serialport/node-serialport #823: Serialport Failure to Install

malleyne, 2016-05-24:
Operating System and Hardware: Raspberry Pi Model B
NodeJS Version: v0.12.6
Serialport version: ??
Python Version (if you're having trouble building serialport): 2.7.9
npm -v: 2.11.2
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior

I expect for serial port to complete a successful install upon
npm install serialport
or
sudo npm install serialport
 
### Actual behavior

running either "npm install serialport" or "sudo npm install serialport" while ssh'd into the raspberry pi from the Max OS version 10.9.5 terminal results in a series of errors: 

Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/Release' (1)
npm ERR! Linux 4.4.9+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport"
npm ERR! node v0.12.6
npm ERR! npm  v2.11.2
npm ERR! code ELIFECYCLE

npm ERR! serialport@3.1.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@3.1.2 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/prosthesis/public/npm-debug.log
[npm-debug.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/278866/npm-debug.txt)

when I attempt to run my main.js file anyway "nodejs main.js" as shown by the instructions here http://www.instructables.com/id/Easy-NodeJS-WebSockets-LED-Controller-for-Raspberr/?ALLSTEPS
I experience the following errors 

/home/pi/node_modules/serialport/node_modules/bindings/bindings.js:83
        throw e
              ^
Error: /home/pi/node_modules/serialport/build/Release/serialport.node: undefined symbol: node_module_register
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at bindings (/home/pi/node_modules/serialport/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/pi/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)

### Steps and/or code to reproduce the issue

I followed the instructions here: http://www.instructables.com/id/Easy-NodeJS-WebSockets-LED-Controller-for-Raspberr/?ALLSTEPS

malleyne, 2016-05-24:
[npm-debug.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/278870/npm-debug.txt)

reconbot, 2016-05-24:
We have a section in our readme which links to a great guide about getting the PI setup. https://github.com/EmergingTechnologyAdvisors/node-serialport#raspberry-pi-linux

Looking at the npm-debug log I'm going to venture a guess you don't have gcc or another c compiler installed. It's kind of hard to tell. 

@nebrius I don't see any mention to installing build tools in your guide. Can you take a look at this issue?

nebrius, 2016-05-24:
@reconbot I haven't come across an instance where it's necessary to install build tools on the pi. I don't think this is missing build tools though, the error for that is more obvious than this. I honestly don't see why it's failing from the logs 😕 

@malleyne can you run `node-pre-gyp install --fallback-to-build` and give us the output from that? Also, which OS are you running on your Pi?

malleyne, 2016-05-25:
1) I installed Raspbian on my Raspberry pi. runnng "uname - a" shows the
following details:

Linux raspberrypi 4.4.9+ #884 Fri May 6 17:25:37 BST 2016 armv6l GNU/Linux

2) I had to install node-pre-gyp which I did by running "sudo npm install
node-pre-gyp -g" while ssh'd into my raspberry pi.

*pi@raspberrypi*:*~ $* sudo npm install node-pre-gyp -g
/usr/local/bin/node-pre-gyp ->
/usr/local/lib/node_modules/node-pre-gyp/bin/node-pre-gyp
node-pre-gyp@0.6.28 /usr/local/lib/node_modules/node-pre-gyp
├── semver@5.1.0
├── nopt@3.0.6 (abbrev@1.0.7)
├── rc@1.1.6 (ini@1.3.4, strip-json-comments@1.0.4, deep-extend@0.4.1,
minimist@1.2.0)
├── mkdirp@0.5.1 (minimist@0.0.8)
├── rimraf@2.5.2 (glob@7.0.3)
├── npmlog@2.0.4 (ansi@0.3.1, gauge@1.2.7, are-we-there-yet@1.1.2)
├── tar-pack@3.1.3 (uid-number@0.0.6, once@1.3.3, debug@2.2.0,
readable-stream@2.0.6, fstream-ignore@1.0.5, fstream@1.0.9)
├── request@2.72.0 (is-typedarray@1.0.0, oauth-sign@0.8.2, aws-sign2@0.6.0,
forever-agent@0.6.1, tunnel-agent@0.4.3, caseless@0.11.0, stringstream@0.0.5,
isstream@0.1.2, json-stringify-safe@5.0.1, extend@3.0.0, aws4@1.4.1,
node-uuid@1.4.7, combined-stream@1.0.5, qs@6.1.0, mime-types@2.1.11,
tough-cookie@2.2.2, form-data@1.0.0-rc4, bl@1.1.2, har-validator@2.0.6,
http-signature@1.1.1, hawk@3.1.3)
└── tar@2.2.1 (inherits@2.0.1, block-stream@0.0.9, fstream@1.0.9)

3) Then I ran the command you suggested above. Below are the errors I see.
Do you think I have to get a different version of node? or something else?
This is my first time working with all of these tools. Thanks for being so
helpful.

*pi@raspberrypi*:*~ $* node-pre-gyp install --fallback-to-build
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.28
node-pre-gyp info using node@0.12.6 | linux | arm
node-pre-gyp ERR! UNCAUGHT EXCEPTION
node-pre-gyp ERR! stack Error: ENOENT, no such file or directory
'./package.json'
node-pre-gyp ERR! stack     at Error (native)
node-pre-gyp ERR! stack     at Object.fs.openSync (fs.js:500:18)
node-pre-gyp ERR! stack     at Object.fs.readFileSync (fs.js:352:15)
node-pre-gyp ERR! stack     at install
(/usr/local/lib/node_modules/node-pre-gyp/lib/install.js:126:38)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function)
[as install]
(/usr/local/lib/node_modules/node-pre-gyp/lib/node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run
(/usr/local/lib/node_modules/node-pre-gyp/bin/node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at Object.<anonymous>
(/usr/local/lib/node_modules/node-pre-gyp/bin/node-pre-gyp:131:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:460:26)
node-pre-gyp ERR! stack     at Object.Module._extensions..js
(module.js:478:10)
node-pre-gyp ERR! stack     at Module.load (module.js:355:32)
node-pre-gyp ERR! System Linux 4.4.9+
node-pre-gyp ERR! command "node" "/usr/local/bin/node-pre-gyp" "install"
"--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi
node-pre-gyp ERR! node -v v0.12.6
node-pre-gyp ERR! node-pre-gyp -v v0.6.28
node-pre-gyp ERR! This is a bug in `node-pre-gyp`.
node-pre-gyp ERR! Try to update node-pre-gyp and file an issue if it does
not help:
node-pre-gyp ERR!     <https://github.com/mapbox/node-pre-gyp/issues>

-mel

nebrius, 2016-05-25:
`node-pre-gyp ERR! stack Error: ENOENT, no such file or directory
'./package.json'`

ack, right, sorry my bad, I left out some important steps 😅. Can you check and see if you have a `node_modules/serialport` directory and run it from that? If not, can you try cloning this repo on your Pi, and run the command there?

This probably won't make a difference since you're running Raspbian, but can you try running `sudo apt-get install build-essential` to install the build tools if they're missing?

malleyne, 2016-05-26:
I don't have  a node_modules/serialport directory. I did try running sudo
apt-get install build-essential in my /home/pi directory which was
successful but running npm install serialport (I also tried running this
command as root) immediately after still did not work for the same reasons
described above. I can't tell from your description if you're suggesting I
clone the node-pre-gyp repo or the serialport repo. I'm also not sure what
command you're suggesting I run after the repo is cloned.

I also took a look at the raspberry pi installation guide on the serialport
github page for serial (
https://github.com/nebrius/raspi-io/wiki/Getting-a-Raspberry-Pi-ready-for-NodeBots)
and it discusses installing johnny-five and raspbi-io which is more than I
need it seems. Also, the sensors I'm adding aren't going to be directly on
the raspberry pi. I'm adding sensors to an arduino mega and want to send
that sensor data somehow to my raspberry pi which is why I started to look
at this serial library.

I appreciate any other suggestions you may have!

nebrius, 2016-05-27:
You are correct about the wiki that you don't need to install Johnny-Five or Raspi IO, and I updated the wiki to make it more clear what the instructions are for node-serialport, thanks!

Sorry for being so unclear, here's the steps I recommend running from your pi:

1. `git clone https://github.com/EmergingTechnologyAdvisors/node-serialport.git`
1. `cd node-serialport`
1. `npm install`
    - I suspect that you will get the same error here that you got previously
1. `node-pre-gyp install --fallback-to-build`

malleyne, 2016-05-27:
@nebrius Below shows the results of executing steps 1-4 above (with some of the details removed to be a little consise). still hitting the same error however when I try to run my main.js script which, atm, only does:

```
/* set up webserver */
var express = require('express'); // web server
app = express();
server = require('http').createServer(app);
io = require('socket.io').listen(server); //web socket server
/* set up serial */
var serialPort = require("serialport").SerialPort
```

I also tried updating pug (aka jade) by running 
```
sudo npm install -g pug
npm install pug-load
```
But that didn't improve anything either. I lastly checked to see the raspberry pi serial installation guide to see the changes you made but I can't tell what's different? It still seems to suggest installing johnny-five and rasp-io?

Thanks,
-mel

Results:
pi@raspberrypi:~ $ git clone https://github.com/EmergingTechnologyAdvisors/node-serialport.git
Cloning into 'node-serialport'...
remote: Counting objects: 3061, done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 3061 (delta 7), reused 6 (delta 6), pack-reused 3046
Receiving objects: 100% (3061/3061), 1.03 MiB | 587.00 KiB/s, done.
Resolving deltas: 100% (1729/1729), done.
Checking connectivity... done.
pi@raspberrypi:~ $ cd node-serialport
pi@raspberrypi:~/node-serialport $ ls
appveyor.yml  AUTHORS.md  bin  binding.gyp  changelog.md  CODE_OF_CONDUCT.md  examples  Gruntfile.js  lib  LICENSE  package.json  PUBLISHING.md  README.md  src  test
pi@raspberrypi:~/node-serialport $ npm install
npm WARN installMany node-pre-gyp was bundled with serialport@4.0.0-beta2, but bundled package wasn't found in unpacked tree
npm WARN peerDependencies The peer dependency eslint@^2.0.0-rc.0 included from eslint-config-standard will no
npm WARN peerDependencies longer be automatically installed to fulfill the peerDependency 
npm WARN peerDependencies in npm 3+. Your application will need to depend on it explicitly.
 
> spawn-sync@1.0.15 postinstall /home/pi/node-serialport/node_modules/spawn-sync
> node postinstall

npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install the latest version of pug instead of jade
-
-
/
> serialport@4.0.0-beta2 install /home/pi/node-serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.0-beta2/serialport-v4.0.0-beta2-node-v14-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.0-beta2 and node@0.12.6 (node-v14 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/node-serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/pi/node-serialport/build'
bindings@1.2.1 node_modules/bindings

.....
.....
.....

pi@raspberrypi:~/node-serialport $ node-pre-gyp install --fallback-to-build
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.28
node-pre-gyp info using node@0.12.6 | linux | arm
node-pre-gyp info validate Running test command: '/usr/local/bin/node --eval 'require(\'/home/pi/node-serialport/build/Release/serialport.node\')''
[serialport] Success: "/home/pi/node-serialport/build/Release/serialport.node" already installed
Pass --update-binary to reinstall or --build-from-source to recompile
node-pre-gyp info ok 

pi@raspberrypi:~/node-serialport $ cd ..
pi@raspberrypi:~ $ cd foo/
pi@raspberrypi:~/foo $ ls
main.js  public
pi@raspberrypi:~/prosthesis $ nodejs main.js 

/home/pi/node_modules/serialport/node_modules/bindings/bindings.js:83
        throw e
              ^
Error: /home/pi/node_modules/serialport/build/Release/serialport.node: undefined symbol: node_module_register
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at bindings (/home/pi/node_modules/serialport/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/pi/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
pi@raspberrypi:~/prosthesis $

reconbot, 2016-05-28:
For the record which raspberry pi distro are you using?

malleyne, 2016-05-28:
Model B. Is that what you mean?

reconbot, 2016-05-28:
Raspbian?

malleyne, 2016-05-28:
Yep! Raspbian.

nebrius, 2016-05-29:
In the installation guide, I added a section specific to serial port. Most people f the instructions are the same for serial port and Johnny-five, the only difference is which module to install from npm, and some post install instructions specific to Raspi IO.

Ok, so looking at the build output, it succeeds this time. The specific error is `node_module_error` part. Normally this happens when a module is compiled with one version of Node and run with another.

By chance are you using nvm or something similar to manage Node versions?

malleyne, 2016-05-30:
I'm not using nvm or any node management software. should I be? what do i
need for?
I flashed raspbian on my raspberry pi's sd card, installed node on my
raspberry pi, then followed the instructions in the instructables page
linked earlier on. I don't have anything else installed.

nebrius, 2016-06-02:
nvm is a platform that makes it pretty straightforward to have multiple versions of node installed on a system simultaneously. It's useful in certain situations, but can also lead to issues. If you don't have a specific need for it (and know that you need it), then I would say don't use it.

So. At this point I'm stumped. I don't know what the problem is I'm afraid.

Maybe a good next step would be to reinstall Raspbian, making sure to use the latest release. Then follow the instructions on the Raspi IO wiki to install the latest version of Node.js, and try again?

jtcmedia, 2016-06-02:
I am running Raspbian Jessie-Lite 2016-05-27 with node 4.4.5 and npm 2.15.5. Serialport installs fine as a local module but fails to install as a global module. See log:

[npm-debug.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/296541/npm-debug.txt)

malleyne, 2016-06-09:
Hey guys, thanks for all the help. I got it to automagically work! maybe it was just plugging the arduino to the rpi but no more errors!

reconbot, 2016-06-09:
yay!

closed 2016-06-09 by reconbot

nebrius, 2016-06-09:
🎉 Glad you got it to work!

lakshyaranganath, 2016-06-16:
Im facing this same issue, any tips on how to make it work? 😢

fivdi, 2016-06-16:
@malleyne there's actually more than one issue mentioned above. It would be better to open a new issue describing exactly what issue you're facing.

lakshyaranganath, 2016-06-16:
oh nvm, i got it to work, i have to install node4.4.5 with arm7 since im using a raspberrypi 2 B 😄

Ipe92, 2016-10-06:
yes

nilesh441, 2017-03-25:
I am getting following error. I have tried almost everything to resolve the error but I am unable to resolve it. I am trying to port nodered with modbus support on Toradex VF61 board with 256MB RAM.

Error Log:

> node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/5.0.0-beta3/serialport-v5.0.0-beta3-node-v48-linux-arm.tar.gz
> node-pre-gyp ERR! Pre-built binaries not found for serialport@5.0.0-beta3 and node@6.10.1 (node-v48 ABI) (falling back to source compile with node-gyp)
> gyp ERR! configure error
> gyp ERR! stack Error: Command failed: /usr/bin/python2 -c import platform; print(platform.python_version());
> gyp ERR! stack Traceback (most recent call last):
> gyp ERR! stack   File "<string>", line 1, in <module>
> gyp ERR! stack   File "/usr/lib/python2.7/platform.py", line 116, in <module>
> gyp ERR! stack     import sys,string,os,re
> gyp ERR! stack ImportError: No module named string
> gyp ERR! stack
> gyp ERR! stack     at ChildProcess.exithandler (child_process.js:204:12)
> gyp ERR! stack     at emitTwo (events.js:106:13)
> gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
> gyp ERR! stack     at maybeClose (internal/child_process.js:886:16)
> gyp ERR! stack     at Socket.<anonymous> (internal/child_process.js:342:11)
> gyp ERR! stack     at emitOne (events.js:96:13)
> gyp ERR! stack     at Socket.emit (events.js:188:7)
> gyp ERR! stack     at Pipe._handle.close [as _onclose] (net.js:501:12)
> gyp ERR! System Linux 4.1.15-v2.5b3+ge6d111c
> gyp ERR! command "/home/root/node-v6.10.1-linux-armv7l/bin/node" "/home/root/node-v6.10.1-linux-armv7l/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/home/root/node-serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/root/node-serialport/build/Release"
> gyp ERR! cwd /home/root/node-serialport
> gyp ERR! node -v v6.10.1
> gyp ERR! node-gyp -v v3.4.0
> gyp ERR! not ok
> node-pre-gyp ERR! build error
> node-pre-gyp ERR! stack Error: Failed to execute '/home/root/node-v6.10.1-linux-armv7l/bin/node /home/root/node-v6.10.1-linux-armv7l/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/root/node-serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/root/node-serialport/build/Release' (1)
> node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/root/node-serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
> node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
> node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
> node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:886:16)
> node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
> node-pre-gyp ERR! System Linux 4.1.15-v2.5b3+ge6d111c
> node-pre-gyp ERR! command "/home/root/node-v6.10.1-linux-armv7l/bin/node" "/home/root/node-serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
> node-pre-gyp ERR! cwd /home/root/node-serialport
> node-pre-gyp ERR! node -v v6.10.1
> node-pre-gyp ERR! node-pre-gyp -v v0.6.34
> node-pre-gyp ERR! not ok
> Failed to execute '/home/root/node-v6.10.1-linux-armv7l/bin/node /home/root/node-v6.10.1-linux-armv7l/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/root/node-serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/root/node-serialport/build/Release' (1)
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! serialport@5.0.0-beta3 install: `node-pre-gyp install --fallback-to-build`
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the serialport@5.0.0-beta3 install script 'node-pre-gyp install --fallback-to-build'.
> npm ERR! Make sure you have the latest version of node.js and npm installed.
> npm ERR! If you do, this is most likely a problem with the serialport package,
> npm ERR! not with npm itself.
> npm ERR! Tell the author that this fails on your system:
> npm ERR!     node-pre-gyp install --fallback-to-build
> npm ERR! You can get information on how to open an issue for this project with:
> npm ERR!     npm bugs serialport
> npm ERR! Or if that isn't available, you can get their info via:
> npm ERR!     npm owner ls serialport
> npm ERR! There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/root/.npm/_logs/2017-03-25T05_50_11_055Z-debug.log

**Any suggestion?**

nebrius, 2017-03-27:
@nilesh441, it looks like something's up with the version of python installed. The core error is this line:

```
gyp ERR! stack ImportError: No module named string
```

Maybe there's some modified version of python being used on this board?

nilesh441, 2017-03-28:
But I am able to import String module standalone.

nebrius, 2017-03-28:
> But I am able to import String module standalone.

Are you able to import `String`, or `string`? The error message has it as `string`, not `String`.