serialport/node-serialport #827: not convenient module_path

Zensey, 2016-05-27:
Operating System and Hardware: Windows 8.1
NodeJS Version:4.4.5
Serialport version: 3.1.2
Python Version (if you're having trouble building serialport): Python 2.7.11

### Expected behavior
`dir \node_modules\serialport\build\Release`
`<DIR>          node-v46-win32-x64`
`<DIR>          node-v46-win32-ia32`

### Actual behavior

`dir \node_modules\serialport\build\Release`
`<DIR>          obj                    `
`serialport.exp     `
`serialport.lib      `
`serialport.map `
`serialport.node`
`serialport.pdb`

### Steps and/or code to reproduce the issue
Upgrade serialport from 2.0.6 to 3.1.2

Earlier it was easy to make 1 universal distribution of program compatible with various systems and architectures, because custom binary builds were accessible under corresponedent directory.
Now it does not work.

It would be great if it works as before.
Previous behaviour can be reverted by changing this:
`    "module_path": "build/{configuration}/",`
to this:
`      "module_path": "./build/{configuration}/{node_abi}-{platform}-{arch}/",.`
in package.json

reconbot, 2016-05-27:
It's the standard path however. We do support alternative paths. See this comment. https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/538#issuecomment-217536942

reconbot, 2016-05-27:
I'm soliciting a download script to download all the prebuilt binaries and put them in the right places too.

closed 2016-05-27 by reconbot