serialport/node-serialport #828: Readme has Incorrect Example

360disrupt, 2016-05-27:
Operating System and Hardware:
NodeJS Version: 5.0.0
Serialport version: 2.0.6

### Expected behavior
Work.

### Actual behavior
`SerialPort is not a function`

### Steps and/or code to reproduce the issue
Wrong:
```
var SerialPort = require("serialport");
var port = new SerialPort("/dev/tty-usbserial1", {
  baudrate: 57600
});
```
Working:
```
var SerialPort = require("serialport");
var port = new SerialPort.SerialPort("/dev/tty-usbserial1", {
  baudrate: 57600
});

```

reconbot, 2016-05-27:
The Readme is for the current master beta, the docs you'd like to be looking at are linked at the top. I highly recommend upgrading to the latest 3.x release.

https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/2.0.6/README.md

closed 2016-05-27 by reconbot

360disrupt, 2016-05-27:
Sry have jumped to quick into the examples.

reconbot, 2016-05-27:
No worries, easy mistake to make. It was a lot harder before we had those links. I made the text calling that out really large because I guarantee you I've made that mistake before.