serialport/node-serialport #838: Getting back 'true' for error when using serialport with electron

jjshoe, 2016-05-31:
Operating System and Hardware: ubuntu 14.04 desktop
NodeJS Version: 4.4.0
Serialport version: 3.1.2
Python Version (if you're having trouble building serialport): 2.7.6
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
The port to open, or at minimum more useful error details

### Actual behavior

Port doesn't open, and I'm given an error of ```true```
### Steps and/or code to reproduce the issue

At first I was getting hit by the npm mismatch error. With the help of #538 I ran ```../node-pre-gyp/bin/node-pre-gyp install --fallback-to-build --runtime=electron --target=1.1.0 --target_arch=x64 --update-binary --dist-url=https://atom.io/download/atom-shell``` in my node_modules/serialport directory. This quickly cleared up this issue.

Good, non-electron code: https://gist.github.com/jjshoe/79b161cd4abef34f23f13b4c79e73044
Bad, electron code: https://gist.github.com/jjshoe/86f99a8a7126c90f7c73db94cd1c7de0
Bad, electron code, output: https://gist.github.com/jjshoe/f9e25a1409001dd31214d0b2373f4b27

I tried to modify the C code to get better output, but C is way beyond my abilities. It would have been nice to see the raw data.

reconbot, 2016-05-31:
It's not even running our c++ code for that error. I'm for better
documentation however, happy to take a PR on the install section of the
readme.

jjshoe, 2016-05-31:
@reconbot thoughts on how to further trace down what's causing that error?

jjshoe, 2016-05-31:
at L209 in my code I see:

```
206   SerialPortBinding.open(this.path, this.options, function(err, fd) {
207     if (err) {
208       debug('SerialPortBinding.open had an error', err);
209       return this._error(err, callback);
210     }
```

I just assumed that that binding is what's used to interact with the C code.

jjshoe, 2016-05-31:
strace seems to indicate that it is indeed making it to the C code, for example, running:
```
strace -f electron serialport-terminal.js -p /dev/ttyUSB0 -b 2400 >> strace.output 2>&1
```

I can see:
```
[pid 31856] open("/dev/ttyUSB0", O_RDWR|O_NOCTTY|O_NONBLOCK|O_SYNC|O_CLOEXECProcess 31855 attached
```

reconbot, 2016-05-31:
I assumed `npm mismatch` meant node api version mismatch

jjshoe, 2016-05-31:
@reconbot my issues comments on how I resolved that to get where I'm at, aka complete steps to reproduce.

reconbot, 2016-06-01:
I'm finally getting to read this through properly. 

A few notes while I try to figure out the larger problem.
 - `baudRate` only needs to be defined once, the camel case is my preference but either works.
-  If you run your app with the env var `DEBUG=*` you'll get those debug outputs. `true` is never returned as an error from the bindings and we have tests around that for `open`. The code for that particular return path is here.

https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/3.1.2/src/serialport.cpp#L136-L157

I'd be willing to guess something is wonkey with your compiled code. Nothing in the examples you provided show that your coercing the error to a a boolean. Maybe try logging `typeof error` right away? If you think there's an issue in serialport, it would help if you passed `false` to `openImmediatly` (the optional 3rd argument in the constructor) and called `open()` with a callback.

jjshoe, 2016-06-01:
I'll see if I get anything different by not auto opening the port in my app, but this is what I get when using DEBUG=* on the serialport-terminal.js app that ships with node-serialport. 

```
[jjshoe@joel-desktop (AWS: prod)] /home/jjshoe/electron/node_modules/serialport/bin$ DEBUG=* electron serialport-terminal.js -p /dev/ttyUSB0 
  serialport SerialPortBinding.open had an error +0ms true
Error true
[jjshoe@joel-desktop (AWS: prod)] /home/jjshoe/electron/node_modules/serialport/bin$ 
```

jjshoe, 2016-06-01:
```
var serialport = require('serialport');
var SerialPort = serialport.SerialPort;

var port = new SerialPort('/dev/ttyUSB0', {baudRate: 2400, autoOpen: false, parser: serialport.parsers.readline('\r')});

port.open(function (err) {
  if (err) {
    console.log(typeof err);
    return console.log('Error opening port: ', err.message);
  }

});

port.on('error', function(err) {
  console.log('port.on(error) Error: ', err.message);
});
```

outputs:
```
port.on(error) Error:  Port is opening
boolean
Error opening port:  undefined
```

reconbot, 2016-06-01:
It's super strange to me in your first output that you could get a stack from a boolean value.

reconbot, 2016-06-01:
`autoOpen: false` is only in the latest beta, make sure you're looking at the right docs.

reconbot, 2016-06-01:
You could try;

```js
var port = new SerialPort('/dev/ttyUSB0', {baudRate: 2400, parser: serialport.parsers.readline('\r')}, false);

port.open(function (err) {
  if (err) {
    console.log(typeof err);
    console.log(err);
  }
});
port.on('error', function(err) {
  console.log('port.on(error) Error: ', err);
});
```

jjshoe, 2016-06-01:
My bad, so here's what I have, to cover all my bases:
```
var serialport = require('serialport');
var SerialPort = serialport.SerialPort;

var port = new SerialPort('/dev/ttyUSB0', {baudRate: 2400, openImmediatly: false, autoOpen: false, parser: serialport.parsers.readline('\r')}, false);

port.open(function (err) {
  if (err) {
    console.log(typeof err);
    return console.log('Error opening port: ', err.message);
  }

});

port.on('error', function(err) {
  console.log('port.on(error) Error: ', err.message);
});
```

And output:
```
boolean
Error opening port:  undefined
```

jjshoe, 2016-06-01:
At one point I tried to see the raw data, by changing null to data, but it complained about not being able to cast it:

https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/3.1.2/src/serialport.cpp#L144

Would be interesting to see what data is at that point, no?

jjshoe, 2016-06-01:
Not sure if this will help:
```
DEBUG=* strace -f electron serialport-terminal.js -p /dev/ttyUSB0
```

Output:
https://gist.github.com/jjshoe/3263c7686ebcd865c31e9f802b05a727

reconbot, 2016-06-01:
The strace shows me we open the port, set it up and set options on it.

Data is not very interesting. It contains the setup info, and for output an int that is the file descriptor if any and the error string if any.

You can find the struct here 
https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/3.1.2/src/serialport.h#L66-L87
And the unix open code here https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/3.1.2/src/serialport_unix.cpp#L125-L142

If you want to debug in the c++ code you can use with [printf](http://www.cplusplus.com/reference/cstdio/printf/) and recompile the local c++ with `npm run gyp-rebuild`

reconbot, 2016-06-01:
@jjshoe try to `printf` `data->errorString`. If we're casting the error object to a boolean it's probably going on in the JS side, as c++ is statically typed and that can't happen by accident. Why that is only happening in electron... I have no idea.

jjshoe, 2016-06-01:
@reconbot - Which block of code do you want me to put that in? and I'm assuming something like

```
printf("%c\n", data->errorString);
```

jjshoe, 2016-06-01:
src/serialport_unix.cpp:

```
125 void EIO_Open(uv_work_t* req) {
126   OpenBaton* data = static_cast<OpenBaton*>(req->data);
127 
128   int flags = (O_RDWR | O_NOCTTY | O_NONBLOCK | O_CLOEXEC | O_SYNC);
129   int fd = open(data->path, flags);
130 
131   printf("unix [%c]\n", data->errorString);
```

src/serialport.cpp 
```
141   v8::Local<v8::Value> argv[2];
142   if (data->errorString[0]) {
143     printf("gen 1 [%c]\n", data->errorString);
144     argv[0] = v8::Exception::Error(Nan::New<v8::String>(data->errorString).ToLocalChecked());
145     argv[1] = Nan::Undefined();
146     // not needed because we're not calling AfterOpenSuccess
147     delete data->dataCallback;
148     delete data->errorCallback;
149     delete data->disconnectedCallback;
150   } else {
151     printf("gen 2 [%c]\n", data->errorString);
152     argv[0] = Nan::Null();
153     argv[1] = Nan::New<v8::Int32>(data->result);
154 
155     int fd = argv[1]->ToInt32()->Int32Value();
156     newQForFD(fd);
157 
158     AfterOpenSuccess(data->result, data->dataCallback, data->disconnectedCallback, data->errorCallback);
159   }
```

output:
```
[jjshoe@joel-desktop (AWS: prod)] /home/jjshoe/examble_b$ DEBUG=* electron index.js 
unix []
gen 2 []
  serialport SerialPortBinding.open had an error +0ms true
boolean
Error opening port:  undefined
```

reconbot, 2016-06-01:
"gen 2"  is the happy path saying there were no errors.

---
Francis Gulotta
wizard@roborooter.com

jjshoe, 2016-06-01:
That's what I thought, but I still get the error you see above! I'm so lost, haha.

jjshoe, 2016-06-01:
I tried changing https://github.com/EmergingTechnologyAdvisors/node-serialport/blame/master/src/serialport.cpp#L171 back to ```argv[0] = Nan::Undefined();``` but that was not well received:

```
[jjshoe@joel-desktop (AWS: prod)] /home/jjshoe/examble_b$ DEBUG=* electron index.js 
unix []
gen 2 []
undefined

#
# Fatal error in , line 0
# unreachable code
#

==== C stack trace ===============================

 1: V8_Fatal
 2: 0x7f9081a1fd87
 3: v8::internal::ToBooleanICStub::UpdateStatus(v8::internal::Handle<v8::internal::Object>)
 4: v8::internal::Runtime_ToBooleanIC_Miss(int, v8::internal::Object**, v8::internal::Isolate*)
 5: 0x1745160092a7
```

reconbot, 2016-06-01:
What's the difference with electron? I'm not very familiar beyond the c++ bindings. Are you you able to communicate with the node process directly or do you have to pass messages between the window and node?

You you should be able to change it back to `Nan::Undefined()` without issue. Tests would fail but besides that it would work.

jjshoe, 2016-06-01:
I'm not super node savy either, so I can't comment on what's different between electron and stock nodejs. You are able to communicate with the node process directly. The code samples I've provided have the JS executing in the main "event loop" (probably bad wording). None of the code samples interact with any browser windows at all. 

As stated above, I can't get Nan::Undefined to work at all.

Another interesting thing I tried, and I assume this is a red flag, but again, this isn't my main skill set:
```
  if (data->errorString[0]) {
    printf("gen 1 [%c]\n", data->errorString);
    argv[0] = v8::Exception::Error(Nan::New<v8::String>(data->errorString).ToLocalChecked());
    argv[1] = Nan::Undefined();
    // not needed because we're not calling AfterOpenSuccess
    delete data->dataCallback;
    delete data->errorCallback;
    delete data->disconnectedCallback;
  } else {
    printf("gen 2 [%c]\n", data->errorString);
    //argv[0] = Nan::Undefined();
    argv[0] = Nan::Null();
    argv[1] = Nan::New<v8::Int32>(data->result);

    int fd = argv[1]->ToInt32()->Int32Value();
    newQForFD(fd);

    AfterOpenSuccess(data->result, data->dataCallback, data->disconnectedCallback, data->errorCallback);
  }

  printf("d [%c]\n", argv[0]);
  data->callback->Call(2, argv);
```

Then I get the following output:
```
[jjshoe@joel-desktop (AWS: prod)] /home/jjshoe/examble_b$ DEBUG=* electron index.js 
unix []
gen 2 []
d [p]
boolean
true
  serialport SerialPortBinding.open had an error +0ms true
boolean
Error opening port:  undefined
```

Last I checked ```p``` would certainly evaluate to a boolean true!

I entered enough debugging to confirm ```argv[0] = Nan::Null();``` is what's setting it to ```p``` for me.

asincrono, 2016-06-01:
You are not alone. Here too inside Electron.

Nodejs system version: 6.2.0.
Electron version: 1.2.0 (packs node inside 6.1.0)
npm version: 3.8.9

Was working right a couple of days ago.
I updated something, I imagine, and now Chaos!

The error I found is even weirder (crashes electron without errors):

### Expected behavior

A list of ports.

### Actual behavior

Returns err true... and the list of ports.
Working with the list of ports leads to Electron crash.

### Steps and/or code to reproduce the issue

```
const serialport = require('serialport')
serialport.list((err, ports) => {
  console.log('err:', err)
  console.log('ports:', ports)
})
```
returns 
```
err: true
ports: Array[2]
```
being the first object the first port with its property `manufacturer: undefined`

Now the cool part: if I try: 
`if (ports[0].manufacturer) {...}`
Electron crashes.

On top of that, if I try:
`console.log (ports[0].manufacturer === undefined)`
the output is:
`false`

asincrono, 2016-06-01:
I went back to electron 1.1.2... I think... and is back working.
(I wrote "I think" cause it's showing node 6.1.0, can't remember if electron 1.1.2 packed that version of node).
Sorry I'm a bit tired not sure how to write properly.

jjshoe, 2016-06-01:
@asincrono which version of node-serialport?

asincrono, 2016-06-01:
Well... confirmed. So looks like is an Electron issue.
**Summary:**
serialport works ok with electron 1.1.2
serialport becomes a nightmare with electron 1.2.0 (as far as I can see).

**What I have:**
In my system I have electron version 1.2.0 -> serialport fails.

**What I did:**
Installed electron-prebulit 1.1.2  (-> got electron 1.1.2 in /app/node_modules/.bin/)
So I could try both versions of electron agains my app/serialport.
To do that I edited  "package.json" and wrote:
```
...
  "dependencies": {
    "electron-prebuilt": "1.1.2",
    "material-design-icons": "^2.2.3",
    **"serialport": "^4.0.0-beta3",**
    "usb-detection": "^1.4.0"
  }
...
```
 **"electron-prebuilt": "1.1.2",** is the important thing, serialport 4.0.0-beta3 isn't really necessary.
Then I ran:
`npm install`

And then, to start my app:
(from the app folder)
`node_modules/.bin/electron .`

Hope could be of any help to the ppl in a hurry.
(sorry if this isn't the right way to report, I'm very tired and this is my fir... second, my second time here)

jjshoe, 2016-06-01:
@asincrono I use serial port 3.1.2, I rolled back to electron-prebuilt 1.1.2 and all is working.

@reconbot do you want to handle making an issue with upstream, or?

asincrono, 2016-06-01:
@jjshoe serialport 4.0.0-beta3 with electron-brebuilt 1.1.2 here and working too.

reconbot, 2016-06-01:
If you could make the issue and ping me I'd apriciate it.

asincrono, 2016-06-02:
Don't know if this will be of any help.

I'm doing an electron project to run in OS X and Linux.
**OS X (el capitán)** Electron 1.2.0/1.2.1 + serialport.list() --> failed: returning err = true and a malformed? array (-> accessing it's contents would cause crash).
**Linux (Ubuntu mate at raspberry pi)** Electron 1.2.1 + serialport.list() --> working ok.

enami, 2016-06-07:
I had similar problem with electron 1.2.1 and serialport 3.1.2 on OSX.  The electron-rebuild command won't rebuild serialport.

The work around I found is to patch electron-rebuild to invoke npm rebuild with --build-from-source option.  After that, electron-rebuild -f seems to rebuild serialport and serialport can open port successfully.

asincrono, 2016-06-11:
@enami what do you mean you patcher electron-rebuild. Which code did you change?
Did you mean to:

1. modify `node_modules/electron-rebuild/src/main.js`
2. and add `--build-from-source` to the command `rebuild` args at line 122(ish)?
```
args.push(
  '--runtime=electron',
  `--target=${nodeVersion}`,
  `--arch=${arch || process.arch}`,
  '--build-from-source'
)
```
(**'--build-from-source'** being the new line).
3. then inside `node_modules/electron-rebuild` `npm install`
???

... because it worked for me too.
(Thanks a lot)

ryanpardieck, 2016-06-13:
I saw this same issue yesterday when trying to use node serialport with nwjs. I assumed it was because I was using node serialport 3.1.2 with a version of nwjs that ran node 6.x, which 3.1.2 doesn't support. I downgraded to the LTS release of nwjs and everything worked.

reconbot, 2016-06-15:
3.1.2 should support node 6.1 at the very least, possibly node 6.2

reconbot, 2016-06-15:
I'm going to close this for now if you'd like to continue with it feel free to reopen.

closed 2016-06-15 by reconbot