serialport/node-serialport #842: Failure to Install - Raspbian Jessie with node.js v6.2.0 and npm v3.8.9

nathanielknopf, 2016-06-07:
Operating System and Hardware: Raspberry Pi 3 B running Raspbian Jessie
NodeJS Version: v6.2.0
Serialport version: 3.1.2
Python Version (if you're having trouble building serialport): 2.7.9
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
should install with npm install serialport or sudo npm install serialport

### Actual behavior
Lots and lots of errors. In particular: 

"
node-pre-gyp ERR! Tried to download: https://github.com/voodootikigod/node-serialport/releases/download/3.1.2/serialport-v3.1.2-node-v48-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@3.1.2 and node@6.2.0 (node-v48 ABI) (falling back to source compile with node-gyp) 
"
Full error message on pastebin [here](http://pastebin.com/HeXN0hbf)
npm-debug.log as attachment: [npm-debug.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/303515/npm-debug.txt)

### Steps and/or code to reproduce the issue
I'm running node.js version 6.2.0 on a raspberry pi 3B running Raspbian Jessie. I'm using npm v3.8.9.

Is this just a version incompatibility issue? It seems like it can't find anything for v6.2.0 of node.js, but maybe I'm misunderstanding the error. Thank you!

nathanielknopf, 2016-06-07:
I have also tried cloning the repo at https://github.com/EmergingTechnologyAdvisors/node-serialport.git with git and running $ npm install but have had no luck with that either.

nathanielknopf, 2016-06-09:
Did the following things and it seemed to work:

    - installed node-pre-gyp with npm
    - changed the owner of ~/node_modules/ to the user rather than root
    - used git to clone the module from https://github.com/EmergingTechnologyAdvisors/node-serialport.git
    - npm install inside the node-serialport folder
    - moved the folder inside node_modules

Seems to work now

closed 2016-06-09 by nathanielknopf

daned33, 2016-06-09:
I also seem to not be able to install it, despite trying your solution.

reconbot, 2016-06-09:
Dan,
That is frustrating. Could you open a new issue with the npm debug log
attached and the steps you've tried and how you setup your pi? Most of the
time it just works. We'll see if we can help you debug.

daned33, 2016-06-09:
Reconbot,
Opened a new issue.