serialport/node-serialport #843: npm update fails when serialport was built manually

infusion, 2016-06-09:
Operating System and Hardware: Raspberry PI B+
NodeJS Version: v4.2.1
Serialport version:2.14.7
Python Version : Python 2.7.9

As said earlier at #833, I installed serialport on Raspberry PI and as there are no pre-built ARM packages, they were compiled manually by npm. Everything works fine with serialport, but another package was updated so I ran `npm update` and it failed for serialport now.

```
➜  ~ npm update
npm WARN engine serialport@1.2.5: wanted: {"node":"0.10.x"} (current: {"node":"4.2.1","npm":"2.14.7"})
-
> serialport@1.2.5 install /home/pi/node_modules/gps/node_modules/serialport
> node-gyp rebuild

make: Entering directory '/home/pi/node_modules/gps/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:0:
../src/serialport.h:30:38: error: ‘Arguments’ in namespace ‘v8’ does not name a type
 v8::Handle<v8::Value> List(const v8::Arguments& args);
                                      ^
../src/serialport.h:31:15: error: variable or field ‘EIO_List’ declared void
 void EIO_List(uv_work_t* req);
               ^
../src/serialport.h:31:15: error: ‘uv_work_t’ was not declared in this scope
../src/serialport.h:31:26: error: ‘req’ was not declared in this scope
 void EIO_List(uv_work_t* req);
                          ^
../src/serialport.h:32:20: error: variable or field ‘EIO_AfterList’ declared void
 void EIO_AfterList(uv_work_t* req);
                    ^
../src/serialport.h:32:20: error: ‘uv_work_t’ was not declared in this scope
../src/serialport.h:32:31: error: ‘req’ was not declared in this scope
 void EIO_AfterList(uv_work_t* req);
                               ^
../src/serialport.h:34:38: error: ‘Arguments’ in namespace ‘v8’ does not name a type
 v8::Handle<v8::Value> Open(const v8::Arguments& args);
                                      ^
../src/serialport.h:35:15: error: variable or field ‘EIO_Open’ declared void
 void EIO_Open(uv_work_t* req);
               ^
../src/serialport.h:35:15: error: ‘uv_work_t’ was not declared in this scope
../src/serialport.h:35:26: error: ‘req’ was not declared in this scope
 void EIO_Open(uv_work_t* req);
                          ^
../src/serialport.h:36:20: error: variable or field ‘EIO_AfterOpen’ declared void
 void EIO_AfterOpen(uv_work_t* req);
                    ^
../src/serialport.h:36:20: error: ‘uv_work_t’ was not declared in this scope
../src/serialport.h:36:31: error: ‘req’ was not declared in this scope
 void EIO_AfterOpen(uv_work_t* req);
                               ^
In file included from ../src/serialport.cpp:3:0:
../src/serialport.h:39:39: error: ‘Arguments’ in namespace ‘v8’ does not name a type
 v8::Handle<v8::Value> Write(const v8::Arguments& args);
                                       ^
../src/serialport.h:40:16: error: variable or field ‘EIO_Write’ declared void
 void EIO_Write(uv_work_t* req);
                ^
../src/serialport.h:40:16: error: ‘uv_work_t’ was not declared in this scope
../src/serialport.h:40:27: error: ‘req’ was not declared in this scope
 void EIO_Write(uv_work_t* req);
                           ^
../src/serialport.h:41:21: error: variable or field ‘EIO_AfterWrite’ declared void
 void EIO_AfterWrite(uv_work_t* req);
                     ^
../src/serialport.h:41:21: error: ‘uv_work_t’ was not declared in this scope
../src/serialport.h:41:32: error: ‘req’ was not declared in this scope
 void EIO_AfterWrite(uv_work_t* req);
                                ^
../src/serialport.h:43:39: error: ‘Arguments’ in namespace ‘v8’ does not name a type
[... 234 lines cut ...]
gyp ERR! cwd /home/pi/node_modules/gps/node_modules/serialport
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok 
npm ERR! Linux 4.4.8+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "update"
npm ERR! node v4.2.1
npm ERR! npm  v2.14.7
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.2.5 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.2.5 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log
```

fivdi, 2016-06-10:
#833 was about the installation of serialport@3.1.2. serialport@3.1.2 is currently the latest non-beta version of serialport and it works fine with Node.js v4.2.1 (which is the version of Node.js for this issue  and #833.) However, this issue is being reported for serialport@1.2.5. serialport@1.2.5 was released many years ago and doesn't support Node.js v4.2.1. The `gps` package which is being installed here appears to depend on serialport@1.2.5, so it can't function correctly with Node.js v4.1.2.

infusion, 2016-06-10:
But why does this problem occur only for updates? The dependency of `gps` works fine for an installation and should use any version (preferable the newest).

fivdi, 2016-06-10:
The information returned by `npm info gps` shows that gps@0.1.0, which is the highest version of `gps`, was released on '2013-10-04T16:29:15.812Z', which was a long time ago. On the other hand, gps@0.0.10 was released on '2016-06-09T15:16:20.695Z', which was yesterday, but there is already the higher version v0.1.0 from 2013.

[npm update](https://docs.npmjs.com/cli/update) will update all the packages to the latest version respecting semver. To me, this means it will attempt to install gps@0.1.0. 

The package.json file for gps@0.1.0 contains the following dependencies which means it depends on serialport@1.2.5:

```
  "dependencies": {
    "serialport": "~1.2.2"
  }
```

Given that you're the author of `gps`, I'd suggest publishing a new version on npm with a version number higher than v0.1.0. Then try everything again to see if it works.

```
pi@raspberrypi:~ $ npm info gps
 
{ name: 'gps',
  description: 'A GPS NMEA parser library',
  'dist-tags': { latest: '0.0.10' },
  versions: [ '0.0.6', '0.0.7', '0.0.8', '0.0.9', '0.0.10', '0.1.0' ],
  maintainers: 
   [ 'andrewnez <andrewnez@gmail.com>',
     'infusion <robert@xarg.org>' ],
  time: 
   { modified: '2016-06-09T15:16:20.695Z',
     created: '2013-10-04T16:29:12.642Z',
     '0.1.0': '2013-10-04T16:29:15.812Z',
     '0.0.6': '2016-03-28T12:11:09.457Z',
     '0.0.7': '2016-04-05T13:40:30.024Z',
     '0.0.8': '2016-04-07T17:14:40.122Z',
     '0.0.9': '2016-05-29T09:54:16.506Z',
     '0.0.10': '2016-06-09T15:16:20.695Z' },
  author: 'Robert Eisele <robert@xarg.org> (http://www.xarg.org/)',
  repository: { type: 'git', url: 'git://github.com/infusion/GPS.js.git' },
  homepage: 'https://github.com/infusion/GPS.js',
  keywords: 
   [ 'nmea',
     'gps',
     'serial',
     'parser',
     'distance',
     'geo',
     'location',
     'rmc',
     'gga',
     'gll',
     'gsa',
     'vtg',
     'gva' ],
  bugs: { url: 'https://github.com/infusion/GPS.js/issues' },
  readmeFilename: 'README.md',
  license: 'MIT OR GPL-2.0',
  title: 'gps.js',
  version: '0.0.10',
  main: 'gps',
  private: false,
  directories: { example: 'examples' },
  engines: { node: '*' },
  scripts: { test: 'mocha tests/*.js' },
  devDependencies: 
   { express: '*',
     'socket.io': '*',
     serialport: '*',
     angles: '*',
     mocha: '*',
     chai: '*',
     byline: '*' },
  gitHead: '6307aaa76f1ca05b00a05a013e60a74c44906e6b',
  dist: 
   { shasum: 'd0cc2fb3d65cc29cfacddcd48172b2d1ce6f0708',
     tarball: 'https://registry.npmjs.org/gps/-/gps-0.0.10.tgz' } }
```

infusion, 2016-06-10:
Ouch! That explains a lot, thanks!

closed 2016-06-10 by infusion