serialport/node-serialport #844: Failure to Install - RPi2 - Raspbian Jessie with node.js v4.4.5 and npm v2.15.5

daned33, 2016-06-09:
Operating System and Hardware: Raspbian Jessie - RPi2
NodeJS Version: v4.4.5
Npm Version: 2.15.5
Serialport version: ?
Python Version (if you're having trouble building serialport): 2.7.3
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior

should install with npm install serialport or sudo npm install serialport

### Actual behavior

Lots errors. In particular:

"
node-pre-gyp ERR! Tried to download: https://github.com/voodootikigod/node-serialport/releases/download/3.1.2/serialport-v3.1.2-node-v48-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@3.1.2 and node@6.2.0 (node-v48 ABI) (falling back to source compile with node-gyp) 
"

Partial error log from terminal: http://pastebin.com/dfZ7sd1M
npm-debug.log as attachment: http://termbin.com/hbds or [npm-debug.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/307835/npm-debug.txt)

### Steps and/or code to reproduce the issue

install with npm install serialport or sudo npm install serialport

openhardwarecoza, 2016-06-09:
https://github.com/openhardwarecoza/LaserWeb3/wiki/Raspberry-Pi-Image-Build

daned33, 2016-06-09:
Tried doing that already, while doing npm install, it throws the same error.

fivdi, 2016-06-10:
It looks like Raspbian Wheezy and gcc/g++ v4.6 are being used. In order to install native addons with Node.js v4+, gcc/g++ 4.8+ are required. See https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons

openhardwarecoza, 2016-06-10:
Wheezy?  @Daniel, why arent you on Jessie yet?  Goodness

daned33, 2016-06-10:
Hmm, odd, I thought it was Jessie. Granted I've not reinstalled said pi in
quite some time. (it was installed using a prebuilt image with octoprint)

Thanks though I'll try it out later.

daned33, 2016-06-10:
Got it working, Thanks.

closed 2016-06-10 by daned33