serialport/node-serialport #845: Unable to install on Raspbian on RPi 3

mimiflynn, 2016-06-10:
Operating System and Hardware:
NodeJS Version: 6.2.1
NPM Version: 3.9.3
Serialport version: 3.1.2
Python Version (if you're having trouble building serialport): 2.7.9
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior

Expected `serialport` to successfully install

### Actual behavior

Returns the following error:

```
$ sudo npm install -g serialport
/usr/local/bin/serialportlist -> /usr/local/lib/node_modules/serialport/bin/serialport-list.js
/usr/local/bin/serialportterm -> /usr/local/lib/node_modules/serialport/bin/serialport-terminal.js

> serialport@3.1.2 install /usr/local/lib/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/voodootikigod/node-serialport/releases/download/3.1.2/serialport-v3.1.2-node-v48-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@3.1.2 and node@6.2.1 (node-v48 ABI) (falling back to source compile with node-gyp)
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.2.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/node_modules/serialport/.node-gyp"
make: Entering directory '/usr/local/lib/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/6.2.1/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/local/lib/node_modules/serialport/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Linux 4.1.21-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/local/lib/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/usr/local/lib/node_modules/serialport/build/Release"
gyp ERR! cwd /usr/local/lib/node_modules/serialport
gyp ERR! node -v v6.2.1
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:852:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
node-pre-gyp ERR! System Linux 4.1.21-v7+
node-pre-gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/local/lib/node_modules/serialport
node-pre-gyp ERR! node -v v6.2.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.28
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/serialport/build/Release' (1)
npm ERR! Linux 4.1.21-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "-g" "serialport"
npm ERR! node v6.2.1
npm ERR! npm  v3.9.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@3.1.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@3.1.2 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.
```

### Steps and/or code to reproduce the issue

Fresh install RPi 3 with latest Raspbian. Update and upgrade apt-get. Install nodejs via command line following the nodejs.org instructions for debian / ubuntu install.

`npm install -g serialport`

fivdi, 2016-06-10:
Why not install serialport locally rather than globally?

If it's essential to install globally try with `sudo npm install serialport -g --unsafe-perm`

fivdi, 2016-06-17:
@mimiflynn did you manage to get serialport installed?

mimiflynn, 2016-06-17:
Sorry for my silence, this is a weekend project and I've not had a chance
to get back to it. I will this coming Sunday.

fivdi, 2016-06-17:
Ah, ok. No problem.

mimiflynn, 2016-06-18:
Thanks for checking in, btw.

I went ahead and installed it locally instead of globally and it appeared to be a success, but when I attempted to output ports [with this script](https://github.com/mimiflynn/arduino-node-temp-humidity/blob/master/nodejs/arduino/portUtil.js) it gave me the following error:

```
[Fri Jun 17 10:57 pm] $ node portUtil.js
/home/mimiflynn/Projects/arduino-node-temp-humidity/nodejs/node_modules/serialport/node_modules/bindings/bindings.js:83
        throw e
        ^

Error: Module version mismatch. Expected 48, got 46.
    at Error (native)
    at Object.Module._extensions..node (module.js:568:18)
    at Module.load (module.js:458:32)
    at tryModuleLoad (module.js:417:12)
    at Function.Module._load (module.js:409:3)
    at Module.require (module.js:468:17)
    at require (internal/module.js:20:19)
    at bindings (/home/mimiflynn/Projects/arduino-node-temp-humidity/nodejs/node_modules/serialport/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/mimiflynn/Projects/arduino-node-temp-humidity/nodejs/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:541:32)
```

fivdi, 2016-06-18:
The `"Error: Module version mismatch. Expected 48, got 46."` is occurring because serialport was installed using Node.js 4.x but `node portUtil.js` was run using Node.js v6.x.

The module version is the [application binary interface (ABI) version number for Node.js](https://nodejs.org/en/download/releases/). Node.js ABI version numbers typically change with every new major Node.js release. The Node.js version used to install native addons like serialport and the Node.js version used at runtime must have the same ABI version number.

Is `nvm` being used and did you perhaps forget to switch to Node.js v6 before installing serialport?

fivdi, 2016-06-18:
or are different user accounts (mimiflynn, pi, root) with different versions of Node.js in their PATH being used at different times?

mimiflynn, 2016-06-21:
Thanks so much for taking the time to look into this. I deleted and reinstalled the `node_modules` and `serialport` is installed and working as expected.

I had installed the project, logged out, then later decided to update via `n` causing the issue.

closed 2016-06-21 by mimiflynn