serialport/node-serialport #847: Could not locate bindings file, rebuild does not help.

Math3v, 2016-06-11:
Operating System and Hardware:
OS: Windows 7 64-bit
Hardware: Custom pressure pad communicating through virtual COM port, connected via USB.

NodeJS Version: v5.10.0
Serialport version: 3.1.2
Python Version (if you're having trouble building serialport): 2.7.11
### Expected behavior
Read data from COM port.
Serialport module is used within Electron application.
### Actual behavior
Error: Could not locate bindings file, after application start.
### Steps and/or code to reproduce the issue
Error message is thrown right after require command.
Reinstalling serialport module does not help, neither do installing node-gyp globally and rebuilding module.

reconbot, 2016-06-11:
if you're rebuilding for electron you need to use `electron-rebuild`

Math3v, 2016-06-12:
Thank you, for your help.

I have tried rebuilding it with electron-rebuild a while ago, but it did not help.

Math3v, 2016-06-14:
`node-gyp configure`
`node-gyp rebuild --target=0.XY.Z --arch=x32 --dist-url=https://atom.io/download/atom-shell`

solved it for me!

closed 2016-06-14 by Math3v