serialport/node-serialport #852: Unable to install on CHIP

jeffersonbenson, 2016-06-12:
Operating System and Hardware: CHIP Debian Jessie
NodeJS Version: v0.10.29 NPM 1.4.21
Serialport version: N/A
Python Version (if you're having trouble building serialport): 2.7.9
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
Serialport should install with little to no issue
### Actual behavior
Serialport fails to install
After multiple attempts, the ability to build from source appears to not be working on the CHIP. 
### Steps and/or code to reproduce the issue
sudo npm install serialport --build-from-source.

```
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/chip/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.11-ntc
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "build" "--fallback-to-build" "--module=/home/chip/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/chip/node_modules/serialport/build/Release"
gyp ERR! cwd /home/chip/node_modules/serialport
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp build --fallback-to-build --module=/home/chip/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/chip/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/chip/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:755:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:822:5)
node-pre-gyp ERR! System Linux 4.4.11-ntc
node-pre-gyp ERR! command "node" "/home/chip/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/chip/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.29
node-pre-gyp ERR! node-pre-gyp -v v0.6.28
node-pre-gyp ERR! not ok 
Failed to execute 'node-gyp build --fallback-to-build --module=/home/chip/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/chip/node_modules/serialport/build/Release' (1)
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! serialport@3.1.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@3.1.2 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.4.11-ntc
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "serialport" "--build-from-source"
npm ERR! cwd /home/chip
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/chip/npm-debug.log
npm ERR! not ok code 0
```

reconbot, 2016-06-13:
Try installing the nodejs-legacy package so the node binary is `node` instead of `nodejs`

jeffersonbenson, 2016-06-13:
So after trying this I receive the following output:

```
/usr/local/bin/serialportlist -> /usr/local/lib/node_modules/serialport/bin/serialport-list.js
/usr/local/bin/serialportterm -> /usr/local/lib/node_modules/serialport/bin/serialport-terminal.js

> serialport@3.1.2 install /usr/local/lib/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/usr/local/lib/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/./serialport.h:7:0,
                 from ../src/serialport.cpp:1:
../node_modules/nan/nan.h:324:47: error: 'REPLACE_INVALID_UTF8' is not a member of 'v8::String'
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
serialport.target.mk:84: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/usr/local/lib/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.11-ntc
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "build" "--fallback-to-build" "--module=/usr/local/lib/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/usr/local/lib/node_modules/serialport/build/Release"
gyp ERR! cwd /usr/local/lib/node_modules/serialport
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp build --fallback-to-build --module=/usr/local/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:755:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:822:5)
node-pre-gyp ERR! System Linux 4.4.11-ntc
node-pre-gyp ERR! command "node" "/usr/local/lib/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/local/lib/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.29
node-pre-gyp ERR! node-pre-gyp -v v0.6.28
node-pre-gyp ERR! not ok 
Failed to execute 'node-gyp build --fallback-to-build --module=/usr/local/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/serialport/build/Release' (1)
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! serialport@3.1.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@3.1.2 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.4.11-ntc
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "serialport" "--build-from-source"
npm ERR! cwd /home/chip/Desktop/LaserWeb3
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/chip/Desktop/LaserWeb3/npm-debug.log
npm ERR! not ok code 0
```

reconbot, 2016-06-13:
(Any reason you're using node 10?)

I'm mobile at the moment, but could you attach the npm-debug.log?

fivdi, 2016-06-13:
The compile error stating that `'REPLACE_INVALID_UTF8' is not a member of 'v8::String'` is occurring because the version of Node.js v0.10.29 that's installed on the system is broken.
```
../node_modules/nan/nan.h:324:47: error: 'REPLACE_INVALID_UTF8' is not a member of 'v8::String'
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
```
Node.js v0.10.29 on Raspbian/Debian for the Raspberry Pi and BeagleBone have the same issue. A solution and workaround for Raspbian is described at https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-Raspberry-Pi. The solution and workaround should also work on CHIP.

jeffersonbenson, 2016-06-14:
So I've been trying to work on it the bit that I can, but for some reason when I install node through apt-get the version I get is .10. Perhaps I need to update my PPA?
Here's a link to the rather long log file: [https://www.dropbox.com/s/0d3c9n30n0b92bu/node-log.rtf?dl=0](https://www.dropbox.com/s/0d3c9n30n0b92bu/node-log.rtf?dl=0) 

Also @fivdi I was able to integrate your workaround into the v8.h file, but this seems to have not have had an effect on the compiling process. 

Progress? Yes. Solved? Not yet.

fivdi, 2016-06-14:
All I can see in the log file is that there were errors running what appears to have been the command `sudo npm install -g --unsafe-perm serialport`. There is no mention of what those errors were.

If it's not possible to install serialport with the broken Node.js v0.10.29 + workaround, I'd suggest installing a non broken version of Node.js. NodeSource provide binary distributions of Node.js.  Installation instructions can be found [here](https://github.com/nodesource/distributions)

reconbot, 2016-06-14:
I think we need to have these on the ubuntu/debian install section, the ppa we mention is out of date

reconbot, 2016-06-14:
https://github.com/EmergingTechnologyAdvisors/node-serialport/pull/853 should make the clearer

jeffersonbenson, 2016-06-15:
While you are correct @reconbot there are still several errors being thrown. I've included an updated log on the off chance that something has changed. Node is now at 6.X and that seems to have helped. Any insight would be greatly appreciated.
[https://www.dropbox.com/s/ojmmcww8qyv1z5b/newnodelog.rtf?dl=0](https://www.dropbox.com/s/ojmmcww8qyv1z5b/newnodelog.rtf?dl=0)

reconbot, 2016-06-15:
the log (which you can upload here btw by dragging it into your comment) still says 

```
2276 error node -v v0.10.29 
2277 error npm -v 1.4.21
```

jeffersonbenson, 2016-06-15:
So I can certainly confirm I have Node v6.2.1 There seems to be some sort of permissions error involving node-pre-gyp
```
/usr/bin/serialportlist -> /usr/lib/node_modules/serialport/bin/serialport-list.js
/usr/bin/serialportterm -> /usr/lib/node_modules/serialport/bin/serialport-terminal.js

> serialport@3.1.2 install /usr/lib/node_modules/serialport
> node-pre-gyp install --fallback-to-build

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.2.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/serialport/.node-gyp"
make: Entering directory '/usr/lib/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/6.2.1/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/lib/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Linux 4.4.11-ntc
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/lib/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/usr/lib/node_modules/serialport/build/Release"
gyp ERR! cwd /usr/lib/node_modules/serialport
gyp ERR! node -v v6.2.1
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:852:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
node-pre-gyp ERR! System Linux 4.4.11-ntc
node-pre-gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/serialport
node-pre-gyp ERR! node -v v6.2.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.28
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/serialport/build/Release' (1)
npm ERR! Linux 4.4.11-ntc
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "--build-from-source" "serialport"
npm ERR! node v6.2.1
npm ERR! npm  v3.9.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@3.1.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@3.1.2 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/chip/Desktop/npm-debug.log
```
Wish I could just upload this stuff, but I'm ssh'd into it for now

reconbot, 2016-06-15:
There's always resin.io for a docker based approach. Do you still have the
old nodejs installed by chance? `which nodejs` return anything?

---
Francis Gulotta
wizard@roborooter.com

fivdi, 2016-06-15:
Try `npm install serialport` or `sudo npm install serialport -g --unsafe-perm`

jeffersonbenson, 2016-06-15:
@reconbot ```which nodejs``` returns /usr/bin/nodejs

@fivdi I believe this has done it! Well...at least globally. NPM install seems to be returning issues in the directory of the project I'm working on, but I'll double check with that particular project. 
Thanks everyone!!!

closed 2016-06-15 by jeffersonbenson

fivdi, 2016-06-15:
The permissions, owners, and groups for the files and subdirectories in the project directory are likely to be messed up now due to attempts to install with sudo. The owner of some of the files and subdirectories is probably now root. Starting with a new project directory and installing locally without sudo is likely to solve these issues.

jeffersonbenson, 2016-06-15:
Seems to be good now. Had to specify a version, but thankfully npm is quite user-friendly compared to other software I've used. Looks all good now. Finally I can get to testing.

level451, 2016-12-07:
I had to install this package before I could get it to compile:
sudo apt-get install build-essential