serialport/node-serialport #854: TypeError: SerialPort is not a function

farell, 2016-06-15:
Operating System and Hardware:Windows 7 & intel i7-4700MQ HM86 Chipset
NodeJS Version: 5.2
Serialport version:3.1.2
Python Version (if you're having trouble building serialport):
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
this two lines of code does not work:

var SerialPort = require('serialport');
var port = new SerialPort('COM16');

I got : 

var port = new SerialPort('COM16');
           ^
TypeError: SerialPort is not a function
    at Object.<anonymous> (C:\Users\bunny\Desktop\AcquireData\serialport.js:20:1
2)
    at Module._compile (module.js:399:26)
    at Object.Module._extensions..js (module.js:406:10)
    at Module.load (module.js:345:32)
    at Function.Module._load (module.js:302:12)
    at Function.Module.runMain (module.js:431:10)
    at startup (node.js:141:18)
    at node.js:977:3

Buuuuuuut, I have no problem running the function:

SerialPort.list(function (err, ports) {
  ports.forEach(function(port) {
    console.log(port.comName);
    console.log(port.pnpId);
    console.log(port.manufacturer);
  });
});

it successfully list my serial port,somebody help!
### Actual behavior

### Steps and/or code to reproduce the issue

reconbot, 2016-06-15:
The docs you're looking at are for the master branch, you want to the docs for 3.1.2
https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/3.1.2/README.md

reconbot, 2016-06-15:
I'm going to close this but if you'd like to continue with it feel free to reopen.

closed 2016-06-15 by reconbot

farell, 2016-06-15:
It works ,thanks buddy！

krishnatgkk, 2016-06-24:
Hi

I am having the same exact issue. can you tell me how did you solve this.
the list function works but when i said new SerialPort it gives me 
TypeError: SerialPort is not a function

please help 

Thanks

reconbot, 2016-06-24:
We changed how you get the function, what version are you using? It sounds like your using 3.x and looking at the docs for 4x.

krishnatgkk, 2016-06-24:
Thank you. not sure how i missed it. works perfect.