serialport/node-serialport #856: Error on opening Port on MacOSX with Electron

po0kie, 2016-06-15:
Operating System and Hardware: Mini Mac

Versions:
ares:"1.10.1-DEV"
atom-shell:"1.2.2"
chrome:"51.0.2704.84"
electron:"1.2.2"
http_parser:"2.7.0"
modules:"48"
node:"6.1.0"
openssl:"1.0.2h"
uv:"1.9.0"
v8:"5.1.281.59"
zlib:"1.2.8"

Serialport version: 4.0.0@beta3

the list method works fine.
the problem is opening the Port.

It throws error:
Port is opening

It only happend on the Mac, on windows its working fine.
Some idea where to look for?

Best regards
Marc

reconbot, 2016-06-15:
Can you share some sample.code?

po0kie, 2016-06-15:
Sure,

i have copied the solutions folder and cleaned out all the files except the node_modules
and main.js + index.html to reproduce the error:

The zip is so big as I would leave the solution the much as similiar to what I have on my pc.
Here the Archive to download if you want:
[https://www.dropbox.com/s/5bdhqinb6krqnn9/serialport-test.zip?dl=0](url)

basically its the following:

main.js
`
const electron = require('electron')
// Module to control application life.
const app = electron.app
// Module to create native browser window.
const BrowserWindow = electron.BrowserWindow
const {ipcMain} = require('electron');

ipcMain.on('list-serial-com-ports', (event, arg) => {
  console.log("requesting serial port to open...");
  var serialport = require('serialport');
 
            
  //this now is also broken and returns a really ugly Error
       
  //serialport.list(function (err, ports) {
    // ports.forEach(function (port) {
     // console.log(port.comName);
     //console.log(port.pnpId);
     //console.log(port.manufacturer);
    //});
  //});
  
   var  delimiter = "\r";
            var serialOptions = {
                baudrate: 57600,
                dataBits: 8,
                parity: 'none',
                autoOpen: false,
                stopBits: 1,
                flowControl: true,
                parser: serialport.parsers.raw
            };
            
            var port = new serialport('/dev/cu.usbserial-KKUyBAng', serialOptions);
              port.on('error', function(err) {
                  console.log('port.on(error) Error: ', err.message);
              });

            port.open(function (err) {
               if (err) {
                    console.log(typeof err);
                   return console.log('Error opening port: ', JSON.stringify(err));
               }
            }); 
});

`

and on index.html is the following to call the function on main.js
`<body >
Serial Port Example

<button type="button" onclick="openSerialPort()"> open Serial Port </button>
</body>

<script>
    const electron = require('electron');
    const remote = electron.remote;

    function openSerialPort() { 
          var ipc = electron.ipcRenderer;
          ipc.send('list-serial-com-ports', '');    
    }
</script>`

Best regards and thanks!

reconbot, 2016-06-16:
You say `SerialPort.list` is now also broken? what error does that give you?

reconbot, 2016-06-16:
Oh, I've we've been seeing issues with electron 1.2.0 and higher. I'd see about reverting back to a 1.1.x and see if that works.

po0kie, 2016-06-16:
Thanks! Downgrading to 1.1.3 its working.

closed 2016-06-17 by reconbot