serialport/node-serialport #859: Testing serialport@4.0.0 for release

reconbot, 2016-06-21:
This is a big release so I'd like to make a call for testers. ☑

Please try to test on the following platforms and report back here.

| Platform / Arch | Node v0.10.x | Node v0.12.x | Node v4.x | Node v5.x | Node v6.x |
|       ---       | --- | --- | --- | --- | --- |
| Linux / ia32    |  [ ]  |  ☑  |  ☑  |  [ ]  |  [ ]  |
| Linux / x64     |  [ ]  |  [ ]  |  [ ]  |  [ ]  |  [ ]  |
| Windows¹ / x86  |  [ ]  |  [ ]  |  [ ]  |  [ ]  |  [ ]  |
| Windows¹ / x64  |  [ ]  |  [ ]  |  [ ]  |  [ ]  |  [ ]  |
| OSX² / x64      |  [ ]  |  [ ]  |  ☑  |  [ ]  |  [ ]  |

reconbot, 2016-06-22:
I got the osx testing done

vicatcu, 2016-06-23:
@reconbot I have an Ubuntu 14.04 x64 machine, and Windows 10 x64 that I could try this with. I've been using 4.0.0-rc1 happily for about a week now on a Raspberry Pi 3 as well.

fivdi, 2016-06-23:
The above list is a thundering disgrace. It doesn't include ARM.

fivdi, 2016-06-23:
It only includes Intel.

reconbot, 2016-06-23:
I opened an issue about that, we haven't officially supported arm yet, I
think we should test for it too. As thundering as possible

reconbot, 2016-06-24:
@vicatcu Raspberry Pi 3 is Arm v8, what version of node?
@fivdi I've thunderously added the other known platforms to the support table and the testing matrix. We're never going to get full coverage, but it's here for the taking.

fivdi, 2016-06-24:
The following tests functioned as expected:

| Platform / Arch | Node v0.10.x | Node v0.12.x | Node v4.x | Node v5.x | Node v6.x |
|       ---       | --- | --- | --- | --- | --- |
| Linux / ia32    |  ☑  |  ☐  |  ☐  |  ☑  |  ☑  |
| Linux / ARM v7 |  ☑  |  ☐  |  ☑  |  ☑  |  ☑  |

Apologies for the ridiculous comment above.

reconbot, 2016-06-25:
Thanks, and no worries, I've updated the table.

fivdi, 2016-06-25:
The following tests also functioned as expected:

| Platform / Arch | Node v0.10.x | Node v0.12.x | Node v4.x | Node v5.x | Node v6.x |
|       ---       | --- | --- | --- | --- | --- |
| Linux / ARM v6 |  ☑  |  ☐  |  ☑  |  ☑  |  ☑  |
| Linux / ARM v7 |  ☐  |  ☑  |  ☐  |  ☐  |  ☐  |

There are five warnings when installing with Node.js v0.10.29 but everything functions as expected at runtime. The warnings are all related to packages that `node-pre-gyp` indirectly depends on. Four of the five warnings could be avoided by upgrading to Node.js >= v0.10.40. To prevent the fifth warning related to `deep-extend`, Node.js >= v0.12.0 is needed. Although everything still functions correctly with Node.js v0.10.x, support for Node.js v0.10.x is starting to get fragile.

```
pi@raspberrypi:~/serialport $ npm install EmergingTechnologyAdvisors/node-serialport
npm WARN engine deep-extend@0.4.1: wanted: {"node":">=0.12.0","iojs":">=1.0.0"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine hawk@3.1.3: wanted: {"node":">=0.10.32"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine hoek@2.16.3: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine cryptiles@2.0.5: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine boom@2.10.1: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
-
> serialport@4.0.0-rc1 install /home/pi/serialport/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.0-rc1/serialport-v4.0.0-rc1-node-v11-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.0-rc1 and node@0.10.29 (node-v11 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/serialport/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  SOLINK_MODULE(target) Release/obj.target/serialport.node: Finished
  COPY Release/serialport.node
make: Leaving directory '/home/pi/serialport/node_modules/serialport/build'
serialport@4.0.0-rc1 node_modules/serialport
├── bindings@1.2.1
├── commander@2.9.0 (graceful-readlink@1.0.1)
├── debug@2.2.0 (ms@0.7.1)
├── es6-promise@3.2.1
├── object.assign@4.0.3 (function-bind@1.1.0, object-keys@1.0.9, define-properties@1.1.2)
├── nan@2.3.5
└── node-pre-gyp@0.6.28 (nopt@3.0.6, semver@5.1.1, rc@1.1.6, mkdirp@0.5.1, tar-pack@3.1.3, rimraf@2.5.2, npmlog@2.0.4, tar@2.2.1, request@2.72.0)
pi@raspberrypi:~/serialport $ 
```

reconbot, 2016-06-25:
Yeah I agree. Thankfully they're not yet causing a problem. A few weeks ago I put a few hours into tracking down and fixing or replacing those packages but not all of them are actively maintained so I wasn't able to get very far. Since it still works ¯\_(ツ)_/¯

jacobq, 2016-06-26:
I'm not sure if I'm doing something wrong, but sometime about 12 hours ago the windows builds of my electron app started blowing up with:
[image]

reconbot, 2016-06-26:
With the beta version?

jacobq, 2016-06-26:
Yes, but it looks like there's something else wrong -- does same with v3.1.2, so it may be an electron thing....just makes me strongly dislike Windows so much *sigh*

reconbot, 2016-07-02:
I did some windows testing (win7 64bit), and have reports with electron working with 4x on windows. Also windows is one of the platforms where we can run the light integration tests in the CI. I think we're good enough to go. I'd love ideas on making this process easier.

closed 2016-07-02 by reconbot

zewish, 2016-07-04:
Hey @reconbot, I updated my 3.x.x code to work with 4.0.0.
I installed 4.0.0 from NPM and I'm not quite sure what's happening but the native promise is still replaced.

Could you please check that you published the right version? Thanks! :)

reconbot, 2016-07-04:
That patch landed a bit after the release, I'll have it out in a few days.

zewish, 2016-07-05:
Cool, thanks for the info.

reconbot, 2016-07-10:
@zewish landed in `4.0.1`