serialport/node-serialport #861: Unable to open serial port from NW.js desktop app engine

1i7, 2016-06-25:
Operating System and Hardware: Ubuntu Linux 14.04 64 bit
NodeJS Version: 6.2.1/NW.js
Serialport version: 4.0.0-rc1, 3.1.2

Hello, Node-serialport fails to open serial port from inside application running above NW.js platform (Node.js desktop application engine http://nwjs.io ). 

```
<!DOCTYPE html>
<html>
  <head>
    <title>Rraptor</title>
    <meta charset="UTF-8"/>
    
    <script type="text/javascript">
      var port;
    
    
      function writeLog(msg) {
        var logElement = document.getElementById("serial_read_data");
        logElement.innerHTML += msg + "<br>";
        logElement.scrollTop = logElement.scrollHeight;
      }
    
      /**
       * Обновить список последовательных портов
       */
      function updateSerialPortList() {
        var SerialPort = require('serialport');
      
	    var combo = document.getElementById("serial_ports");
	    combo.options.length = 0;
	
        SerialPort.list(function (err, ports) {
          ports.forEach(function(port) {
            console.log(port.comName);
            console.log(port.pnpId);
            console.log(port.manufacturer);
            
            var option = document.createElement("option");
	        option.text = port.comName;
	        option.value = port.comName;
	        combo.add(option, null);
          });
        });
      }
      
      /**
       * Подключаемся к устройству на последовательном порте
       */
      function connectDevice(portName) {
        var SerialPort = require('serialport');

        port = new SerialPort(portName, {
          // скорость
          baudRate: 9600,
          // получать данные по одной строке
          parser: SerialPort.parsers.readline('\n'),
          // не открывать порт сразу здесь
          autoOpen: false,
          lock: true
        });

        // 
        // события
        // 

        // порт открылся
        port.on('open', function () {
          writeLog('Opened port: ' + portName);
  
          // пишем данные отсюда, результат получаем дальше 
          // в колбэке на событие data
          port.write('help');
        });

        // пришли данные
        port.on('data', function (data) {

          document.getElementById('serial_read_data').innerHTML += data;
          writeLog(data);
        });

        // 
        // Действия
        // 

        // открываем порт
        port.open(function(err) {
          if (err) {
            writeLog('Error opening serial connection ' + portName + ': ' + err);
          }
        });      
      }
      
      function sendCmd(cmd) {
        // пишем данные отсюда, результат получаем в колбэке на событие data
          port.write(cmd);
      }
    </script>
  </head>
  <body>
    <h1>Рраптор: панель управления</h1>
    <div>
      Устройство: <select name="serial_ports" id="serial_ports"></select> 
      <a href="javascript:updateSerialPortList()">обновить</a>
      
      <a href="javascript:connectDevice(document.getElementById('serial_ports').value)">Подключиться</a>
    </div>
    
    <a href="javascript:sendCmd('help')">Попросить о помощи</a>
    
    <div id="serial_read_data">
    </div>
  </body>
</html>
```
copy above code to index.html, then run application from same dir:

`
~/apps/nwjs-sdk-v0.16.0-beta2-linux-x64/nw .
`
The app would show window with dropdown list, click on "Update serial port list" link to fill the list of available serial ports (this would work), then select "/dev/ttyUSB0" (or other valid device connected to serial port), then click on "Connect device" link to connect to serial port.

On the first connection attempt it would return error event with "true" as _err_ variable value (and it has boolean datatype), the second and all further connection attempt would return error event with _err_ with "unable to lock device" message.

Same code runs just fine when run from command line with regular nodeJS: I am able to communicate device (Arduino), send commands, receive result etc. Getting list of serial ports also works just fine with NW.js.

Tried both with NW.js stable v0.15.4 and with latest v0.16.0-beta2 (sdk download option allows to use built-in debugger when run app).
http://nwjs.io/downloads/

Serialport version: 4.0.0-rc1, 3.1.2 - tried both, both fail

reconbot, 2016-06-25:
You know, I'm not sure how they fixed it but #838 had the same problem with electron. I'd love to figure out what's going on and document it.

1i7, 2016-06-27:
Reported same issue to NW.js project
https://github.com/nwjs/nw.js/issues/5035

switched to Electron 1.1.3 for now which works (latest Electron 1.2.5 does not work with same problem). Looks like the problem was introduced with Electron 1.1.3->1.20 version switch. In NW,js, probably, mostly at the same time as they should share same codebase.

Mike-Dax, 2016-07-13:
I was able to get serialport 4.0.0 to work with Electron 1.2.6 by installing electron-rebuild and making a small change, editing: `node_modules/electron-rebuild/lib/main.js`

I Replaced `args.push('--runtime=electron', '--target=' + nodeVersion, '--arch=' + (arch || process.arch));`

with `args.push('--runtime=electron', '--target=' + nodeVersion, '--arch=' + (arch || process.arch), '--build-from-source');`

(Alternatively you can edit the source of main.js at the same point, I just had some version conflicts with babel so I edited the compiled version.)

Jump back into the root directory and run

`./node_modules/.bin/electron-rebuild`

This rebuilds serialport against electron's version.

closed 2016-10-05 by reconbot