serialport/node-serialport #863: Make windows building easier by talking about windows-build-tools

reconbot, 2016-06-29:
https://www.npmjs.com/package/windows-build-tools <3 this is a great amount of work by @felixrieseberg. Updating our windows section of our readme would be of great help.

giseburt, 2016-07-21:
AFAICT, node-serialport doesn't compile with --msvs_version=2015, which is what that installs. Am I missing something?

reconbot, 2016-07-21:
I probably am, I haven't had time to dig into it. Why don't we install with
it?

giseburt, 2016-07-21:
This appears to be the source of the problem I'm seeing: https://github.com/nodejs/node-gyp/issues/972

Good news: `npm -g install npm@latest` fixed it. This is then a transient error.

reconbot, 2017-01-13:
We delegated to node-gyp's instructions for windows with some pointers.

closed 2017-01-13 by reconbot