serialport/node-serialport #866: Sending BRK signal

lubino, 2016-07-09:
Operating System and Hardware:
NodeJS Version: v5.11.0
npm version: 3.8.6
Serialport version: 4.0.0.0
Python Version (if you're having trouble building serialport): 2.7.9
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
There should exist some "sumbol for sending BRK"
### Actual behavior
There is some set brk options, but I don't know how to use it properly. Maybe it is not an issue, but please help me.

### Steps and/or code to reproduce the issue
I am trying to send "BRK FF 55 01 01 00 40 00 0a a0" to my Denon DRA-F109 but it ignores my command :(

reconbot, 2016-07-09:
I'd try this;

```js
port.set({brk:false}, function(){
  port.write([0xFF, 0x55, ...]);
});
```

reconbot, 2016-07-09:
After doing some reading, you probably have to set `brk` back to false before doing any writing.

lubino, 2016-07-10:
Wow, that was fast answer. Thank you for that, I tried that already, but it is not working, I think there is some communication gap witch makes it not understandable by receiver. Is that possible?

reconbot, 2016-07-11:
¯\_(ツ)_/¯

Can you share some code?

reconbot, 2016-07-11:
It would't hurt to share you hardware setup too.

lubino, 2016-07-12:
I am trying to connect my C.h.i.p ( https://www.nextthing.co/pages/chip ) over USB TTL rs232 dongle ( http://img.photo138.com/H5/CEG004200_4_zps474d3da9-40.jpg ) to Denon DRA-F109 HiFi audio amplifier. When I get it working I will publish everything here on github. My goal is to create an AirPlay,UPnP and media player device with HiFi standards represented by Denon amplifier. So I need to send controlling signals to the amplifier.
I found some ruby code for that https://github.com/kfigiela/denon-raspberry and I couldn't get it work on C.h.i.p. and I need to do that in node anyway, because of other WebUI staff I already implemented there.

I tried to send BRK ff 55 01 01 00 40 00 05 9b using code:

```javascript
port.on('data', function (data) {
  console.log('Received', data);
});
port.set({brk:true}, function() {
  var data = new Buffer([0x55, 0x01, 0x01, 0x00, 0x40, 0x00, 0x05, 0x9b]); 
  console.log("Sending", data);
  port.write(data);
  port.set({brk:false}, function(){
  });
});
```
The receiver echoes everything it gets, so I can see:
```
Sending Buffer << ff 55 01 01 00 40 00 05 9b >>
Received Buffer << 00 ff 55 01 01 00 40 00 05 9b >>
```
So I think that there is no HW problem on wire, because I am able to receive the echo. BRK signal is represented by 00 on UNIX systems. It looks like I have the same problem other had when they were sending 00 instead of BRK, so the receiver echoes it back, but doesn't react on it (doesn't change the volume or anything else).

I also tried to set BRK to false before sending data, but with the seme non working result :(
```javascript
port.set({brk:true}, function() {
  port.set({brk:false}, function(){
    var data = new Buffer([0x55, 0x01, 0x01, 0x00, 0x40, 0x00, 0x05, 0x9b]); 
    console.log("Sending", data);
    port.write(data);
  });
});
```

Here is also a very useful description of that protocol, Kamil did a great work.
https://gist.github.com/kfigiela/e5555ebc47ce4dd8a5b7

reconbot, 2016-07-12:
According to the protocol you need to keep it in `break` for `16.25 ms`. I would guess that any time longer than that would be ok. I'd try a `setTimeout` of 16ms and if that doesn't work reliably try 17ms.

```js
var port = new SerialPort('/');
var data = new Buffer([0x55, 0x01, 0x01, 0x00, 0x40, 0x00, 0x05, 0x9b]);

function breakFor16ms(cb){
  port.set({brk:true}, function() {
    setTimeout(function(){
      port.set({brk:false}, cb);
    }, 16);
  });
}

breakFor16ms(function(){
  console.log("Sending", data);
  port.write(data);
});
```

jacobrosenthal, 2016-07-12:
Yes more should be fine (better even) 100 even

Do some checks on settimeout, its not particularly fine grained or
repeatable.

https://nodejs.org/api/timers.html#timers_settimeout_callback_delay_arg

"The callback will likely not be invoked in precisely delay milliseconds.
Node.js makes no guarantees about the exact timing of when callbacks will
fire, nor of their ordering. The callback will be called as close as
possible to the time specified."

reconbot, 2016-07-12:
It will usually be at least that much time and if the event loop isn't
super busy it will be within a few ms. Let me know how it works!

lubino, 2016-07-13:
Thank you Francis, thank you a lot. I have no idea why I didn't try to delay that signal before, whatever, it works like a charm. So thank you very very much.

Right now my code looks like it needs some refactoring, yes it works, but looks ugly. So I will create a nice npm package and I will publish it on github, npm and this support issue also.

lubino, 2016-07-16:
Sorry for disturbing you again. Can you please explain me more about setting BRK to false and true?

Because it works only once, then I need to restart NodeJS (closing and reopening port in JS doesn't work and creating a new instance of SerialPort after closing the previous one also doesn't work)

This is how I am sending the BRK signal:
```javascript
    //nextLog is console.log with time in ms which shows delay from lastly called console.log

    function sendBRK(cb) {
        console.log('going to send BRK');
        port.set({brk: true}, function () {
            nextLog('BRK sent, going to delay for 16ms');
            setTimeout(function () {
                nextLog('timeout , going to clear BRK');
                port.set({brk: false}, function () {
                    nextLog('BRK cleared, going to send data');
                    cb();//here is callBack which writes data
                });
            }, 16/*16.25*/);
        });
    }
    function onData(data) {
        if (data.length == 1 && data[0] == 0) {
            nextLog('received BRK');
        } else {
            nextLog('received data '+toString(data));
        }
    }

     sendBRK(function () {
            var data = toArray('ff 55 01 01 00 01 02 00 59');
            port.write(data, function (err) {
                if (!err) nextLog('data sent (' + data.length + ' bytes) ' + toString(data));
            });
     });
```
When I run that first time after NodeJS starts, the timing looks like this and it works with my receiver:
```
going to send BRK
17.45ms BRK sent, going to delay for 16ms
19.01ms received BRK
10.95ms timeout , going to clear BRK
5.7ms BRK cleared, going to send data
7.87ms data sent (9 bytes) ff 55 01 01 00 01 02 00 59
```
But when I call it second time and any other time, it looks like this, completely different signal timing and it doesn't work with my receiver, only solution I have found is to kill NodeJs and run it again (not a solution for my app):
```
going to send BRK
3.08ms received BRK
2.07ms BRK sent, going to delay for 16ms
17.51ms timeout , going to clear BRK
4.59ms BRK cleared, going to send data
3.13ms data sent (9 bytes) ff 55 01 01 00 01 02 00 59
```

So I can see, that when I call port.set({brk: true}) for first time it took about 17ms to execute, and any other time only 3ms. I am also calling port.set({brk: false}) so I don't understand where is the difference?

PS: It also stops working after any communication on the port, if the device sends me some data, after that my BRK signal is ignored by the device also when I call it first time and the timing looks identical like second call :(

Thanks for any help.

lubino, 2016-07-17:
After testing I found the answer. The first call to serial port do some initialization, and the receiver is waiting the data after about 16ms, so is important to count with that delay took by calling port.set({brk: false}). Now I am calculating how long should I wait in setTimeout and it woks :)

reconbot, 2016-07-17:
Glad to hear it!

closed 2016-07-22 by reconbot