serialport/node-serialport #867: Disabling dtr on open

lynchmaniac, 2016-07-10:
Operating System and Hardware: Windows 7
NodeJS Version: 5.4.0
Serialport version: 4.0.0
Python Version (if you're having trouble building serialport):3.4.4
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
Writte data on a LCD plug on an Arduino Uno

### Actual behavior
When i write into serial the arduino reset. It's because of dtr which reset teh arduino.
There are some material workaround like this http://playground.arduino.cc/Main/DisablingAutoResetOnSerialConnection
But for my test, it will be good to do this with serialport.
There are an option to do this
myPort.set({"dtr": false}, callback);

but it seems that there are no effect on the arduino.

### Steps and/or code to reproduce the issue
There is my code for the serial part
```
portName = "COM5";
var myPort = new serialport(portName, {
   baudRate: 9600,
   // look for return and newline at the end of each data packet:
   parser: serialport.parsers.readline("\n")
 });

myPort.on('open', showPortOpen);
myPort.on('data', sendSerialData);
myPort.on('close', showPortClose);
myPort.on('error', showError);

function showPortOpen() {
   	console.log('port open. Data rate: ' + myPort.options.baudRate);
   	myPort.set({dtr:false}, function(){
  		myPort.write("some data");
	});
}
```
So it's seem that the ``` myPort.set({dtr:false}, function(){ ``` have no effect. The arduino still reset after the write data.

But maybe i doing something wrong, or i misunderstood the option.

reconbot, 2016-07-11:
All my research points to this being enforced by linux's serialport drivers. On windows I've seen a few examples of how it might work, but this problem is going to be difficult to address.

I do recommend modifying your arduino or using it's TX and RX pins with a USB TTL cable ([sparkfun](https://www.sparkfun.com/products/9718)|[Adafruit](https://www.adafruit.com/products/70)) instead of USB serial if you don't want it to restart.

You mentioned on the other thread that you have python code that doesn't enable DTR on open. Can you share that?

lynchmaniac, 2016-07-11:
Thanks for the reply.
I think i'll make some hardware modification on my arduino to stop this reset.
This my python code :
```
# writeSerial.py
import time
import argparse
import string
import json
import serial
import json
import sys
from pprint import pprint

class MyListener():
    ser = serial.Serial()
    ser.port = 'COM5'
    ser.baudrate = 9600
    ser.timeout = 1
    ser.setDTR(False)
    ser.open()
    ser.write(sys.argv[1].encode())
    time.sleep(1)

```

reconbot, 2016-07-11:
Reading the source of pyserial I can find that on windows they disable/enable rts immediately after opening with `createFile` (regardless of when you set it). I bet it's done fast enough to make a difference to the arduino.

[windows pyserial](https://github.com/pyserial/pyserial/blob/master/serial/serialwin32.py#L206-L209)
```python
        if self._dsrdtr:
            comDCB.fDtrControl = win32.DTR_CONTROL_HANDSHAKE
        else:
            comDCB.fDtrControl = win32.DTR_CONTROL_ENABLE if self._dtr_state else win32.DTR_CONTROL_DISABLE
```

[Unix does too...](https://github.com/pyserial/pyserial/blob/master/serial/serialposix.py#L592-L597)
```python
    def _update_dtr_state(self):
        """Set terminal status line: Data Terminal Ready"""
        if self._dtr_state:
            fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
        else:
            fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_DTR_str)
```

lynchmaniac, 2016-08-01:
Hi,

Have tou got some news on this issue ? Is it possible to do that with serialport ?

reconbot, 2016-08-01:
Not any faster then we can currently do it.

closed 2016-08-22 by reconbot

duzc2, 2017-06-02:
hi @lynchmaniac 
Hi , I still get this problem on the last 4.0.7.
Have we fixed the problem ?

kealist, 2017-08-02:
I know I'm necroing this, but the documentation is quite bizarre for disabling dtr.   As @duzc2 mentioned (I figured it out by looking through cpp code), it can be disabled on opening (or creation) by setting `hupcl: false` as in:

```
var port = new SerialPort('COM5', {
    baudRate: 57600, 
    dataBits: 8, 
    hupcl: false, 
    stopBits: 1, 
    parity: "none"
},false);`
```

I would say this option should be listed among the `SerialPort~openOptions : Object` docs.

reconbot, 2017-08-04:
Can you file a bug for that? It should be fixed.

kealist, 2017-08-05:
Sorry, I started here, but did end up making the other issue and PR after more investigation.

reconbot, 2017-08-05:
Sorry was reviewing emails on my phone, wasn't sure who did what. =)