serialport/node-serialport #870: Increase in delay for closed connection causes port hopping (OSX)

noopkat, 2016-07-13:
Operating System and Hardware: OSX El Capitan 10.11.5
NodeJS Version: 4.2.1
Serialport version: 4.0.1
Python Version (if you're having trouble building serialport): n/a

### Expected behavior
1. Perform a software reset (not a physical pulling out of the cable) of connected USB serial device (example - Arduino Leonardo/Micro)
2. Serial device should promptly* disappear from `Serialport.list` of devices, or at least faster than 400ms on average.
3.  In almost all cases, device should return to same port when finished rebooting the USB emulator

*not a scientifically exact measure of time

### Actual behavior
1. Device takes up to 400 milliseconds to disappear from the list of exisiting devices
2. As a result, when the device finishes rebooting (generally faster than this), it hops to another port eg. from `/dev/cu.usbmodem97` to `/dev/cu.usbmodem98`
3. It seems that the device can exist on two different listed ports simultaneously for a short period of time, and only the latest port will allow a new connection to the device (which is the most correct port in this case, considering).
4. "Port hopping" is common on Windows, but I have never seen it happen on OSX before.

### Steps and/or code to reproduce the issue
1. Plug in an Arduino Micro, Leonardo, or Sparkfun Pro Micro
2. `npm install avrgirl-arduino@1.7.4`
3. `cd node/modules/avrgirl-arduino`
4. `node bin/cli.js list` to see which port the arduino is on
5. Run `node tests/demos/micro.js` (or the appropriate file for your arduino board)
6. When finished, run `node bin/cli.js list` again to see which port the arduino is on. It should have skipped at least one port number assignment, and most likely has bumped up two numbers (due to the device coming out of bootloader mode when done flashing).

### Additional musings

Essentially each time I flash one of these 32u4 devices with Serialport 4.0.1, I am incrementing that port number by 3, which kinda starts looking weird. I'm not sure if this feel wonky from others' perspectives, though?

Is a 400ms+ closing time deemed slow, or is this normal? It's great because it surfaced some less than robust implementations within avrgirl, but outside of this, closing time regressions are less than ideal.

I can run some proper benchmarks to see what the previous closing time was pre v4.x if you're interested in having more numbers for your consideration :grin:

reconbot, 2016-07-13:
Thank you. =)

I haven't touched any of the list or closing code in a long while, at least not significantly. (I did try to! but man it's hard to uncouple c++) So I'm not convinced something has changed. But you never know what interacts with what so a regression would be the first thing I check.

What your describing isn't closing a port as you never said the port was open. (Also rules out disconnect, which did have some changes.) It would be the registering and deregistering of the port with the OS. 

For some background.

With linux the kernel updates file entries in `/dev` and we poll the directory.  We don't use any of the libudev stuff yet.

With windows we query the registry. It has been reported to cache the entries for up to a few days (with windows 8 and 10). It should be noted that COM 1-3 always exist regardless of what they have attached.

We use the [IOServiceGetMatchingServices](https://developer.apple.com/library/mac/documentation/IOKit/Reference/IOKitLib_header_reference/#//apple_ref/c/func/IOServiceGetMatchingServices) function to get ports on osx. That doesn't help me figure this out at all. Never mind about that. This shouldn't have changed recently either.

I'd blame osx or the ftdi drivers, but lets test for that regression.

noopkat, 2016-07-13:
@reconbot ah thanks, this is really useful info.

I realise I need to clarify when the port is open:

1. Connect to the device at baud rate 1200
2. Device is reset while connected by cycling DTR settings
3. This triggers a disconnect, followed by a close event
4. Reconnect to device at different baud rate, once it shows up on new port
5. Flash device
6. Node process ends which closes the connection again.

noopkat, 2016-07-13:
I think I agree, from what you're suggesting, that perhaps this is not related to the connection closing timing at all. I'll report back once I have an isolated script to replicate more helpful output with, and some time comparisons 😁

reconbot, 2016-07-16:
Was there a regression?

noopkat, 2016-07-16:
this week got weird, so stay tuned for an update this weekend as I had to put this down for a couple of days. Thanks  😁

noopkat, 2016-07-18:
The convo I started here is misleading as to what this issue is now about 😄  I'm going to close this for now and reopen a fresh issue if I find a regression when I get to this. Thanks!

closed 2016-07-18 by noopkat