serialport/node-serialport #871: DEPRECATION: Please use 'require('serial')' instead of 'require('serialport').SerialPort'

wjrivera, 2016-07-13:
Operating System and Hardware:  Windows 7 64-bit
NodeJS Version: v4.4.7
NPM Version: v3.10.5
Serialport version:
Python Version : 3.4.3

Getting error below everytime I execute the code below:
Error:
DEPRECATION: Please use 'require('serial')' instead of 'require('serialport').SerialPort'
Code:
var serialport = require('serialport');
var SerialPort = serialport.SerialPort;
var portname = process.argv[2];

//list serial ports:
var myPort = new SerialPort(portname, {
  baudRate: 6900,
  options: false,
  parser: serialport.parsers.readline("\r\n")
});

myPort.on('open', function() {
  console.log('port is open');
});
myPort.on('close', function() {
  console.log('port is closed');
});
myPort.on('error', function() {
  console.log('something went wrong');
});
myPort.on('data', function(data) {
  console.log(data);
  myPort.write('x');
});

reconbot, 2016-07-13:
You're using serialport 4.x you no longer need to do `var SerialPort = serialport.SerialPort;`

reconbot, 2016-07-13:
I do see that `require('serial')` is wrong.

wjrivera, 2016-07-13:
Thank you so much, that fixed the issue.

closed 2016-07-13 by wjrivera