serialport/node-serialport #872: Can't install serialport...

drj-io, 2016-07-14:
Operating System and Hardware: Macbook Pro OSX Yosemite 
NodeJS Version: tested 4.2.2, 5.12.0, 6.3.0 / currently npm 3.8.6 (node 5.12.0)
Serialport version: Latest
Python Version (if you're having trouble building serialport): Python 2.7.10
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior

```
npm install serialport
```

### Actual behavior

```
npm install serialport
bash-3.2$ npm install serialport

> serialport@4.0.1 install /Users/david/dev/r/testst/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! install error
node-pre-gyp ERR! stack Error: Unsupported target version: 0.34.2
node-pre-gyp ERR! stack     at get_runtime_abi (/Users/david/dev/r/testst/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:156:23)
node-pre-gyp ERR! stack     at Object.module.exports.evaluate (/Users/david/dev/r/testst/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:277:19)
node-pre-gyp ERR! stack     at install (/Users/david/dev/r/testst/node_modules/serialport/node_modules/node-pre-gyp/lib/install.js:148:31)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as install] (/Users/david/dev/r/testst/node_modules/serialport/node_modules/node-pre-gyp/lib/node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run (/Users/david/dev/r/testst/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at Object.<anonymous> (/Users/david/dev/r/testst/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:131:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:413:34)
node-pre-gyp ERR! stack     at Object.Module._extensions..js (module.js:422:10)
node-pre-gyp ERR! stack     at Module.load (module.js:357:32)
node-pre-gyp ERR! stack     at Function.Module._load (module.js:314:12)
node-pre-gyp ERR! System Darwin 14.5.0
node-pre-gyp ERR! command "/usr/local/bin/node" "/Users/david/dev/r/testst/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /Users/david/dev/r/testst/node_modules/serialport
node-pre-gyp ERR! node -v v5.12.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok
Unsupported target version: 0.34.2
```

### Steps and/or code to reproduce the issue
Just trying to install

reconbot, 2016-07-16:
Are you doing anything with electron?

drj-io, 2016-07-18:
Yes, though, in this case I was just trying to install without electron.

Thanks for your help, any ideas welcome.

drj-io, 2016-07-19:
@reconbot I've tried just about everything and every version of node.  It's very strange.  I've been able to install in the past, without any problem.  But I can't install whatsoever since I set up electron. 

```bash
~/d/sptest> node -v
v6.3.0
~/dev> mkdir sptest
~/dev> cd sptest
~/d/sptest> ls
~/d/sptest> npm init
(node:13593) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13593) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13593) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13593) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
This utility will walk you through creating a package.json file.
It only covers the most common items, and tries to guess sensible defaults.

See `npm help json` for definitive documentation on these fields
and exactly what they do.

Use `npm install <pkg> --save` afterwards to install a package and
save it as a dependency in the package.json file.

Press ^C at any time to quit.
name: (@madwire/sptest) sptest
version: (1.0.0)
description:
entry point: (index.js)
test command:
git repository:
keywords:
license: (ISC)
About to write to /Users/david/dev/sptest/package.json:

{
  "name": "sptest",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "author": "David Jones <null@rackfx.com> (rackfx.com)",
  "license": "ISC"
}

Is this ok? (yes)
~/d/sptest>
~/d/sptest> npm install serialport --save
(node:13598) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13598) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13598) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13598) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13598) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13598) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13598) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13598) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:13598) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.

> serialport@4.0.1 install /Users/david/dev/sptest/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! install error
node-pre-gyp ERR! stack Error: Unsupported target version: 0.34.2
node-pre-gyp ERR! stack     at get_runtime_abi (/Users/david/dev/sptest/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:156:23)
node-pre-gyp ERR! stack     at Object.module.exports.evaluate (/Users/david/dev/sptest/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:277:19)
node-pre-gyp ERR! stack     at install (/Users/david/dev/sptest/node_modules/serialport/node_modules/node-pre-gyp/lib/install.js:148:31)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as install] (/Users/david/dev/sptest/node_modules/serialport/node_modules/node-pre-gyp/lib/node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run (/Users/david/dev/sptest/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at Object.<anonymous> (/Users/david/dev/sptest/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:131:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:541:32)
[... 90 lines cut ...]
npm ERR! Darwin 14.5.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport" "--save"
npm ERR! node v4.4.7
npm ERR! npm  v2.15.8
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/david/dev/sptest/npm-debug.log
~/d/sptest>
```

drj-io, 2016-07-22:
Tried uninstalling and reinstalling Node.  

still get 
```
Unsupported target version: 0.34.2
```

That's fine, but what does that mean?  Version of what?

I cloned the github repo and ran npm install and get the same error.  I have not given up yet.

drj-io, 2016-07-22:
I've tried a few things.  I tried on another laptop and it works fine on there... 

I'm now getting the same error with serialport and sqlite3 on this repo, so I'm certain it's localized. Obviously node-pre-gyp can't build the module on my machine for some reason.

Good night for now...

drj-io, 2016-07-22:
Updated to El Capitan on this machine and updated X Code too.  Still get the same error

drj-io, 2016-07-25:
I was able to build j5/serialport on a separate, virgin OSX Yosemite machine for electron and import the module into my primary dev box.  

Not sure how to resolve the target issues for the specific machine in question, but this worked for me. 

@reconbot, feel free to let me know if you would like more information to debug the issue.  Closing this for now.   I may revisit in the future, too.

closed 2016-07-25 by drj-io