serialport/node-serialport #873: .on('data', function(data) not returning values

wjrivera, 2016-07-14:
Operating System and Hardware: Windows 7 64-bit
NodeJS Version: v4.4.7
NPM Version: v3.10.5
Serialport version:
Python Version : 3.4.3

I'm working on a board that when I send a buffer array, it replies back with a message.  I'm using Device Monitoring Studio on my COM port to monitor traffic.  I see my message going in using the code below.

myPort.on('open', function() {
  console.log('port is open\n');
  myPort.write(buffer, function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('Diagnostics info for software & configuration message sent.\n');
  });
});

myPort.on('data', function(data) {
  console.log('entered on data\n');
  console.log('Data; ' + data);
  myPort.write('x');
});

Device Monitoring Studio tells me that the node.js program it still reading (Ordinal: 14449 in picture) from COM port and that my board is replying with a HEX number (Ordinal: 14451 and 14453 in picture).  Is there something I'm doing wrong?  Does the 'data' take HEX values?
Thank you

[image]

reconbot, 2016-07-14:
The constant reading is normal behavior to check if new data is there.

Serialport data is any binary data and can be human readable or not, so the
monitor is using hex to be safe. We use buffers in node serialport and
unless you use a parser to change the behavior, you'll get `Buffer`s on the
data event. You can call `toString('utf8')` to turn it to a string if you
like.

wjrivera, 2016-07-14:
Thank you so much!!!  I used raw and converted my data.toString('hex'), solved the problem.

reconbot, 2016-07-14:
Yay!

closed 2016-07-14 by reconbot