serialport/node-serialport #874: Unable to Install serialport on Raspberry PI 3

jonhardman, 2016-07-14:
Operating System and Hardware: Raspberry PI 3 - Raspian
NodeJS Version: v6.3.0, NPM 3.10.3
Serialport version: Any 
Python Version (if you're having trouble building serialport): 2.7.9
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
Trying to Install Serial Port with the following command:
sudo npm install serialport

### Actual behavior
The install fails, with the following errors (debug log attached)

I have followed all of the instructions described here for a raspberry pi - with a fresh install of Raspbian: https://github.com/nebrius/raspi-io/wiki/Getting-a-Raspberry-Pi-ready-for-NodeBots
However this has not fixed the problem, I have even tried older versions of serialport and have the same issues.

Are there any further steps I need to complete for it to work on the raspberry pi 3 ? I am really keen to get this working :-)

Thank you in advance,

Output:

> serialport@4.0.1 install /home/pi/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-node-v48-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and node@6.3.0 (node-v48 ABI) (falling back to source compile with node-gyp) 
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.3.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/serialport/.node-gyp"
make: Entering directory '/home/pi/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/6.3.0/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/home/pi/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/node/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Linux 4.4.13-v7+
gyp ERR! command "/usr/local/node/bin/node" "/usr/local/node/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/pi/node_modules/serialport/build/Release"
gyp ERR! cwd /home/pi/node_modules/serialport
gyp ERR! node -v v6.3.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/node/bin/node /usr/local/node/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:852:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
node-pre-gyp ERR! System Linux 4.4.13-v7+
node-pre-gyp ERR! command "/usr/local/node/bin/node" "/home/pi/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node_modules/serialport
node-pre-gyp ERR! node -v v6.3.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/node/bin/node /usr/local/node/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/Release' (1)
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.
npm ERR! Linux 4.4.13-v7+
npm ERR! argv "/usr/local/node/bin/node" "/usr/local/bin/npm" "install" "serialport"
npm ERR! node v6.3.0
npm ERR! npm  v3.10.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
[npm-debug.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/364830/npm-debug.txt)

### Steps and/or code to reproduce the issue
sudo npm install serialport

reconbot, 2016-07-14:
@nebrius this looks like a permissions error but the user is root?

jonhardman, 2016-07-14:
Yes I am running this with the standard root 'pi ' user.

jonhardman, 2016-07-15:
I've tried all sorts and in the end the following command worked for me:

sudo npm install serialport -unsafe-perm

I was unsure what -unsafe-perm actually did and found an explantion here:
http://nodered.org/docs/hardware/raspberrypi.html

_Note: the reason for using the --unsafe-perm option is that when node-gyp tries to recompile any native libraries it tries to do so as a "nobody" user and often fails to get access to certain directories. This causes alarming warnings that look like errors... but sometimes are errors. Allowing node-gyp to run as root using this flag avoids this - or rather shows up any real errors instead._

Is this approach recommended?

reconbot, 2016-07-15:
Yeah it's fine as long as your trust serialport. =)

Kidding aside, it's just as safe as running our JS code as root.

jonhardman, 2016-07-16:
Thats great I'm happy with that :-) I think you should perhaps look to include this in your documentation for the raspberry pi install.

reconbot, 2016-07-16:
@nebrius what do you think about `unsafe-perm`?

closed 2016-07-16 by reconbot

razvan-theodor, 2018-02-05:
Hi. I wonder why
`sudo npm install serialport --unsafe-perm`
works on a Raspberry Pi and global install:
`sudo npm install serialport --unsafe-perm -g`
doesn't.