serialport/node-serialport #875: Distribute serial mocking classes with package

qdot, 2016-07-18:
Right now I'm mocking serialport for my tests via pulling the git repo instead of the npm package, and using the darwin-hardware emulator, as the library tests themselves use. It'd be nice to have the mocks distributed with the npm package instead of having to pull the git repo. There's other mocking libraries out there (virtual-serialport, mock-serialport), but none of them seem to be up to date with the current API, or to cover all of it as expected.

reconbot, 2016-07-22:
I'm 100% for this

reconbot, 2016-07-23:
For the record you can `require('serialport/test/mocks/darwin-hardware')` but that's hardly a stable API.

qdot, 2016-07-23:
Oh, yeah, that's what I'm doing right now. Having it in npm from now on would be nice so that the mock lines up with the package version though, just in case there's hard version requirements and API changes (for instance, virtual-serialport still uses the openImmediately parameter instead of autoOpen option, and someone on there was complaining they still use it when I changed that :) ).

Anyways, so I try to integrate/PR this, or you got it?

reconbot, 2016-07-23:
Reading isn't isolated to the binding yet but other than that we could
support a pluggable back end with out too much trouble.

Moving read to bindings.js would be step 1. Go for it!

reconbot, 2016-07-23:
I'd support just exposing the mock binding as is but it's sandbox stuff we
really don't want in prod.

reconbot, 2016-07-28:
I did a spike on cleaning up the binding layer enough to provide a mocking backend. It was not easy. I think this will have to come with the streams3 rewrite.

reconbot, 2016-08-22:
Which is currently on #906

reconbot, 2016-11-05:
[`MockBinding`](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/lib/bindings-mock.js) is now part of the main package =)

```js
var SerialPort = require('serialport/lib/serialport')
SerialPort.Binding = require('serialport/lib/bindings-mock');

var port = new SerialPort('/pretend/port');
port.binding.emitData(new Buffer("hi!"));
port.on('data', console.log);
```

This will be part of the 5x release and should make it into `5.0.0-beta3`

closed 2016-11-05 by reconbot