serialport/node-serialport #876: can't catch "can't connect"

gbaumgart, 2016-07-18:
Hi,
is there a way to catch "can't connect to /dev/...." error ? I put the routine in a try/catch clause but my whole app is still crashing on this which it shouldn't of course.

I am using 2.1.0. 

thanks a lot.

reconbot, 2016-07-18:
Upgrading to 3x will let you use the `on('error')` event more reliably with
minimal API changes. It was sometimes broken in 2x (see our upgrade guide
linked in the readme). Opening happens asynchronously so try/catch can't
catch the open event. Upgrading to 4x will give you synchronous errors
during construction, but even then open is still an asynchronous event.

gbaumgart, 2016-07-18:
thanks a lot, did update to 4.0.1 and now it works as expected.

closed 2016-07-18 by gbaumgart