serialport/node-serialport #877: It seems that on Windows OS vendorId always set to undefined

fduman, 2016-07-20:
Operating System and Hardware: Windows 10
NodeJS Version: 5.10.0
Serialport version: 2.1.2

I use serialport's list method and I cannot get vendorId information from that function. I had to use the pnpId property.

It seems that serialport_win.cpp has no code to get the vendor id.

I can get the vendor id information on OS X and this is an inconsistent behavior.

reconbot, 2016-07-22:
This might be a bug. 

 - We totally don't get them https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/src/serialport_win.cpp#L493-L504
 - I have no idea if we can get them

We're using a library called http://github.com/todbot/usbSearch/ to find ports and get info. This area is ripe for cleaning them up.

Zensey, 2016-11-20:
Also there is no productId and locationId

closed 2016-12-21 by reconbot