serialport/node-serialport #879: Serialport tests fail on Windows

joaocgreis, 2016-07-22:
Operating System and Hardware: Windows 10, Arduino Uno
NodeJS Version: all (https://ci.nodejs.org/view/All/job/serialport-daily-master/79/)
Serialport version: master

### Expected behavior

Tests should pass.

### Actual behavior

```
  76 passing (7s)
  4 pending
  1 failing

  1) SerialPortBinding #close errors when providing a bad file descriptor:
     Uncaught AssertionError: expected null to be an instance of Error
    at Function.assert.instanceOf (D:\workspace\serialport-test-commit-windows\nodes\serialport-win10\node_modules\chai\lib\chai\interface\assert.js:800:35)
    at D:\workspace\serialport-test-commit-windows\nodes\serialport-win10\test\bindings.js:141:16
```
From: https://ci.nodejs.org/job/serialport-test-commit-windows/536/nodes=serialport-win10/consoleFull

### Steps and/or code to reproduce the issue

https://ci.nodejs.org/view/All/job/serialport-daily-master/ is set up to run the tests on a Windows machine everyday, for the master branch. The job is now stable, but the same test fails everyday for all node versions when running `npm test`.

cc @munyirik

reconbot, 2016-07-22:
Hi this is great,
Leaves me with a lot of questions however. We pass tests on windows on appveryor all the time. [Master is currently passing](https://ci.appveyor.com/project/j5js/node-serialport/branch/master). 

 - Is there a setup I can see for how this is being run? (things need to be changed including `npm install` to `npm install --build-from-source`
 - I notice the url has changed for the repo since it was setup
 - Can github hooks kickoff a build?
 - Is there notifications I can hook into to report back to github or at least a badge?

reconbot, 2016-07-22:
I found some env vars, what's on `COM6`?

munyirik, 2016-07-22:
COM6 has an Arduino Uno

reconbot, 2016-07-22:
Nice! Could you ensure this is loaded onto it?
https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/test/arduinoTest/arduinoEcho/arduinoEcho.ino

---
Francis Gulotta
wizard@roborooter.com

munyirik, 2016-07-23:
I will when I get back to the office next week. I'll let you know when I have.

joaocgreis, 2016-07-25:
> Is there a setup I can see for how this is being run?

The console output has all the commands that are run. It is divided in 4 stages:
- rebase if needed
- kill node processes that might have been left behind
- set up a temporary node and npm
- npm install and run the tests, currently boils down to this:
```
call npm install --build-from-source
call npm install colors
call npm install memwatch-next --msvs_version=2013
call npm test
call npm run stress
```

> (things need to be changed including `npm install` to `npm install --build-from-source`

Done, any more?

> I notice the url has changed for the repo since it was setup

Done.

> Can github hooks kickoff a build?
> Is there notifications I can hook into to report back to github or at least a badge?

Currently the build is started daily for master, other than that it can only be started by a node core collaborator. There is some discussion going on about that in https://github.com/nodejs/build/issues/448 .

> env vars

Build parameters are env vars in the job (e. g. https://ci.nodejs.org/view/All/job/serialport-test-commit-windows/553/parameters/ ).

reconbot, 2016-07-25:
I recently made some changes

```
npm install --build-from-source
npm test
```

Is all we should be running, the stress tests are unreliable. I'd love to include updating the uno on `com6` but we can save that for another time.

reconbot, 2016-07-25:
Wait... it's an existing repo?

I'd add the following to the preamble

```
rm -rf ./node_modules ./build
npm install --build-from-source
npm test
```

reconbot, 2016-07-25:
The changes I made will run all the integration tests that require an Arduino if you have `TEST_PORT`defined.

joaocgreis, 2016-07-25:
> Is all we should be running

Done, removed the others.

> it's an existing repo?

The workdir is clean and the repo cloned every time the job is run.

`TEST_PORT` is always defined, it's a job parameter.

There's a new error (timeout): https://ci.nodejs.org/view/All/job/serialport-test-commit-windows/560/nodes=serialport-win10/console

reconbot, 2016-07-25:
I bet the Arduino doesn't have the `ArduinoEcho` program loaded on it. That would cause the timeout.

@noopkat what would be the best way to script flashing our echo program onto an arduino at the beginning of every build? I'd love to automate that step so we can continue to improve that code without human intervention.

The not throwing an error on port close of a invalid FD...  I'll merge a change to that test that tires a different strategy. It's possible the behavior of `CloseHandle` changed slightly in win 10 when given a negative FD.

noopkat, 2016-07-25:
@reconbot a question for you WRT your query: It looks like `ArduinoEcho` is designed to be board agnostic. Which boards would we be likely supporting for this preloading step?

munyirik, 2016-07-25:
I just uploaded arduinoEcho.ino to the board.

reconbot, 2016-07-25:
IT's an Uno

---
Francis Gulotta
wizard@roborooter.com

noopkat, 2016-07-25:
@reconbot if we're always going to suggest this test is done with an uno, then I would recommend compiling the sketch for the Uno, and placing the hex file alongside the sketch location in this repository. 

As for the rest, is this test always a manual process? Ie. does someone sit with the uno and run the test, or is this also run remotely too, on a computer with an uno attached? I ask because for manual testing a quick "I'm going to erase your arduino and then run a test on it, type 'yes' to proceed" followed by an avrgirl flash call would be a nice addition for the stress tests. etc. Perhaps a flag to override the warning might work for any automated remote machine testing. 

This is mostly a blob of my thinking aloud, but let me know what you think.

reconbot, 2016-07-25:
This is an automated process, In specific this issue is a window 10 build server. I want to put in a pre test script that flashes the arduino.

The manual co-pilot stuff would benefit from this too. =)

reconbot, 2016-07-25:
So Avrgirl as a dep, and a comitted precompiled hex file.

noopkat, 2016-07-25:
@reconbot yep! That should do just fine. This is very meta mind you - avrgirl-arduino uses serialport as a dep 😉

reconbot, 2016-07-25:
But it uses a stable release I can live with that

reconbot, 2016-07-26:
Looks like we have passing builds!

@noopkat were bootstrapping! Like pypy or gcc building a new version of gcc which is then used to build itself again.

I'll see if I can work up a build script to flash the arduino.

joaocgreis, 2016-07-26:
All node versions green: https://ci.nodejs.org/job/serialport-daily-master/84/

reconbot, 2016-07-28:
@joaocgreis I think we can close this issue but there are a few steps to work out

 - how can I be notified of failures?
 - we should have a way to load firmware onto the arduino

joaocgreis, 2016-07-28:
I don't know of a way to enable notifications with Jenkins, when https://github.com/nodejs/build/issues/448 resolves we might have more room to try new things. For now the simplest way is to keep checking https://ci.nodejs.org/job/serialport-daily-master/ , but you could take a look at https://ci.nodejs.org/job/serialport-daily-master/api/ if you want to build a custom tool.

When you have a list of commands to run to load the firmware let me know, I can add them to the job.

If you want me to run it on some specific PR like I did with https://github.com/EmergingTechnologyAdvisors/node-serialport/pull/883 please mention me, I'll start the job when possible.

Thanks!

reconbot, 2016-08-06:
@joaocgreis I have a branch `arduino-pipeline` that has a new `npm run` command that flashes the arduino. If the command doesn't work on windows then we can change it up. `npm run arduino:load` it will load the compiled arduino sketch onto an uno on `TEST_PORT`.

Can you run  before testing as part of the build on the travis machine and see if it works?

joaocgreis, 2016-08-10:
Failed: https://ci.nodejs.org/view/All/job/serialport-test-commit-windows/661/nodes=serialport-win10/console

Passing a env var in npm scripts [seems to be a problem](https://github.com/npm/npm/issues/2800): 
```
d:\workspace\serialport-test-commit-windows\nodes\serialport-win10>call npm run arduino:load   || true

> serialport@4.0.1 arduino:load d:\workspace\serialport-test-commit-windows\nodes\serialport-win10
> node bin/require-test-port.js && avrgirl-arduino flash -f test/arduinoTest/arduinoEcho/arduinoEcho.ino.hex -a uno -p $TEST_PORT -v

[Error: Opening $TEST_PORT: File not found]
```

joaocgreis, 2016-08-10:
I expanded the commands in the test job: https://ci.nodejs.org/view/All/job/serialport-test-commit-windows/663/nodes=serialport-win10/console

It looks like that the env var is the only thing that needs fixing. After that, it's back to green.

reconbot, 2016-08-10:
Cross platform shell scripting...

Time to research the available solutions.

munyirik, 2016-09-23:
FYI. I'm updating the machine so it's offline now (upgrading to Win 10 [version 1607](https://technet.microsoft.com/en-us/windows/release-info)). I'll update when it's back online.

munyirik, 2016-09-23:
It's back online.

munyirik, 2016-12-10:
The machine will be offline until 12/22 (I'm out of the office until then and the machine is getting moved during that time).

munyirik, 2016-12-27:
And it's back online.

reconbot, 2017-07-06:
Hey, since we have no feedback mechanism with this build server, I'm ok with shutting it down. CI's are fragile beasts.

closed 2017-07-06 by reconbot