serialport/node-serialport #880: Can't use with atom

Sorunome, 2016-07-23:
I cannot install this with atoms package manager `apm`:
```
[sorunome@sorunome-desktop arduino-upload]$ apm --version
apm  1.9.2
npm  2.13.3
node 0.10.40
python 3.5.1
git 2.9.0
```
The output is the following:
```
[sorunome@sorunome-desktop arduino-upload]$ apm install
Installing modules ✗
> serialport@4.0.1 install /home/sorunome/repos/arduino-upload/node_modules/serialport
> node-pre-gyp install --fallback-to-build

Unsupported target version: 0.36.12

node-pre-gyp ERR! install error 
node-pre-gyp ERR! stack Error: Unsupported target version: 0.36.12
node-pre-gyp ERR! stack     at get_runtime_abi (/home/sorunome/repos/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:156:23)
node-pre-gyp ERR! stack     at Object.module.exports.evaluate (/home/sorunome/repos/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:277:19)
node-pre-gyp ERR! stack     at install (/home/sorunome/repos/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/install.js:148:31)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as install] (/home/sorunome/repos/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run (/home/sorunome/repos/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at Object.<anonymous> (/home/sorunome/repos/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:131:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:456:26)
node-pre-gyp ERR! stack     at Object.Module._extensions..js (module.js:474:10)
node-pre-gyp ERR! stack     at Module.load (module.js:356:32)
node-pre-gyp ERR! stack     at Function.Module._load (module.js:312:12)
node-pre-gyp ERR! System Linux 4.4.14-1-lts
node-pre-gyp ERR! command "node" "/home/sorunome/repos/arduino-upload/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/sorunome/repos/arduino-upload/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.40
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok 
npm ERR! Linux 4.4.14-1-lts
npm ERR! argv "/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/sorunome/.atom/.apm/.apmrc" "--userconfig" "/home/sorunome/.atom/.apmrc" "install" "--target=0.36.12" "--arch=x64"
npm ERR! node v0.10.40
npm ERR! npm  v2.13.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/sorunome/repos/arduino-upload/npm-debug.log
```
Relevant part of npm-debug.log:
```
5009 error Linux 4.4.14-1-lts
5010 error argv "/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/sorunome/.atom/.apm/.apmrc" "--userconfig" "/home/sorunome/.atom/.apmrc" "install" "--target=0.36.12" "--arch=x64"
5011 error node v0.10.40
5012 error npm  v2.13.3
5013 error code ELIFECYCLE
5014 error serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
5014 error Exit status 1
5015 error Failed at the serialport@4.0.1 install script 'node-pre-gyp install --fallback-to-build'.
5015 error This is most likely a problem with the serialport package,
5015 error not with npm itself.
5015 error Tell the author that this fails on your system:
5015 error     node-pre-gyp install --fallback-to-build
5015 error You can get their info via:
5015 error     npm owner ls serialport
5015 error There is likely additional logging output above.
```

reconbot, 2016-07-25:
Node 0.10 and atom? That sounds weird to me, they're usually on the latest build of node, their v8 is often newer than node 6.

Do other binary modules work with atom? Like `leveldown` or `sqlite3`?

Sorunome, 2016-07-25:
Just checked, leveldown installs fine, sqlite3 yields the same error

reconbot, 2016-07-28:
I really don't have any background on using npm modules with the atom editor. I'm happy to help debug and even make changes to serialport to better support them, but I think you should probably open a ticket with them (reference this one). Maybe once its figured out we can add some docs about it.

reconbot, 2016-08-13:
For the record we work on node 0.10

reconbot, 2016-08-14:
Atom 1.9 is out and I hear 1.10-beta has a note you might like.

http://blog.atom.io/2016/08/01/atom-1-9-and-1-10-beta.html

```
Packages that depend on native modules now install more reliably.
```

Sorunome, 2016-08-16:
I still can't install under 1.9, i guess waiting for 1.10 then?

reconbot, 2016-09-06:
I suppose so, we do work with electron which atom uses and would be happy to support atom but there's not much we can do for them it seems.

I'm going to close this for now as there's nothing we can do, but please do report back!

closed 2016-09-06 by reconbot

Sorunome, 2016-09-07:
I just tried, it seems to install with atom 1.10, I have not verified the functionality yet, though