serialport/node-serialport #884: to produce delay in successive at commands.

msohaib11, 2016-07-26:
SerialPort version:
ACPI\PNP0501\1
NodeJS Version:
v6.3.0 npm 3.10.3
Operating System and Hardware:
Are you using Electron/Atom/NwJS?:

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem
I am trying to interface modem to nodejs. In using to post AT commands successively a delay is required to view the correct response. I want to start serial communication and then I want to stop and then restart again step by step. please any one can resolve the issue.

### Expected Behavior
serial:'AT\r\n'
response:OK
.
.
.
.

### Actual Behavior

### Steps and Code to Reproduce the Issue

reconbot, 2016-07-27:
You can wait for the data event that has the response. No need to delay anything, when it's sent you'll see it.

msohaib11, 2016-07-27:
thank you for the response. I am using the code her. It only gives the response of first AT command, the remaining commands were not responded. please help:
serialPort.on("open",function(){
serialPort.write('AT', function(err) {
   if (err) {
     return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
serialPort.write('\r', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
serialPort.write('\n', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
setTimeout(function2, 10000);

serialPort.write('AT+CMGF=1', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
serialPort.write('\r', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
serialPort.write('\n', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
serialPort.write('AT+CMGS="03458189964"', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
serialPort.write('\r', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
serialPort.on('data', function (data) {
  console.log('Data: ' + data);});

serialPort.write('^z', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
 });

});
serialPort.on('data', function (data) {
  console.log('Data: ' + data);
});

reconbot, 2016-07-27:
You want to combine your writes, the order is probably guaranteed but since everything is asynchronous this will get messy fast. 

It's late in my time zone so here is a poor and untested example code, but it might lead you the right direction. It would benefit from [async](http://caolan.github.io/async/) or another flow control library.

```js
'use strict';

var SerialPort = require('serialport');
var port = new SerialPort('/dev/cu.USBXXXX', {
  parser: SerialPort.parsers.readline('\r\n')
});

port.on('open',function(){
  port.write('AT\r\n');
  port.once('data', function(buffer){
    if (buffer.toString() !== 'OK') { throw 'Not OK' }
    this.write('AT+CMGF=1\r\n');
    port.once('data', function(buffer){
      if (buffer.toString() !== 'OK') { throw 'Not OK' }
      port.write('AT+CMGS="555555"\r\n');
      port.once('data', function(buffer){
        if (buffer.toString() !== 'OK') { throw 'Not OK' }
        console.log('done!');
      });
    });
  });
});

```

msohaib11, 2016-07-27:
wonderful bro, thanks a lot I try it but finally I want to ask .once is approved method of serialport.

reconbot, 2016-07-27:
It's just like on but happens only once. It's part of the streams interface
and nodejs event emitter we inherit from. It's safe to use and won't be
going away.

reconbot, 2016-07-28:
Let me know how it goes.

msohaib11, 2016-07-28:
sorry I cannoot respond. please help me. It results in same. I tried the code you suggested.It throws error for not okay if I use read line parser. but if I use raw parser then only for the first AT command It results OK and the response for the remaining commands was not shown pleasehelp me.

msohaib11, 2016-08-14:
Can anyone help how we can merge asysnc and serialport to produce delay between two serial writing in extending the code given below. please help me in this regard.is it coorect or not?please help
async.auto({
  
    firstatcommand: port.on('open', function() {
  port.write('AT', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
});,..........................)

msohaib11, 2016-08-16:
can anyone respond to it
async1.auto(
{secondatcommand:function(){
send1;
},thirdrdat:function()
{
port.on('data',sendSerialData(data));
},final:['secondatcommand','3rdat',function(results,final){final;}]

});

function showPortOpen() {
   console.log('port open. Data rate: ' + port.options.baudRate);
}
function send1()
{
port.write('AT\r\n');
}
function final()
{
console.log('final');
}

reconbot, 2017-01-21:
I'm going to close this issue due to it's age, I think the underlying feature request would be a custom parser that supports AT commands. It's a bit more than I'm able to do. I leave it open for others to contribute.

closed 2017-01-21 by reconbot