serialport/node-serialport #885: Flush doesn't clear parser buffer.

flaminggoat, 2016-07-26:
SerialPort version: 4.0.1
NodeJS Version: 4.2.1
Operating System and Hardware: RPi 2
Are you using Electron/Atom/NwJS?: No

### Summary of Problem
Data remains in buffer of parser after flush is called.
### Expected Behavior
No data in parser buffer after flush.
### Actual Behavior
Data in parser buffer after flush.
### Steps and Code to Reproduce the Issue
Use newline parser. Receive some data without newline. Call flush. Receive data + newline. Data event contains the data from before the flush as well as the data from after it.

reconbot, 2016-07-27:
It flushes the operating system buffer so it wont have any effect on the parser. We don't have a built in parser that lets you flush it. (not yet anyway)

closed 2016-07-27 by reconbot

flaminggoat, 2016-07-27:
Surely that is important functionality. If you know there is garbage in the parser, it seems the only way to empty it is to close and reopen the serial port.

reconbot, 2016-07-28:
This hasn't come up in an issue yet. You're the first. If you really want, you can "flush" the readline parser by calling it directly with a new line character. It will think it came from the port and emit a data event.

```js
port.options.dataCallback(new Buffer('\n'));
```