serialport/node-serialport #886: Opened port immediately disconnects with an ENXIO error on OS X

codybonney, 2016-07-27:
SerialPort version: `4.0.1`
NodeJS Version: `4.4.7`
Operating System and Hardware: `OS X El Capitan 10.11.2` `MacBook Pro`
Python Version: `2.7.10`
Are you using Electron/Atom/NwJS?: `No`
Device: [Mini Maestro 12-Channel USB Servo Controller](http://www.pololu.com/product/1352)

### Summary of Problem
When a port is opened, it disconnects almost immediately with an ENXIO error.

### Expected Behavior
A port can be opened and stay open, without it immediately disconnecting with an error.

### Actual Behavior
I'm able to detect and open a port. However, it disconnects very shortly after it's opened (~100-400ms) with the following error:
```
{ [Error: ENXIO: no such device or address, read] errno: -6, code: 'ENXIO', syscall: 'read' }
```

### Notes
- I'm able to write to the port in that very short window before it disconnects and closes. The device does respond.
- Running the script quickly in succession results in it not being able to detect the port in the first place. It detects it again after waiting for 1-2sec
- After running the script quickly in succession, `ls -la /dev/cu.usbmodem00043071` returns `No such file or directory` but reappears after 1-2sec
- I had no problems communicating with the device on an older version of OS X, node, and node-serialport (<v1.0.0).
- The user account on OS X has admin privileges

### Steps and Code to Reproduce the Issue
1. Plug in Mini Maestro Device
2. Copy/Paste path to serial port into script
3. Run script `node SerialPortTest.js` (see below)

### Script 
```
// Requires
var SerialPort  = require("serialport");

var portOpenMS = 0,
    portOpenTimer = null,
    startPortOpenInterval = function() {
        portOpenTimer = setInterval(function() {
            portOpenMS += 1;
        }, 1)
    },
    stopPortOpenInterval = function() {
        clearInterval(portOpenTimer);
    },
    resetPortOpenInterval = function() {
        portOpenMS = 0;
    };

serialPort = new SerialPort('/dev/cu.usbmodem00043071', {
    autoOpen : false
});

/**
 * Setup event handlers
 */
serialPort.on('open', function() {
    console.log("Port was opened");
});

serialPort.on('data', function(data) {
    console.log("Port received data - " + data)
});

serialPort.on('disconnect', function(err) {
    stopPortOpenInterval();
    console.log("Port was disconnected - ");
    console.error(err);
});

serialPort.on('error', function(err) {
    console.log("Port threw error - ");
    console.error(err);
});

serialPort.on('close', function() {
    stopPortOpenInterval();
    console.log("Port was closed - port was open for " + portOpenMS + "ms");
    resetPortOpenInterval();
});

// Manually open the port
serialPort.open(function (err) {

    // short circuit if error
    if (err) {
        return console.log('Error opening port: ', err.message);
    }

    if (serialPort.isOpen()) {
        console.log("Port is open");
        startPortOpenInterval();
    }
    else {
        console.log("Port should be open, but is not");
    }
});
```

## Output
```
Port was opened
Port is open
Port was disconnected - 
{ [Error: ENXIO: no such device or address, read] errno: -6, code: 'ENXIO', syscall: 'read' }
Port was closed - port was open for 236ms
```

reconbot, 2016-07-27:
I'm willing to bet your browning out the USB to serial chip and it restarts. Are you using external power?

reconbot, 2016-07-27:
In fact I know that at some point OSX limited current draw on the USB ports (except for iPhones) so that might be the reason it used to work.

codybonney, 2016-08-06:
Thank you for the reply! I didn't know about OS X limiting current draw!

I ended up trying several different configurations for powering the controller and servos, including an externally powered USB hub. Unfortunately, I kept running into the same issue.

After some more research, I found out that [Polulu released a firmware update](https://www.pololu.com/docs/0J40/4.f) for the device very recently. 

The solution was to spin up a Windows VM in VirtualBox, install Pololu's software, and upgrade the device to the new firmware version. After that, everything worked perfectly with the latest version of node-serialport on OS X!

I wrote up a little guide on how to [upgrade the pololu firmware using OS X and VirtualBox](https://codybonney.com/upgrade-firmware-on-pololu-mini-maestro-using-os-x-and-virtualbox/) in case anyone else gets stuck in the same situation. A few parts were a little tricky.

Thanks again so much for your help and for maintaining this project!

closed 2016-08-06 by codybonney

reconbot, 2016-08-06:
Yay!