serialport/node-serialport #889: Build error for OSX 10.11.4 (15E65)

duncup, 2016-07-28:
SerialPort version:4.0.1
NodeJS Version:4.4.2
npm Version:3.10.6
Python Version:2.7.10
Operating System and Hardware:
Mac book pro for 2015.
OSX 10.11.4 (15E65).
Are you using Electron/Atom/NwJS?:
I'm using Atom editor.
I'm using this project to nwjs.
(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem
I'm using nwjs tool nw-gyp to build this project.
But i get some error.
### Expected Behavior
Ok,i want to build this project and use this to control my printer.
### Actual Behavior
I got this error massage.
```
gyp info it worked if it ends with ok
gyp info using nw-gyp@3.4.0
gyp info using node@4.4.2 | darwin | x64
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
In file included from ../../nan/nan.h:48:0,
                 from ../src/./serialport.h:7,
                 from ../src/serialport_unix.cpp:1:
/Users/steppenwolf/.nw-gyp/0.16.0/src/node_buffer.h:16:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
 #pragma clang diagnostic push
 ^
/Users/steppenwolf/.nw-gyp/0.16.0/src/node_buffer.h:17:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
 #pragma clang diagnostic ignored "-Wignored-attributes"
 ^
/Users/steppenwolf/.nw-gyp/0.16.0/src/node_buffer.h:19:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
 #pragma clang diagnostic pop
 ^
/Users/steppenwolf/.nw-gyp/0.16.0/src/node_buffer.h:18:64: warning: 'visibility' attribute ignored [-Wattributes]
 NODE_EXTERN typedef void (*FreeCallback)(char* data, void* hint);
                                                                ^
In file included from /System/Library/Frameworks/IOKit.framework/Headers/usb/IOUSBLib.h:27:0,
                 from ../src/serialport_unix.cpp:14:
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:612:23: error: too many #pragma options align=reset
 #pragma options align=reset
                       ^
In file included from /System/Library/Frameworks/IOKit.framework/Headers/usb/IOUSBLib.h:27:0,
                 from ../src/serialport_unix.cpp:14:
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:782:23: error: too many #pragma options align=reset
 #pragma options align=reset
                       ^
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:834:23: error: too many #pragma options align=reset
 #pragma options align=reset
                       ^
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:852:23: error: too many #pragma options align=reset
 #pragma options align=reset
                       ^
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:872:23: error: too many #pragma options align=reset
 #pragma options align=reset
                       ^
../src/serialport_unix.cpp: In function 'OpenBatonPlatformOptions* ParsePlatformOptions(const v8::Local<v8::Object>&)':
../src/serialport_unix.cpp:44:109: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /Users/steppenwolf/.nw-gyp/0.16.0/deps/v8/include/v8.h:8185): Use maybe version [-Wdeprecated-declarations]
   result->vmin = Nan::Get(options, Nan::New<v8::String>("vmin").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
                                                                                                             ^
../src/serialport_unix.cpp:45:111: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /Users/steppenwolf/.nw-gyp/0.16.0/deps/v8/include/v8.h:8185): Use maybe version [-Wdeprecated-declarations]
   result->vtime = Nan::Get(options, Nan::New<v8::String>("vtime").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
                                                                                                               ^
make: *** [Release/obj.target/serialport/src/serialport_unix.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/nw-gyp/lib/build.js:294:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 15.4.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/bin/nw-gyp" "build" "--target=0.16.0"
gyp ERR! cwd /Users/steppenwolf/gbl/posOrderChecker/node_modules/serialport
gyp ERR! node -v v4.4.2
gyp ERR! nw-gyp -v v3.4.0
gyp ERR! not ok 
```
### Steps and Code to Reproduce the Issue
I'll install the nw-builder and the nw-gyp.
Use npm install this project.
Change dir to this project path.
Run the commad `nw-gyp rebuild --target=0.16.0`.
This is all.
Thanks.

duncup, 2016-07-28:
I'll use the node-pre-gyp build this project too.
`node-pre-gyp build --runtime=node-webkit --target=0.16.0`
And this is error massage.
```
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.29
node-pre-gyp info using node@4.4.2 | darwin | x64
gyp info it worked if it ends with ok
gyp info using nw-gyp@3.4.0
gyp info using node@4.4.2 | darwin | x64
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
In file included from ../../nan/nan.h:48:0,
                 from ../src/./serialport.h:7,
                 from ../src/serialport_unix.cpp:1:
/Users/steppenwolf/.nw-gyp/0.16.0/src/node_buffer.h:16:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
 #pragma clang diagnostic push
 ^
/Users/steppenwolf/.nw-gyp/0.16.0/src/node_buffer.h:17:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
 #pragma clang diagnostic ignored "-Wignored-attributes"
 ^
/Users/steppenwolf/.nw-gyp/0.16.0/src/node_buffer.h:19:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
 #pragma clang diagnostic pop
 ^
/Users/steppenwolf/.nw-gyp/0.16.0/src/node_buffer.h:18:64: warning: 'visibility' attribute ignored [-Wattributes]
 NODE_EXTERN typedef void (*FreeCallback)(char* data, void* hint);
                                                                ^
In file included from /System/Library/Frameworks/IOKit.framework/Headers/usb/IOUSBLib.h:27:0,
                 from ../src/serialport_unix.cpp:14:
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:612:23: error: too many #pragma options align=reset
 #pragma options align=reset
                       ^
In file included from /System/Library/Frameworks/IOKit.framework/Headers/usb/IOUSBLib.h:27:0,
                 from ../src/serialport_unix.cpp:14:
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:782:23: error: too many #pragma options align=reset
 #pragma options align=reset
                       ^
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:834:23: error: too many #pragma options align=reset
 #pragma options align=reset
                       ^
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:852:23: error: too many #pragma options align=reset
 #pragma options align=reset
                       ^
/System/Library/Frameworks/IOKit.framework/Headers/usb/USB.h:872:23: error: too many #pragma options align=reset
 #pragma options align=reset
                       ^
../src/serialport_unix.cpp: In function 'OpenBatonPlatformOptions* ParsePlatformOptions(const v8::Local<v8::Object>&)':
../src/serialport_unix.cpp:44:109: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /Users/steppenwolf/.nw-gyp/0.16.0/deps/v8/include/v8.h:8185): Use maybe version [-Wdeprecated-declarations]
   result->vmin = Nan::Get(options, Nan::New<v8::String>("vmin").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
                                                                                                             ^
../src/serialport_unix.cpp:45:111: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /Users/steppenwolf/.nw-gyp/0.16.0/deps/v8/include/v8.h:8185): Use maybe version [-Wdeprecated-declarations]
   result->vtime = Nan::Get(options, Nan::New<v8::String>("vtime").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
                                                                                                               ^
make: *** [Release/obj.target/serialport/src/serialport_unix.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/nw-gyp/lib/build.js:294:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 15.4.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/bin/nw-gyp" "build" "--runtime=node-webkit" "--target=0.16.0" "--module=/Users/steppenwolf/gbl/posOrderChecker/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/Users/steppenwolf/gbl/posOrderChecker/node_modules/serialport/build/Release"
gyp ERR! cwd /Users/steppenwolf/gbl/posOrderChecker/node_modules/serialport
gyp ERR! node -v v4.4.2
gyp ERR! nw-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'nw-gyp build --runtime=node-webkit --target=0.16.0 --module=/Users/steppenwolf/gbl/posOrderChecker/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/Users/steppenwolf/gbl/posOrderChecker/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:827:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Darwin 15.4.0
node-pre-gyp ERR! command "/usr/local/bin/node" "/usr/local/bin/node-pre-gyp" "build" "--runtime=node-webkit" "--target=0.16.0"
node-pre-gyp ERR! cwd /Users/steppenwolf/gbl/posOrderChecker/node_modules/serialport
node-pre-gyp ERR! node -v v4.4.2
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok 
Failed to execute 'nw-gyp build --runtime=node-webkit --target=0.16.0 --module=/Users/steppenwolf/gbl/posOrderChecker/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/Users/steppenwolf/gbl/posOrderChecker/node_modules/serialport/build/Release' (1)
```

reconbot, 2016-07-28:
I don't know enough about nwjs to help debug this without assistance from nwjs. I know people have gotten it to work, and I'll happily make changes to better support it but I think you should open an issue on their project and ping this issue or me.

duncup, 2016-08-01:
Ok,Thanks.

duncup, 2016-08-01:
I can build this now.
FYI: Don't use mp-gcc to build this.

reconbot, 2016-08-01:
Great!

closed 2016-08-01 by reconbot