serialport/node-serialport #894: Serialport on NW.js spontaneously closes app on some Windows machines when accessing 'undefined' port properties

nanowizard, 2016-08-05:
SerialPort version: 3.1.2
Operating System and Hardware: Dell Latitude E5520, Win 7 Pro, Core i5-2520M 2.5GHZ, 4GB RAM.
Are you using Electron/Atom/NwJS?: NW.js version 0.15.0

### Problem
After making some changes to our app code, everything would work perfectly fine on most of our PCs, but on the Dell Latitude referenced above the app would close almost immediately after opening. After investigating with the dev tools on that machine I discovered it was occurring when processing the result of the `SerialPort.list` function. Specifically, it occurrs on the following line, where I create a label based on the port `comName` and `manufacturer` if it exists.

    p.label = p.comName + (p.manufacturer ? ' ('+p.manufacturer+')' : '');

It appears to occur whenever I attempt to do any kind of boolean operation on a port property that shows up with the mysterious `hole` type. The following is a screenshot from some tests I was doing in the developer console, where I stored a port object as a global variable `temp1`:
[image]
After executing the command `!temp1.manufacturer`, the program immediately exited.

No errors are ever printed to the console. And the process closes with an exit code of 0.

### Steps and Code to Reproduce the Issue
I am unsure specifically what kind of system configuration causes this to occur. As I mentioned above, everything worked fine on almost all of our other Windows machines. It could have been due to the fact that the `manufacturer` property on that port object was a `hole` type. I saw the same behavior when trying to do operations on the other properties set to `hole`.

On such a machine, the following code should reproduce the issue:
```
SerialPort.list(function(err, ports){
    ports.forEach(function(p){
        console.log(!p.manufacturer);
    });
});
```

reconbot, 2016-08-05:
I have never in my life seen a `hole` data type. I know it's a concept with arrays but...

I'd open an issue on nwjs. We provide a guaranteed `undefined` when information isn't available.

reconbot, 2016-08-05:
And totally report back here! Making it easy for people to use SP with whatever ideas they have is important to us. I'm happy to help debug on a nwjs ticket too.

nanowizard, 2016-08-09:
Yeah I was completely baffled when I saw that `hole` type. I [just opened the issue on nwjs](https://github.com/nwjs/nw.js/issues/5201). My workaround for the time being is to just not do any kind of boolean operations on those variables. So I changed this line:

`p.label = p.comName + (p.manufacturer ? ' ('+p.manufacturer+')' : '');`

To this:

p.label = p.comName + ' ' + p.manufacturer;

Which doesn't cause any problems.

reconbot, 2016-09-06:
Browser serialport as suggested in that ticket will soon be merged with node-serialport with a chrome serialport backend. It's not as up to date with the api, but it does work well. I'm going to close this ticket as it's an issue with NW.js

closed 2016-09-06 by reconbot