serialport/node-serialport #895: errors running serialport on linux (using rfxcom)

Averelll, 2016-08-06:
SerialPort version: 2.1.2, 3.1.1, 4.01
NodeJS Version: 6.2.2
Operating System and Hardware: Debian Jessie, under Xen PV
Are you using Electron/Atom/NwJS?: no

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem
When running RFXCOM that uses serialport, on start I get the following error:

==== JS stack trace =========================================

Security context: 0x187acbbc9e59 <JS Object>#0#
    1: .node [module.js:568] [pc=0xf4f8846df04] (this=0xfc33d00bd31 <an Object with map 0x698bb517769>#1#,module=0xfc33d138799 <a Module with map 0x698bb517d41>#2#,filename=0xfc33d138731 <String[74]: /etc/nodejs/apps/rfx/node_modules/serialport/build/Release/serialport.node>)
    2: load [module.js:458] [pc=0xf4f88434f92] (this=0xfc33d138799 <a Module with map 0x698bb517d41>#2#,filename=0xfc33d138731 <String[74]: /etc/nodejs/apps/rfx/node_modules/serialport/build/Release/serialport.node>)
    3: tryModuleLoad(aka tryModuleLoad) [module.js:417] [pc=0xf4f88434abd] (this=0x187acbb04189 <undefined>,module=0xfc33d138799 <a Module with map 0x698bb517d41>#2#,filename=0xfc33d138731 <String[74]: /etc/nodejs/apps/rfx/node_modules/serialport/build/Release/serialport.node>)
    4: _load [module.js:409] [pc=0xf4f884306a2] (this=0xfc33d004f39 <JS Function Module (SharedFunctionInfo 0x27458f922d79)>#3#,request=0xfc33d137d41 <String[74]: /etc/nodejs/apps/rfx/node_modules/serialport/build/Release/serialport.node>,parent=0xfc33d129679 <a Module with map 0x698bb517d41>#4#,isMain=0x187acbb04299 <false>)
    5: require [module.js:468] [pc=0xf4f8843eb13] (this=0xfc33d129679 <a Module with map 0x698bb517d41>#4#,path=0xfc33d137d41 <String[74]: /etc/nodejs/apps/rfx/node_modules/serialport/build/Release/serialport.node>)
    6: require(aka require) [internal/module.js:20] [pc=0xf4f8843e846] (this=0x187acbb04189 <undefined>,path=0xfc33d137d41 <String[74]: /etc/nodejs/apps/rfx/node_modules/serialport/build/Release/serialport.node>)
    7: bindings [/etc/nodejs/apps/rfx/node_modules/bindings/bindings.js:76] [pc=0xf4f884699b6] (this=0x187acbbe3fc1 <JS Global Object>#5#,opts=0x27458f9a7e61 <String[15]: serialport.node>)
    8: /* anonymous */ [/etc/nodejs/apps/rfx/node_modules/serialport/lib/bindings.js:3] [pc=0xf4f8846822e] (this=0xfc33d122ba1 <an Object with map 0x35d9907ac1>#6#,exports=0xfc33d122ba1 <an Object with map 0x35d9907ac1>#6#,require=0xfc33d1246d9 <JS Function require (SharedFunctionInfo 0x27458f953911)>#7#,module=0xfc33d122b51 <a Module with map 0x698bb517d41>#8#,__filename=0xfc33d122af9 <String[60]: /etc/nodejs/apps/rfx/node_modules/serialport/lib/bindings.js>,__dirname=0xfc33d124671 <String[48]: /etc/nodejs/apps/rfx/node_modules/serialport/lib>)
    9: _compile [module.js:541] [pc=0xf4f8843da90] (this=0xfc33d122b51 <a Module with map 0x698bb517d41>#8#,content=0xfc33d123cf1 <String[784]\: 'use strict';\n\nvar bindings = require('bindings')('serialport.node');\nvar listUnix = require('./list-unix');\n\nvar linux = process.platform !== 'win32' && process.platform !== 'darwin';\n\nfunction listLinux(callback) {\n  callback = callback || function(err) {\n    if (err) { this.emit('error', err) }\n  }.bind(this);\n  return listUnix(callback);\n};\n\nvar platformOptions = {};\nif (process.platform !== 'win32') {\n  platformOptions = {\n    vmin: 1,\n    vtime: 0\n  };\n}\n\nmodule.exports = {\n  close: bindings.close,\n  drain: bindings.drain,\n  flush: bindings.flush,\n  list: linux ? listLinux : bindings.list,\n  open: bindings.open,\n  SerialportPoller: bindings.SerialportPoller,\n  set: bindings.set,\n  update: bindings.update,\n  write: bindings.write,\n  platformOptions: platformOptions\n};\n>,filename=0xfc33d122af9 <String[60]: /etc/nodejs/apps/rfx/node_modules/serialport/lib/bindings.js>)
   10: .js [module.js:550] [pc=0xf4f8843658b] (this=0xfc33d00bd31 <an Object with map 0x698bb517769>#1#,module=0xfc33d122b51 <a Module with map 0x698bb517d41>#8#,filename=0xfc33d122af9 <String[60]: /etc/nodejs/apps/rfx/node_modules/serialport/lib/bindings.js>)
   11: load [module.js:458] [pc=0xf4f88434f92] (this=0xfc33d122b51 <a Module with map 0x698bb517d41>#8#,filename=0xfc33d122af9 <String[60]: /etc/nodejs/apps/rfx/node_modules/serialport/lib/bindings.js>)
   12: tryModuleLoad(aka tryModuleLoad) [module.js:417] [pc=0xf4f88434abd] (this=0x187acbb04189 <undefined>,module=0xfc33d122b51 <a Module with map 0x698bb517d41>#8#,filename=0xfc33d122af9 <String[60]: /etc/nodejs/apps/rfx/node_modules/serialport/lib/bindings.js>)
   13: _load [module.js:409] [pc=0xf4f884306a2] (this=0xfc33d004f39 <JS Function Module (SharedFunctionInfo 0x27458f922d79)>#3#,request=0x27458f965d49 <String[10]: ./bindings>,parent=0xfc33d004fa1 <a Module with map 0x698bb517d41>#9#,isMain=0x187acbb04299 <false>)
   14: require [module.js:468] [pc=0xf4f8843eb13] (this=0xfc33d004fa1 <a Module with map 0x698bb517d41>#9#,path=0x27458f965d49 <String[10]: ./bindings>)
   15: require(aka require) [internal/module.js:20] [pc=0xf4f8843e846] (this=0x187acbb04189 <undefined>,path=0x27458f965d49 <String[10]: ./bindings>)
   16: /* anonymous */ [/etc/nodejs/apps/rfx/node_modules/serialport/lib/serialport.js:16] [pc=0xf4f8844b6ae] (this=0xfc33d005279 <an Object with map 0x35d9907ac1>#10#,exports=0xfc33d005279 <an Object with map 0x35d9907ac1>#10#,require=0xfc33d005069 <JS Function require (SharedFunctionInfo 0x27458f953911)>#11#,module=0xfc33d004fa1 <a Module with map 0x698bb517d41>#9#,__filename=0xfc33d005221 <String[62]: /etc/nodejs/apps/rfx/node_modules/serialport/lib/serialport.js>,__dirname=0xfc33d0051f9 <String[48]: /etc/nodejs/apps/rfx/node_modules/serialport/lib>)
   17: _compile [module.js:541] [pc=0xf4f8843da90] (this=0xfc33d004fa1 <a Module with map 0x698bb517d41>#9#,content=0xfc33d008821 <Very long string[13557]>#12#,filename=0xfc33d005221 <String[62]: /etc/nodejs/apps/rfx/node_modules/serialport/lib/serialport.js>)
   18: .js [module.js:550] [pc=0xf4f8843658b] (this=0xfc33d00bd31 <an Object with map 0x698bb517769>#1#,module=0xfc33d004fa1 <a Module with map 0x698bb517d41>#9#,filename=0xfc33d005221 <String[62]: /etc/nodejs/apps/rfx/node_modules/serialport/lib/serialport.js>)
   19: load [module.js:458] [pc=0xf4f88434f92] (this=0xfc33d004fa1 <a Module with map 0x698bb517d41>#9#,filename=0xfc33d005221 <String[62]: /etc/nodejs/apps/rfx/node_modules/serialport/lib/serialport.js>)
   20: tryModuleLoad(aka tryModuleLoad) [module.js:417] [pc=0xf4f88434abd] (this=0x187acbb04189 <undefined>,module=0xfc33d004fa1 <a Module with map 0x698bb517d41>#9#,filename=0xfc33d005221 <String[62]: /etc/nodejs/apps/rfx/node_modules/serialport/lib/serialport.js>)
   21: _load [module.js:409] [pc=0xf4f884306a2] (this=0xfc33d004f39 <JS Function Module (SharedFunctionInfo 0x27458f922d79)>#3#,request=0x27458f9562c9 <String[10]: serialport>,parent=0xfc33d00bdd1 <a Module with map 0x698bb517d41>#13#,isMain=0x187acbb04299 <false>)
   22: require [module.js:468] [pc=0xf4f8843eb13] (this=0xfc33d00bdd1 <a Module with map 0x698bb517d41>#13#,path=0x27458f9562c9 <String[10]: serialport>)
   23: require(aka require) [internal/module.js:20] [pc=0xf4f8843e846] (this=0x187acbb04189 <undefined>,path=0x27458f9562c9 <String[10]: serialport>)
   24: /* anonymous */ [/etc/nodejs/apps/rfx/node_modules/rfxcom/lib/rfxcom.js:1] [pc=0xf4f88448d52] (this=0xfc33d00c031 <an Object with map 0x35d9907ac1>#14#,exports=0xfc33d00c031 <an Object with map 0x35d9907ac1>#14#,require=0xfc33d00be99 <JS Function require (SharedFunctionInfo 0x27458f953911)>#15#,module=0xfc33d00bdd1 <a Module with map 0x698bb517d41>#13#,__filename=0xfc33d00bfe1 <String[54]: /etc/nodejs/apps/rfx/node_modules/rfxcom/lib/rfxcom.js>,__dirname=0xfc33d00bfb9 <String[44]: /etc/nodejs/apps/rfx/node_modules/rfxcom/lib>)
   25: _compile [module.js:541] [pc=0xf4f8843da90] (this=0xfc33d00bdd1 <a Module with map 0x698bb517d41>#13#,content=0xfc33d015131 <Very long string[36936]>#16#,filename=0xfc33d00bfe1 <String[54]: /etc/nodejs/apps/rfx/node_modules/rfxcom/lib/rfxcom.js>)
   26: .js [module.js:550] [pc=0xf4f8843658b] (this=0xfc33d00bd31 <an Object with map 0x698bb517769>#1#,module=0xfc33d00bdd1 <a Module with map 0x698bb517d41>#13#,filename=0xfc33d00bfe1 <String[54]: /etc/nodejs/apps/rfx/node_modules/rfxcom/lib/rfxcom.js>)
   27: load [module.js:458] [pc=0xf4f88434f92] (this=0xfc33d00bdd1 <a Module with map 0x698bb517d41>#13#,filename=0xfc33d00bfe1 <String[54]: /etc/nodejs/apps/rfx/node_modules/rfxcom/lib/rfxcom.js>)
[... cut ...]
(node) and will stop working in the next major release.

After that, it actually works, I do get input from the serial port, but the error occurs every single time I start my node module. It happens with all version of serialport that I tried.

### Expected Behavior

### Actual Behavior

### Steps and Code to Reproduce the Issue

Install rfxcom (npm install rfxcom)
Use a simple example like:

var rfxcom = require('rfxcom');
var rfxtrx = new rfxcom.RfxCom("/dev/ttyUSB0", {debug: true});
rfxtrx.initialise();

reconbot, 2016-08-06:
> pc=0xf4f8846822e 9: _compile [module.js:541]
> pc=0xf4f8843da90('serialport.node');\nvar listUnix =
> platformOptions\n};\n>,filename=0xfc33d122af9 ) 10: .js [module.js:550]
> pc=0xf4f8843658b 11: load [module.js:458] pc=0xf4f88434f92 12:
> tryModuleLoad(aka tryModuleLoad) [module.js:417] pc=0xf4f88434abd 13: _load
> [module.js:409] pc=0xf4f884306a2 14: require [module.js:468]
> pc=0xf4f8843eb13 15: require(aka require) [internal/module.js:20]
> pc=0xf4f8843e846 16: /* anonymous
> pc=0xf4f8844b6ae 17: _compile [module.js:541] pc=0xf4f8843da90 18: .js
> [module.js:550] pc=0xf4f8843658b 19: load [module.js:458] pc=0xf4f88434f92
> 20: tryModuleLoad(aka tryModuleLoad) [module.js:417] pc=0xf4f88434abd 21:
> _load [module.js:409] pc=0xf4f884306a2 22: require [module.js:468]
> pc=0xf4f8843eb13 23: require(aka require) [internal/module.js:20]
> pc=0xf4f8843e846 24: /* anonymous
> pc=0xf4f88448d52 25: _compile [module.js:541] pc=0xf4f8843da90 26: .js
> [module.js:550] pc=0xf4f8843658b 27: load [module.js:458] pc=0xf4f88434f92
> 28: tryModuleLoad(aka tryModuleLoad) [module.js:417] pc=0xf4f88434abd 29:
> _load [module.js:409] pc=0xf4f884306a2 30: require [module.js:468]
> pc=0xf4f8843eb13 31: require(aka require) [internal/module.js:20]
> pc=0xf4f8843e846 32: /* anonymous
> pc=0xf4f88445ae2 33: _compile [module.js:541] pc=0xf4f8843da90 34: .js
> [module.js:550] pc=0xf4f8843658b 35: load [module.js:458] pc=0xf4f88434f92
> 36: tryModuleLoad(aka tryModuleLoad) [module.js:417] pc=0xf4f88434abd 37:
> _load [module.js:409] pc=0xf4f884306a2 38: require [module.js:468]
> pc=0xf4f8843eb13 39: require(aka require) [internal/module.js:20]
> [... 2 lines cut ...]
> pc=0xf4f8843658b 43: load [module.js:458] pc=0xf4f88434f92 44:
> tryModuleLoad(aka tryModuleLoad) [module.js:417] pc=0xf4f88434abd 45: _load
> [module.js:409] pc=0xf4f884306a2 46: require [module.js:468]
> pc=0xf4f8843eb13 47: require(aka require) [internal/module.js:20]
> pc=0xf4f8843e846 48: /* anonymous */ [/etc/nodejs/apps/rfx/rfx.js:1]

2 things.

What exactly is xen pv? My hunch is it doesnt like our precompiled
binaries. If you run `npm install --build-from-source` it would work around
that.

Rfxcom is using serialport 2x. 3x fixed major issues with minimal API
changes. 4x is even better. If you open an issue with then an ping me, I'll
gladly help upgrade.

Averelll, 2016-08-06:
Xen is similar to VMware, PV just means it runs para virtualised (in contrast to HVM, which mean full hardware virtualisation. I have serial ports working, just the bug keeps occurring. I tried multiple versions of serialport, I also tried the 'npm install --build-from-source`, even though it builds ok, it still gives me this issue on run. Btw, the same errors I get when running on a Raspberry pi.

reconbot, 2016-08-06:
When you run the same code on the RPI exactly?

reconbot, 2016-08-06:
My question is are you able to reproduce this error on multiple systems without `rfxcom`?

Averelll, 2016-08-06:
I was trying just this, when I noticed that I was actually looking at a warning this is a know issue: https://github.com/EmergingTechnologyAdvisors/node-serialport/pull/777/commits. I forked node-rfxcom, make the proper changes to use serialport v4.01 (use of autoOpen, not openImmediately). Now the issue is gone. I will inform the developers of node-Rfxcom on this. Appreciate the quick responses, consider the issue closed.

reconbot, 2016-08-06:
great!

closed 2016-08-06 by reconbot