serialport/node-serialport #902: [feature request] idle-based parser

iurly, 2016-08-10:
### Summary of Problem
The default raw parser calls onData() upon reception of at most 8 bytes. I believe this is due to the intrinsic nature of the UART.
For PDU-based protocols (e.g. Modbus RTU), this makes the user responsible for reconstructing bigger frames and implementing their own timeout mechanisms.
Notice how none of the existing parsers fits this purpose, as such protocols do not specify a predefined length or an explicit terminator / byte delimiter.

### Expected Behavior
For such protocols, it would be nice to have some mechanism by which the whole "frame" is read, using a mere user-configurable "silence" (e.g. 100ms) as a delimiter. This way, data is only emitted after the line has been idle for such amount of time, leaving this complexity out from such applications.

reconbot, 2016-08-10:
This would make an interesting "silence" transform stream

reconbot, 2016-09-06:
In our streams rewrite we now have transform streams for parsers, this makes it a little easier to have a self contained parser. This isn't one I'd publish with serialport unless there's some demand (I don't really know too much about PDU based protocols but I'm willing to be schooled.) It would be straightforward to publish it as a separate package. 

The following appears to work, and you're free to steal it but I didn't write tests for it, so ymmv.
```js
'use strict';
var Transform = require('readable-stream').Transform;
var inherits = require('inherits');

function TimeoutParser(options) {
  if (!(this instanceof TimeoutParser)) {
    return new TimeoutParser(options);
  }
  Transform.call(this, options);

  options = options || {};

  if (options.time === undefined) {
    throw new TypeError('time not set');
  }
  this.time = options.time;
  this.buffer = new Buffer(0);
  this.timeout = null;
}

inherits(TimeoutParser, Transform);

TimeoutParser.prototype._transform = function(chunk, encoding, cb) {
  this.buffer = Buffer.concat([this.buffer, chunk]);
  cb();
  if (this.timeout) {
    clearTimeout(this.timeout);
    this.timeout = null;
  }
  this.timeout = setTimeout(function() {
    this.timeout = null;
    this.push(this.buffer);
    this.buffer = new Buffer(0);
  }.bind(this), this.time);
};

// When the stream is closed
TimeoutParser.prototype._flush = function(cb) {
  if (this.timeout) {
    clearTimeout(this.timeout);
    this.timeout = null;
  }
  this.push(this.buffer);
  this.buffer = new Buffer(0);
  cb();
};

module.exports = TimeoutParser;
```

closed 2016-09-06 by reconbot

reconbot, 2016-09-06:
Upon reflection If someone wants to do a PR with tests I'd be happy to merge this. 

I do think modbus support is a cheap and useful thing to support. [Modbus RTU](http://www.rtaautomation.com/technologies/modbus-rtu/) makes me a little sad but so it goes.