serialport/node-serialport #903: [feature request] Wrapper to enable RS-485

jcwren, 2016-08-11:
### Summary of Problem

Possibly add a wrapper to enable the RS-485 functionality on a Linux serial port. RS-485 half-duplex operation uses the RTS line to switch the RS-485 transciever between receive and transmit. This functionality is built-in to the kernel serial driver, and is managed through ioctl() calls.  https://www.kernel.org/doc/Documentation/serial/serial-rs485.txt

reconbot, 2016-08-22:
What is this used to talk to?

Unless this is a common use case we probably won't land support for it specifically in serialport. 
I'm in the middle of pulling out our c++ bindings into their own objects. When that is done it will be a lot easier to fork the linux binding into it's own module and add this functionality and use it with serialport. 

I also see a future where you can do arbitrary ioctl commands at the binding level (for tuning or options like this).

reconbot, 2016-09-06:
I'm going to close this for now, but if you'd like to continue with it feel free to comment and we'll reopen. I'd look into using [`ioctl`](https://www.npmjs.com/package/ioctl) with serialport to achive this.

closed 2016-09-06 by reconbot