serialport/node-serialport #904: Electron and serialport on raspberry pi 3, Module version missmatch error

styoe, 2016-08-11:
Python: 2.7.9
Serialport: 3.8.6
electron prebuilt: 3.8.6
raspberry 3 linux 4.4.16-v7+ 
Node: 5.12.0
npm: 3.8.6

I cannot get the app to run without the Module version missmatch error. I followed a bunch of tutorials here, suggesting rebuilding the module, changing node & npm & electron versions, and so far i have managed to change the error from: 
"Expected 49, got 46" to  "Expected 49, got 47" to "Expected 49, got 48".

Do you please have any more ideas, or point me in the right direction as i am clueless by now.

I tried clearing the npm cache, rebuilding all modules, rebuilding just serialport, deleting node modules and reinstalling them. Using node-gyp to rebuild the binaries... I think i tried every solution out there.

So far, by logging the line before the line that breaks in the node_modules/bindings/bindings.js have managed to isolate the problem ( i think ) to serialport/dist/Release/serialport.node

Here is the log:

> eres-one@1.0.0 start /home/pi/eres-one/app
> electron main.js

#MY LOG
/home/pi/eres-one/app/node_modules/serialport/build/serialport.node
/home/pi/eres-one/app/node_modules/serialport/build/Debug/serialport.node
/home/pi/eres-one/app/node_modules/serialport/build/Release/serialport.node
#END MY LOG

App threw an error during load
Error: Module version mismatch. Expected 49, got 47.
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:167:20)
    at Object.Module._extensions..node (module.js:568:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:167:20)
    at Module.load (module.js:458:32)
    at tryModuleLoad (module.js:417:12)
    at Function.Module._load (module.js:409:3)
    at Module.require (module.js:468:17)
    at require (internal/module.js:20:19)
    at bindings (/home/pi/eres-one/app/node_modules/bindings/bindings.js:77:44)
A JavaScript error occurred in the main process
Uncaught Exception:
Error: Module version mismatch. Expected 49, got 47.
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:167:20)
    at Object.Module._extensions..node (module.js:568:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:167:20)
    at Module.load (module.js:458:32)
    at tryModuleLoad (module.js:417:12)
    at Function.Module._load (module.js:409:3)
    at Module.require (module.js:468:17)
    at require (internal/module.js:20:19)
    at bindings (/home/pi/eres-one/app/node_modules/bindings/bindings.js:77:44)

Please help me solve this,
Thanks in advance

styoe, 2016-08-11:
And the 46, 47, 48 versions i managed to get correlate to the node versions i installed.

46 for node < 4, 47 for node 5.* and 48 for node 6.*

The problem is that it is still 48 for the latest node arm build, and as far as i can see. te official page states 
Version	LTS	Date	V8	npm	NODE_MODULE_VERSION[1]	
Node.js v6.3.1		2016-07-21	5.0.71.57	3.10.3	48
https://nodejs.org/en/download/releases/

P

reconbot, 2016-08-11:
You're looking for electron-rebuild the version of "node" it uses doesn't
match anything released.

styoe, 2016-08-11:
But i tried many versions of electron-rebuild and electron... none of them worked. Are you telling me it cannot be solved?

reconbot, 2016-08-11:
It very much can be solved and many people have figured a way to do it. But
I think you'll need to open an issue with electron rebuild. I'm happy to
make any changes to better support electron and help debug.

styoe, 2016-08-11:
Thank you, i will. Is there nothing else i can do? and can you pls, if you have the time :) explain why can't i modify the serialport module somehow to use the same version of node as my electron app? As no other modules i am using are having this problem... 
I don't mean to be a drag, i'm just trying to understand this problem... i read this https://nodejs.org/api/addons.html and i understand Electron comes with it's own version of node bundled up, and serialport uses a different node version when it is creating the bindings? Could this be changed?

reconbot, 2016-08-13:
npm uses one version of node to download or compile the serialport binary and electron uses a different version to run it. You need to get electron to build the binary and then you're set.

styoe, 2016-08-13:
https://github.com/electron/electron-rebuild/issues/97 
I tried everything you see here, still nothing. Is there something i am doing wrong? 
What determines the version of the package? can i change something in the c++ code to trick electron into thinking that the serialport package is version 49?

styoe, 2016-08-14:
https://github.com/juliangruber/require-rebuild/issues/12 
even with require-rebuild it doesnt work,
Please help

styoe, 2016-08-14:
Allright! i managed to get it working by running sudo npm rebuild --runtime=electron --target=1.2.5 --disturl=https://atom.io/download/atom-shell --build-from-source
and by bumping my electron version to 1.2.6 :)
https://www.youtube.com/watch?v=U5TqIdff_DQ
Thank you for your help

reconbot, 2016-08-14:
great!

closed 2016-08-14 by reconbot

mcammaert, 2016-08-19:
@styoe thanks so much :)

Arrow7000, 2016-10-15:
Hi, is there any chance any of you would be able to take a look at a similar issue but for NodeJS on Ubuntu x64? I would really appreciate it! 

The issue is here: #964