serialport/node-serialport #911: Trouble installing serialport on Debian/Beaglebone

killabyte, 2016-08-18:
SerialPort version: 1.7.4
NodeJS Version: v0.12.6
Operating System and Hardware: Debian / Beaglebone
Are you using Electron/Atom/NwJS?: No
NPM : 2.11.2
Python: 2.7.9

Also running NVM which manages another NodeJS version 6.4.0
### Summary of Problem
Getting a build error from gyp:

serialport.target.mk:88: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/debian/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Linux 4.4.9-ti-r25
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/debian/node_modules/serialport/build/serialport/v1.7.4/Release/node-v48-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/home/debian/node_modules/serialport/build/serialport/v1.7.4/Release/node-v48-linux-arm"
gyp ERR! cwd /home/debian/node_modules/serialport
gyp ERR! node -v v6.4.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/debian/node_modules/serialport/build/serialport/v1.7.4/Release/node-v48-linux-arm/serialport.node --module_name=serialport --module_path=/home/debian/node_modules/serialport/build/serialport/v1.7.4/Release/node-v48-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/debian/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:852:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
node-pre-gyp ERR! System Linux 4.4.9-ti-r25
node-pre-gyp ERR! command "/usr/local/bin/node" "/home/debian/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/debian/node_modules/serialport
node-pre-gyp ERR! node -v v6.4.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/debian/node_modules/serialport/build/serialport/v1.7.4/Release/node-v48-linux-arm/serialport.node --module_name=serialport --module_path=/home/debian/node_modules/serialport/build/serialport/v1.7.4/Release/node-v48-linux-arm' (1)
npm WARN enoent ENOENT: no such file or directory, open '/home/debian/package.json'
npm WARN debian No description
npm WARN debian No repository field.
npm WARN debian No README data
npm WARN debian No license field.
npm ERR! Linux 4.4.9-ti-r25
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "--unsafe-perm" "serialport@1.7.4"
npm ERR! node v6.4.0
npm ERR! npm  v3.10.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.

### Expected Behavior

I have serialport running on an rPi using SerialPort version: 1.7.4 and NodeJS Version: v0.12.6. I got a beaglebone shipped with Node 6.4.0 and it was failing trying to install serialport. So I tried NVM to install the version I have working on the rPi, v0.12.6. If I run node -v on the beaglebone, it tells me I'm running v0.12.6. However, when I run 

sudo npm install --unsafe-perm serialport@1.7.4

I'm getting a build error. I'm not sure if the build process thinks I'm using Node 6.4.0. There is a line in the error:

gyp ERR! node -v v6.4.0

I've attached the log file.

### Actual Behavior

### Steps and Code to Reproduce the Issue

[npm-debug.log.zip](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/425799/npm-debug.log.zip)

killabyte, 2016-08-18:
There is this bit too in the failure output:

npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

killabyte, 2016-08-19:
It looks like I was able to fix this. I'm using xbee-rx which has a dependency for the serialport@1.7.4 version. I basically performed the steps from nathanielknopf from issue https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/842. 

> - installed node-pre-gyp with npm
- changed the owner of ~/node_modules/ to the user rather than root
- used git to clone the module from https://github.com/EmergingTechnologyAdvisors/node-serialport.git
- npm install inside the node-serialport folder
- moved the folder inside node_modules

> 
In my case I just downloaded a 1.7.x version from 
https://github.com/EmergingTechnologyAdvisors/node-serialport/releases?after=v2.0.1, compiled it and removed all other serialport directories before copying in the new compiled version.

matijagaspar, 2016-09-02:
I tried all of the options mentioned and I still keep getting: g++: internal compiler error: Segmentation fault (program cc1plus)

On Raspberry Pi B

reconbot, 2016-09-02:
What version of gcc do you have?

matijagaspar, 2016-09-03:
gcc version 4.9.2 (Raspbian 4.9.2-10)

reconbot, 2016-09-03:
Good, you need gcc at least 4.8. Bad, I'm not sure how to track down the issues with your g++

matijagaspar, 2016-09-03:
Ill try to downgrade the gcc if there is any change? I to have no idea how to get more useful information about what is wrong.

reconbot, 2016-09-06:
I completely missed that you're using serialport@1.7.4, you need to use at least serialport 3x for node 4 and above.

reconbot, 2016-09-06:
There's an [upgrade guide](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/UPGRADE_GUIDE.md) from 2x to 3x to 4x but nothing for 1x as that.. was a very long time ago.

fivdi, 2016-09-17:
The BeagleBone appears to be configured incorrectly here. The account used for the logon appears to use Node.js v0.12.6 and the root account appears to use Node.js v6.4.0.

fivdi, 2016-09-17:
What is the output of the following commands?
```
node -v
sudo node -v
```

closed 2016-10-11 by reconbot