serialport/node-serialport #912: msbuild.exe failed with exit code: 1

dfsq1311, 2016-08-20:
SerialPort version: 4.0.1
NodeJS Version: 6.3.1
Operating System and Hardware: windows7
Are you using Electron/Atom/NwJS?: Electron
python Version: 2.7.6
msvs_version: 2013

###  msbuild.exe failed with exit code: 1

when I run 'node-pre-gyp install --fallback-to-build', the serialport start pre-building, but throw err
```
\.node-gyp\iojs-1.3.3\Release\iojs.lib : fatal error LNK1107: invalid or corrupt file: cannot read at 0x108A76 [D:\workspace\electron-quick-start\node_modules\serialport\build\serialport.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\12.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (D:\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internalild_process.js:204:12)
gyp ERR! System Windows_NT 6.1.7601
```

When I just run 'npm install serialport --build-from-source --msvs_version=2013', every thing is OK.

### Expected Behavior

[image]

### Actual Behavior

[image]

### Steps and Code to Reproduce the Issue

#### Code
https://github.com/dfsq1311/electron-quick-start-serialport

#### Steps
run command on win32 x64
```
npm install
```

dfsq1311, 2016-08-20:
my package.json below
`  "scripts": {
    "install": "node-pre-gyp install --fallback-to-build --build-from-source --runtime=electron --target=1.3.3 --target_arch=x64 --directory=node_modules/serialport --update-binary --dist-url=https://atom.io/download/atom-shell",
    "start": "electron main.js"
  },`

reconbot, 2016-08-22:
Why is it building iojs.lib? That seems out of place. Hopefully a read herring.

dfsq1311, 2016-08-29:
I delete ~/.node-gyp, then run `npm install node-gyp -g`, it work.

closed 2016-08-29 by dfsq1311

reconbot, 2016-08-29:
Yay!