serialport/node-serialport #914: Bug when more than three ports on win10

konyee, 2016-08-23:
SerialPort version: 4.0.1 
NodeJS Version: 4.4.7 and 6.3.1
Operating System and Hardware: Win10 64bit
Are you using Electron/Atom/NwJS?: no

(Run `node -v && npm -v && python --version` to get version numbers)
v4.4.7
2.15.8
Python 2.7.10

### Summary of Problem
Cannot create more then 3 serialports on Win10. 

### Expected Behavior
Be open more ports.

### Actual Behavior
If I create 4 or more it cannot be closed. 
It also crashed my express webserver.

### Steps and Code to Reproduce the Issue
If you put more than 3 portname to ports, the test will stuck after 'closing'.
If less then 4, everything is ok.

As I see the problem maybe at UV_THREADPOOL_SIZE 
http://docs.libuv.org/en/v1.x/threadpool.html#

[serialTest.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/433438/serialTest.txt)

konyee, 2016-08-25:
I found a workaround for this.
http://stackoverflow.com/questions/30983672/node-serialport-on-windows-with-multiple-devices-hangs

Set UV_THREADPOOL_SIZE enviroment variable larger than count of serial ports BEFORE you start your app.

reconbot, 2016-09-02:
I'm glad you found this. Sorry it's an issue. I'm going to keep this open pending the 5x release which might be able to address this issue on windows.

konyee, 2016-09-09:
You're welcome. :)

deandob, 2016-09-22:
Would be great if 5.0 release could include a fix for this rather than setting environmental variables outside the program.

reconbot, 2016-09-22:
The current bindings require a thread per port, there's not much we can do about that other than rearchitect the bindings.

deandob, 2016-09-23:
Ok thanks for confirming. I'll factor the environment variable into the setup for my app.

Zensey, 2016-11-02:
To work with many ports on windows one should use IO completion ports (IOCP) API.

1. Create completion port: CreateIoCompletionPort
2. Open port with overlapped flag, set port params. Add port fd to IOCP.
4. Issue ReadFile for port fd.
3. In separate thread do GetQueuedCompletionStatus in loop and get i/o state, grab data, emit events.

reconbot, 2016-11-02:
I completely agree, our current method of windows reading is far from optimal and this is the right way to go. I haven't explored it but it but I think [`uvpoll_t`](http://docs.libuv.org/en/v1.x/poll.html) basically does that, and I'm hoping that we can use it. I haven't gotten anything to work yet, and I welcome any patches.

Zensey, 2016-11-03:
AFAIK libuv didn't have IOCP support, at least 1 year ago.

jaredrolson, 2017-03-31:
Hi, I'm having this same problem and would really like to have 4+ ports open simultaneously. 

I understand that I need to change an environment variable before the app is setup, but I don't understand how to do that. I think I'm supposed to paste this line of code somewhere:

`process.env.UV_THREADPOOL_SIZE = 6;`

But I don't understand where to put that code. Can someone point me in the right direction? I pasted at the very top of my main app, but it's not working.

Thanks

reconbot, 2017-03-31:
The environment variable needs to be set before node starts.

---
Francis Gulotta
wizard@roborooter.com

jaredrolson, 2017-04-01:
Perfect thanks. I have it working now. I'm running my code from the command prompt, and it worked when I entered this into the prompt before running my app:

`set UV_THREADPOOL_SIZE=8`

jupe, 2017-04-01:
I've same problem. Please mention this limitaion and env variable in release notes so people know about this..  Unfortunately it takes long time before I saw this issue and possible workaround fix ..

reconbot, 2017-04-01:
PRS welcome

reconbot, 2017-04-01:
Also not an issue for 5x beta

reconbot, 2017-07-06:
I've figured out the cause of this. I've opened #1221 as a feature request to fix this. If I can I'l have it out in the 5.0.0 release.

closed 2017-07-06 by reconbot