serialport/node-serialport #916: Getting the problem while installing the seriaport in Raspberry pi3

titusece, 2016-08-25:
Dear All,
I got the below error while installing the seriaport for Xbee in node-red.

I have also tried like this "npm install serialport"

**Version details:**
root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red# 
root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red# 
root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red# npm -v
n3.8.6
root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red# node- v
bash: node-: command not found
root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red# node -v
v5.12.0
root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red# nodejs -v
v5.12.0
root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red#

**LOG**

root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red# 
root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red# npm install serialport@2.1.2

> serialport@4.0.1 install /home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-node-v47-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and node@5.12.0 (node-v47 ABI) (falling back to source compile with node-gyp) 
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/5.12.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/.node-gyp"
make: Entering directory '/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/5.12.0/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:100:13)
gyp ERR! stack     at ChildProcess.emit (events.js:185:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Linux 4.1.18-v7+
gyp ERR! command "/usr/bin/nodejs" "/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/build/Release"
gyp ERR! cwd /home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport
gyp ERR! node -v v5.12.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:100:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:185:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:850:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
node-pre-gyp ERR! System Linux 4.1.18-v7+
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport
node-pre-gyp ERR! node -v v5.12.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/nodejs /home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/johnny-five/node_modules/serialport/build/Release' (1)
npm WARN install:serialport@4.0.1 serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
npm WARN install:serialport@4.0.1 Exit status 1

> serialport@4.0.1 install /home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/firmata/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-node-v47-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and node@5.12.0 (node-v47 ABI) (falling back to source compile with node-gyp) 
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/5.12.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/firmata/node_modules/serialport/.node-gyp"
make: Entering directory '/home/pi/Desktop/rPi_Titus/node-red-new/node-red/node_modules/firmata/node_modules/serialport/build'
[... 73 lines cut ...]
npm ERR! Linux 4.1.18-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "serialport@2.1.2"
npm ERR! node v5.12.0
npm ERR! npm  v3.8.6
npm ERR! code ELIFECYCLE

npm ERR! serialport@2.1.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@2.1.2 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/Desktop/rPi_Titus/node-red-new/node-red/npm-debug.log
root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red# 
root@raspberrypi:/home/pi/Desktop/rPi_Titus/node-red-new/node-red#

reconbot, 2016-09-06:
This is a common debian problem as they have the node binary named `nodejs` instead of `node`.  See our [readme for more information](https://github.com/EmergingTechnologyAdvisors/node-serialport#ubuntudebian-linux). Also serialport@2.1.2 might not work with node 4x and above. I'd recommend using at least 3x, if you can link me to the package I'd be happy to help them upgrade.

closed 2016-09-06 by reconbot

titusece, 2016-09-07:
Thanks for the reply reconbot.
I'm able to install the serial port now.

Thank you!

reconbot, 2016-09-07:
Can you share the xbee library? I'd like to take a look.

titusece, 2016-09-10:
I have taken the Xbee from the below link.
https://github.com/jankolkmeier/NR-XBeeAPI

And also I have got help from this group.
https://groups.google.com/forum/#!topic/node-red/KJZBMjjPhbc

I hope this helps.