serialport/node-serialport #917: [feature request] byteDelimiter with fixed length trailing bytes

mariotacke, 2016-08-26:
### Summary of Problem
I'm working with a Japanese (SANYO) serial protocol where messages are byte delimited and include a four byte checksum **after** the delimiter.

**Example**
Byte delimiter in the following messages would be `[ 2, 3 ]`. Both messages include a trailing checksum of length 4 (`[ 66, 67, 51, 65 ]` and `[ 51, 66, 70, 57 ]` respectively).

```js
message1 = [ 2, 48, 48, 52, 128, 49, 48, 48, 3, 66, 67, 51, 65 ]
message2 = [ 2, 48, 48, 55, 128, 49, 48, 49, 50, 51, 51, 3, 51, 66, 70, 57 ]
```

### Expected Behavior
Define a byteDelimiter and optional fixed length after to determine a message. The parser should not emit until the buffer consists of the regular delimited data plus fixed length after.

Possible backwards compatible API
```js
var SerialPort = require('serialport');

var port = new SerialPort('/dev/tty-usbserial1', {
  parser: SerialPort.parsers.byteDelimiter([ 2, 3 ], 4)
});
```

reconbot, 2016-08-26:
I'd create a transform stream to do this. Eventually the parsers will be
replaced with this approach.

mariotacke, 2016-08-26:
Do you happen to know of an example implementation on how to do this?

reconbot, 2016-09-02:
This is the current rewrite of line parser as a transform stream. I make use of the `inherits` and `readable-stream` packages both of which will be included in serialport in the next release.

```js
'use strict';
var Transform = require('readable-stream').Transform;
var inherits = require('inherits');

function ReadLineParser(options) {
  if (!(this instanceof ReadLineParser)) {
    return new ReadLineParser(options);
  }
  Transform.call(this, options);

  options = options || {};

  if (options.delimiter === undefined) {
    options.delimiter = new Buffer('\n', 'utf8');
  }

  this.delimiter = new Buffer(options.delimiter);
  this.setEncoding(options.encoding || 'utf8');
  this.buffer = new Buffer(0);
}

ReadLineParser.prototype._transform = function(chunk, encoding, cb) {
  var data = Buffer.concat([this.buffer, chunk]);
  var position;
  while ((position = data.indexOf(this.delimiter)) !== -1) {
    this.push(data.slice(0, position));
    data = data.slice(position + this.delimiter.length);
  }
  this.buffer = data;
  cb();
};

ReadLineParser.prototype._flush = function(cb) {
  this.push(this.buffer);
  this.buffer = new Buffer(0);
  cb();
};

inherits(ReadLineParser, Transform);

module.exports = ReadLineParser;
```

reconbot, 2016-09-06:
I've updated my comment to reflect the byte delimiter transform stream. You'd want to modify this part.

```js
  while ((position = bindexOf(data, this.delimiter)) !== -1) {
    this.push(data.slice(0, position));
    data = data.slice(position + this.delimiter.length);
  }
```

To look for the delimiter and then 4 extra bytes, and if it matches remove the delimiter and the 4 extra bytes

```js
  while ((position = bindexOf(data, this.delimiter)) !== -1) {
   if ((position + 4) > data.length) { break; }
    this.push(data.slice(0, position));
    data = data.slice(position + this.delimiter.length + 4);
  }
```

This is untested, but hopefully puts you on the right path. Let me know how it works out!

reconbot, 2016-09-06:
And if you publish the stream let me know and I'll link to it!

closed 2016-09-06 by reconbot

mariotacke, 2016-09-06:
Thank you very much. I looked into the source and wrote my own custom parser instead. I definitely need to ramp up on transform streams and your suggested code helps a ton!

Here is my implementation for the byteDelimiter + trailing 4 bytes.
```
module.exports = {
  sanyo: () => {
    let buf = [];
    let length = 0;

    return function (emitter, buffer) {
      for (let i = 0; i < buffer.length; i++) {
        buf[buf.length] = buffer[i];

        if (buf.length && buffer[i] === 3) {
          length = buf.length + 4;
        }

        if (length && buf.length === length) {
          emitter.emit('data', buf);
          buf = [];
          length = 0;
        }
      }
    };
  }
};
```