serialport/node-serialport #919: weird application exit/crash if usb-to-serial cable is disconnected

taramid, 2016-08-30:
SerialPort version: 4.0.1
NodeJS Version: 6.4.0
Operating System and Hardware: Win7 x64, Intel Core i5
Are you using Electron/Atom/NwJS?: no

### Summary of Problem
trying to create a robust application I have found that usb-to-serial cable disconnect causes application to exit with out stack trace.

perhaps this is not serialport library issue though

### Expected Behavior 
the library emits 'error'/'disconnect' event 

### Actual Behavior
unhandled application exit/crash with console output:
```
undefined:0

TypeError: undefined is not a function
```

### Steps and Code to Reproduce the Issue
a code:
```
	var SerialPort = require('serialport');
	var serial = new SerialPort(
		path,
		{
			baudRate: 9600,
			dataBits: 8,
			stopBits: 1,
			parity: 'none',
			rtscts: true
		}
	);
	// nothing fancy - just a standard creation of a SerialPort instance;

```
steps: run the code and disconnect the cable (I use usb-to-serial Profilic one)

taramid, 2016-08-30:
on Linux (Raspbian on Raspberry Pi 3; Node v6.4.0) the same situation is handled correctly, emitting two events: 'disconnect' (with error "Error bad file descriptor on polling..." and 'close' 

it makes me pretty sure that the issue happens beyond the library; where can it be addressed to?

reconbot, 2016-09-02:
It's the library's fault, we have #587 open to track it. I've been working on cleaning up how this works for our next major release. The behavior isn't yet completely consistent between platforms.

closed 2016-09-02 by reconbot