serialport/node-serialport #921: serialport-term -b 19200 returns Error: SetCommState: Unknown error code 87

davidchristine88, 2016-08-31:
SerialPort version: 4.0.1
NodeJS Version: 4.5.0
Operating System and Hardware: Windows 7 Professional SP1 64 bit
Are you using Electron/Atom/NwJS?: Yes

(Run `node -v && npm -v && python --version` to get version numbers)
v4.5.0
2.1.6
Python 2.7.12

### Summary of Problem
serialport-term -p COM8 works at default 9600.   When adding -b 19200 option, it returns the error
Error [Error: SetCommState: Unknown error code 87]

### Expected Behavior
>serialport-term -p COM8 -b 19200

### Actual Behavior
>serialport-term -p COM8 -b 19200
Error [Error: SetCommState: Unknown error code 87]

>
### Steps and Code to Reproduce the Issue
serialport-term -p COM8 -b 19200

reconbot, 2016-08-31:
What are you talking to? This comes from windows when it can't change the baud rate. The only relevant thread I've found is this one which blames the hardware that's being hooked up. https://social.msdn.microsoft.com/Forums/en-US/ad18e2a1-c25b-43e7-8b8d-019073db2c6f/why-setcommstate-error-87-?forum=vssmartdevicesnative

davidchristine88, 2016-09-01:
I am using a FTDI USB to RS232 converter.   The manufacturer and SW driver provider is FTDI and Driver Date is 3/9/2016.  Driver Version 2.12.16.0.   Digital Signer is MS Windows Hardware Compatibility Publisher.   This converter is been working with other Terminal program such as Putty and TerraTerm.

radio-miskovice, 2016-12-03:
I have the same issue. I suspect it may be related to **commander** module, as the -b option is always parsed as NaN ("not a number") regardless of the actual value.

reconbot, 2017-07-10:
We did get this fixed serialport-4.0.7

closed 2017-07-10 by reconbot