serialport/node-serialport #925: close midstream results in "Cannot read property 'length' of undefined"

baffo32, 2016-09-08:
SerialPort version: 4.0.1
NodeJS Version: 4.5.0
Operating System and Hardware: Debian Linux, RFDuino
Are you using Electron/Atom/NwJS?: No

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem

When I close my serial port while data is arriving, often an error is thrown.

### Expected Behavior

Close gracefully.

### Actual Behavior
```
buffer.js:281
      length += list[i].length;
                       ^

TypeError: Cannot read property 'length' of undefined
    at Function.Buffer.concat (buffer.js:281:24)
    at SerialPort.<anonymous> (node_modules/serialport/lib/serialport.js:287:32)
    at SerialPort.<anonymous> (node_modules/serialport/lib/serialport.js:303:7)
    at FSReqWrap.wrapper [as oncomplete] (fs.js:576:17)
```

### Steps and Code to Reproduce the Issue

This happens intermittently for me.  I haven't found yet how to reliably reproduce it.

reconbot, 2016-09-08:
Wierd.. [it looks like](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/4.0.1/lib/serialport.js#L287) the internal buffer is getting cleared on `resume()` and that might cause this issue. I'm not sure why resume would be getting called. It also look like we never initialize  `this.buffer` that is a total bug.

Thanks for the bug report!

baffo32, 2016-09-08:
I was guessing that the point of this.buffer was to hold data during pause to be emitted on resume, and that it is kept undefined when running.  I'm using this patch:
```
diff --git a/lib/serialport.js b/lib/serialport.js
index b5c36ee..485d5a3 100644
--- a/lib/serialport.js
+++ b/lib/serialport.js
@@ -348,7 +348,10 @@ if (process.platform !== 'win32') {
 
         // do not emit events if the stream is paused
         if (this.paused) {
-          this.buffer = Buffer.concat([this.buffer, b]);
+          if (this.buffer)
+            this.buffer = Buffer.concat([this.buffer, b]);
+          else
+            this.buffer = b;
           return;
         }
         this._emitData(b);
```

reconbot, 2016-09-08:
it shouldn't be buffering anything at all tbh, that's the responsibility of
the stream internals

---
Francis Gulotta
wizard@roborooter.com

N4TY, 2016-09-16:
unsubscribe

ysyyork, 2016-09-28:
Hi, I came across the same issue. It seems not solvable from serialPort end but is there a workaround? Now it seems random when I am trying to close the port and this error occurs.

reconbot, 2016-09-28:
@ysyyork we'll solve it pretty soon, no known workarounds right now, sorry.

reconbot, 2016-09-28:
ok this should be fixed in master you can use that or I'll have `5.0.0-beta2` out in a few days probably.

closed 2016-09-28 by reconbot

ysyyork, 2016-09-28:
awesome, thanks!

reconbot, 2016-10-05:
I've released [Serialport@4.0.3](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/4.0.3) which includes this fix.

PR #945 (2016-09-28): Plug a hole in the linux/unix read function
`this.buffer` is never initialized so if we close the port while reading we'll get an error. This has no tests but doesn't fail any of our current tests. With the streams rewrite this is no longer an issue and instead of testing it fully here I'd rather concentrate on a new implementation of it.

Closes #925