serialport/node-serialport #926: Load node-serialport with Electron+Angular 2

victorhsn, 2016-09-08:
SerialPort version: 4.0.1
NodeJS Version: 6.5
Operating System and Hardware: Windows, desktop.
Are you using Electron/Atom/NwJS?: Electron 1.3.5
Python: 2.7

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem

Currently I'm using Electron + Angular 2, whole application is working, but when I'll to load the module node-serialport, the ng2 stop all features, as the "ajax" had stop. On the other hand, if I click under some component as input the ng2 back to work again.

### Expected Behavior

That's the angular 2 should to load with Electron + node-serialport.

### Actual Behavior

### Steps and Code to Reproduce the Issue

reconbot, 2016-09-08:
I don't know too much about Angular 2. "the ng2 stop all features" What does that mean? 

Are you using webpack or browserify?

victorhsn, 2016-09-09:
Hi @reconbot, It means that some angular's support stop. 

But I found the problem. There is something wrong, because when I execute the command "npm install serialport", the archives downloaded, specifically the file in the path serialport/lib/list-unix is not like the archive that is on github (list-linux), so if I comment the following lines code;

`var linux = process.platform !== 'win32' && process.platform !== 'darwin';

function listLinux(callback) {
  callback = callback || function(err) {
    if (err) { this.emit('error', err) }
  }.bind(this);
  return listUnix(callback);
};`

`module.exports = {
... 
 list: linux ? listLinux : bindings.list
...
}`

My application back to works normally... Now I'll to do download of github and rebuild the module to use in my app.

reconbot, 2016-09-09:
What error does this section of code produce? It's valid javascript.

The code on github in master is for the beta releases, you want to look for the tag of your version of serial port you installed.

reconbot, 2016-09-09:
If you got it working, I guess I'll close the issue. We'll have a beta release in a week or two.

closed 2016-09-09 by reconbot

victorhsn, 2016-09-09:
Thank you @reconbot, the version that is coming go to fix the problem with ng2.

vicatcu, 2017-06-12:
Hey @victorhsn got any pointers on what you did to get this to work? I think I'm experiencing a similar problem. See this gist for my error output: https://gist.github.com/vicatcu/1a33e942511342e5b925fdaa0ebd16e6