serialport/node-serialport #931: Raspbian SerialPort NPM Install Error

alex-ac2, 2016-09-13:
SerialPort Version: n/a
NodeJS Version: v4.5.0
npm Version: 2.15.9
Operating System and Hardware: Raspbian -- Raspberry Pi3 
Are you using Electron/Atom/NwJS?: n/a

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem
Unable to install the serialport module using npm.  

### Expected Behavior
To install the serialport module as a local dependency for a node application using johnny-five 

### Actual Behavior
ac@berry1:~/j5 $ sudo npm install serialport --save
npm WARN package.json testj5@1.0.0 No repository field.
npm WARN package.json testj5@1.0.0 No README data
-
> serialport@4.0.1 install /home/ac/j5/node_modules/serialport
> node-pre-gyp install --fallback-to-build
 
node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-node-v46-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and node@4.5.0 (node-v46 ABI) (falling back to source compile with node-gyp)
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.5.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/ac/j5/node_modules/serialport/.node-gyp"
make: Entering directory '/home/ac/j5/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/4.5.0/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/home/ac/j5/node_modules/serialport/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.4.11-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/ac/j5/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/ac/j5/node_modules/serialport/build/Release"
gyp ERR! cwd /home/ac/j5/node_modules/serialport
gyp ERR! node -v v4.5.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home
/ac/j5/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/ac/j5/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/ac/j5/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:829:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 4.4.11-v7+
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/ac/j5/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/ac/j5/node_modules/serialport
node-pre-gyp ERR! node -v v4.5.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/ac/j5/node_modules/serialport/
build/Release/serialport.node --module_name=serialport --module_path=/home/ac/j5/node_modules/serialport/build/Release' (1)
npm ERR! Linux 4.4.11-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "serialport" "--save"
npm ERR! node v4.5.0
npm ERR! npm  v2.15.9
npm ERR! code ELIFECYCLE
 
npm ERR! serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.
 
npm ERR! Please include the following file with any support request:
npm ERR!     /home/ac/j5/npm-debug.log
 
ac@berry1:~/j5 $

### Steps and Code to Reproduce the Issue
ac@berry1:~/j5 $ sudo npm install serialport --save

secretdog, 2016-09-13:
I have the same issue.

alex-ac2, 2016-09-13:
Hey, have you come across any documentation to help troubleshoot this?

reconbot, 2016-09-13:
How have you found the guide in the readme?

reconbot, 2016-09-13:
https://github.com/nebrius/raspi-io/wiki/Getting-a-Raspberry-Pi-ready-for-NodeBots

reconbot, 2016-09-13:
@nebrius Alex has followed your guide but is still hitting issues. Any ideas?

nebrius, 2016-09-15:
@alex-ac2 which version of Raspbian are you using? Also, can you list the output of `gcc -v` for us?

fivdi, 2016-09-17:
The command being used to install serialport here is `sudo npm install serialport` (note the `sudo`.) Understanding the implications of `sudo` here requires fairly deep knowledge of some aspects of how Linux works. For more information see [here](https://github.com/nodejs/node/issues/8545#issuecomment-247335172). Alternatively, install serialport with `npm install serialport` instead.

justinwright, 2016-09-26:
I've run into this issue as well. I'm able to get around it by just side loading the plugin. Assuming you've got an existing project, clone the serialport plugin somewhere else on your computer. Run `sudo npm link` in that plugin folder, then change to your project location and link it with `sudo npm link serialport`. I've also had to run `sudo node-gyp configure build` in the serialport location to get rid of some more errors in my project.

reconbot, 2016-10-05:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2016-10-05 by reconbot

gbaumgart, 2016-10-24:
still doesn't work with 4.0.3, npm install serialport -d shows that it tries to download https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.3/serialport-v4.0.3-node-v46-linux-arm.tar.gz which doesnt exists.....

Do you think forking and fixing the installer may work ?

reconbot, 2016-10-25:
> node-serialport/releases/download/4.0.3/serialport-v4.
> 0.3-node-v46-linux-arm.tar.gz which doesnt exists.....

We don't have arm binaries, they need to be build on the local machine.

---
Francis Gulotta
wizard@roborooter.com