serialport/node-serialport #932: 5.0.0-beta1 - parser with regex support missing

jupe, 2016-09-15:
SerialPort version: 5.0.0-beta1

### Summary of Problem
previous release 4.0.1 allow to use [readline](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/4.0.1/lib/parsers.js#L12) parser with regex delimiter, but seems that it is not possible anymore. It would be useful in some cases where delimiter is not "hardcoded" string. For example one of my application uses delimiter `/\r\n|\n\r|\n|\r/` which eats all _carriage return_ (`CR`) and _line feed_ (`LF`) characters away in what ever order they come.. 

I think it shouldn't be hard to further develop DelimiterParser a bit to use regex instead of string..

### Expected Behavior
possible to use DelimiterParser with regex

### Actual Behavior
regex delimiter is not possible to use

reconbot, 2016-09-19:
A+ catch. I missed this was a feature. It might have been a side effect. =)

RomanSaveljev, 2016-09-19:
Maybe it is ought to be a custom parser type to avoid bogging down the readline streaming parser performance?

reconbot, 2016-09-20:
100% agreed. I was thinking maybe a transform stream that takes a function to do it's dirty work? Like something you'd pass to `array.reduce`?

reconbot, 2017-07-10:
Closed by @jessicaquynh

closed 2017-07-10 by reconbot