serialport/node-serialport #935: Could not locate the bindings file under win7 Electron desktop app

zhoutianwu, 2016-09-19:
I have an Electron desktop app that requires some nodejs packages including serialport. But only serialport is failed to require.

	* Operating System and Hardware: windows7
	* Electron version: 1.3.1
	* electron-prebuilt: 0.36.12
	* serialport: 4.0.1
	* node version : 6.3.1
	* python version: 2.7.11

Electron desktop app require serialport,
var SerialPort = require("serialport") 

run app, show error:
main.js:125632 Uncaught Error: Could not locate the bindings file. Tried:
→ C:\Users\Administrator\Hozz5\build\serialport.node
→ C:\Users\Administrator\Hozz5\build\Debug\serialport.node
→ C:\Users\Administrator\Hozz5\build\Release\serialport.node
→ C:\Users\Administrator\Hozz5\out\Debug\serialport.node
→ C:\Users\Administrator\Hozz5\Debug\serialport.node
→ C:\Users\Administrator\Hozz5\out\Release\serialport.node
→ C:\Users\Administrator\Hozz5\Release\serialport.node
→ C:\Users\Administrator\Hozz5\build\default\serialport.node
→ C:\Users\Administrator\Hozz5\compiled\6.3.0\win32\x64\serialport.node

it not locate the bindings file under foler C:\Users\Administrator\Hozz5\node_modules\serialport\build\Release
C:\Users\Administrator\Hozz5\node_modules\serialport\build\Release has serialport.node file

I copy C:\Users\Administrator\Hozz5\node_modules\serialport\build\Release folder to   C:\Users\Administrator\Hozz5 ,it also show same error!

ZhangYunP, 2016-09-25:
I have the same problem .

reconbot, 2016-09-25:
You need to rebuild them. Look at electron-rebuild

zhoutianwu, 2016-09-26:
C:\Users\Administrator\Hozz5>node_modules\.bin\electron-rebuild

then npm start, also show error Could not locate the bindings file.

reconbot, 2016-09-28:
Please check out #538 people have shared how they have gotten electron to work there

closed 2016-09-28 by reconbot