serialport/node-serialport #937: TypeError: SerialPort.parsers.ReadLine is not a function

JMishou, 2016-09-20:
SerialPort version: 4.x
NodeJS Version: 4.5
Operating System and Hardware: raspian jesse lite & pi zero
Are you using Electron/Atom/NwJS?: No

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem

Reading the serial port  without a parser works just fine; but, when running the example code for the readline parser I receive this error:

    var parser = SerialPort.parsers.ReadLine();
                                    ^
    TypeError: SerialPort.parsers.ReadLine is not a function
        at Object.<anonymous> (/home/pi/rfid/serialport.js:4:33)
        at Module._compile (module.js:409:26)
        at Object.Module._extensions..js (module.js:416:10)
        at Module.load (module.js:343:32)
        at Function.Module._load (module.js:300:12)
        at Function.Module.runMain (module.js:441:10)
        at startup (node.js:139:18)
        at node.js:974:3

### Expected Behavior

That the example code for the readline parser would run:

    var SerialPort = require('serialport');
    var ReadLine = SerialPort.parsers.ReadLine;
    var port = new SerialPort('/dev/serial0');
    var parser = SerialPort.parsers.ReadLine();
    port.pipe(parser);
    parser.on('data', console.log);

Any idea what could be causing this error?  I have installed the code using

`npm install serialport`

and 

`npm install serialport --build-from-source`

I installed node two different ways with the same results... self compiled and downloading the arm6 binaries.

reconbot, 2016-09-20:
You're looking at the docs for the latest master not for 4.x https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/4.0.1/README.md

JMishou, 2016-09-20:
Got it! Thanks!

closed 2016-09-20 by JMishou

saisurya76, 2017-08-03:
Hi NMishou, 

serialport.parsers.readline("\n") gives error 
"TypeError: serialport.parsers.readline is not a function".

can u share how you could clear this, please?

Thanks,
Surya

reconbot, 2017-08-03:
@saisurya76 they were looking at the wrong docs for their version of serialport. The api has changed between versions.

islenny, 2017-08-04:
@saisurya76 reinstall serialport with version 4.0.7, it works for me.

reconbot, 2017-08-04:
Or upgrade your code to 5x, which is a lot faster and less buggy.

jtleaming, 2017-08-07:
I'm getting a similar error from readline('\r\n'). 
"TypeError: RealSerialPort.parsers.readline is not a function"

I'm updated to the latest 5.0.0

reconbot, 2017-08-07:
Please see the parsers documentation for 5x, the way you use them have changed.
https://github.com/EmergingTechnologyAdvisors/node-serialport#module_serialport--SerialPort.parsers

jtleaming, 2017-08-07:
serialport.pipe(new Readline({ delimiter: '\r\n' }))

Now I'm getting an error that pipe is not a function: "TypeError: serialport.pipe is not a function"

reconbot, 2017-08-07:
What is the variable serialport? if it's a port object then it will have a
pipe.

---
Francis Gulotta
wizard@roborooter.com

jtleaming, 2017-08-07:
const Readline = RealSerialPort.parsers.Readline;

  const serialport = new SerialPort({
    io: io,
    route: '/port/COM'+port_number,
    captureFile: './COM'+port_number+'.out',
    retryPeriod: 1000,
    device: 'COM'+port_number,
    options: {
      baudrate: 38400,
      //parser: new Readline({delimiter:'\r\n'}) //RealSerialPort.parsers.readline('\r\n')
    }
  });
    const parser = serialport.pipe(new Readline({ delimiter: '\r\n' }));
    parser.on('data', console.log);

reconbot, 2017-08-07:
If `const serialport = new SerialPort()` doesn't give you a stream with a pipe method then you're not using serialport 5. I'd console log it out and see what you've got. You can also play around in the repl by running `node` and pasting your program in, it will let you see what the objects look like.

What's `RealSerialPort` vs `SerialPort`?

jtleaming, 2017-08-07:
SerialPort is a 'socket.io-serialport' while RealSerialPort is a 'serialport' 5.

reconbot, 2017-08-07:
I don't know what a `socket.io-serialport` is.

jtleaming, 2017-08-07:
Another NPM package. The code I was running used it to open the specified serial port. I had to roll back to version 4.0.1 to get it to work.

reconbot, 2017-08-07:
If anyone else has trouble with that package, you can open issues with it directly at it's github. Maybe ask them to upgrade.

https://github.com/pghalliday/socket.io-serialport