serialport/node-serialport #938: Error: Permission denied, cannot open /dev/ttyS1

FLCos, 2016-09-22:
SerialPort version:^4.0.1
NodeJS Version: v4.4.7
Operating System and Hardware: CentOS 7(virtual machine)
Are you using Electron/Atom/NwJS?: no, Just learn what is NwJS. 

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem
Can't open serial port, permission denied. 
I have tried to add my user to the group dialout:
`sudo usermod -a -G dialout pi`
But the problem still occurred.

### Expected Behavior
Open port successfully, write data.
### Actual Behavior
Can't open port.
### Steps and Code to Reproduce the Issue
Here is my code: 
```
var SerialPort = require('serialport');
SerialPort.list(function (err, ports) {
    if(err){
        console.log('Error: ' + err);
    }
    ports.forEach(function(port) {
        console.log(port.comName);
        console.log(port.pnpId);
        console.log(port.manufacturer);
        console.dir(port);
    });
});

var port = new SerialPort('/dev/ttyS0');
// var port = new SerialPort('/dev/ttyS1');
// var port = new SerialPort('/dev/ttyS2');
// var port = new SerialPort('/dev/ttyS3');

port.on('open', function(openErr) {
    if(openErr){
        return console.log('Error opening port: ', openErr.message);
    }
    port.write('main screen turn on', function(err) {
        if (err) {
            return console.log('Error on write: ', err.message);
        }
        console.log('message written');
    });
});
```
I tried four times with changing the port name. And the result:
```
Error:  Error: Permission denied, cannot open /dev/ttyS3
Error:  Error: Permission denied, cannot open /dev/ttyS2
Error:  Error: Permission denied, cannot open /dev/ttyS1
Error:  Error: Permission denied, cannot open /dev/ttyS0
/dev/ttyS0
undefined
undefined
/dev/ttyS1
undefined
undefined
/dev/ttyS2
undefined
undefined
/dev/ttyS3
undefined
undefined

```

FLCos, 2016-09-22:
Actually, I have tried the example code in windows 7 system, it's OK: 
var port = new SerialPort('COM2');
I think main problem is about the centOS system, maybe another access right need to add?

reconbot, 2016-09-22:
Yeah, looks like you need to be in a group with access to the ports.

closed 2016-09-22 by reconbot